StarLabsLtd/firmware #4: Consider trackpad input when determining keyboard backlight timeout

cassidyjames, 2020-11-15:
Not sure if this is feasible, but it's a nice touch I've seen on other laptops: the system should consider trackpad input when determining keyboard backlight timeout. The idea being:

1. It's easier to find the trackpad than a specific key without the backlight; and less "destructive" to wiggle the cursor to get the backlight to come back than to hit a (potentially wrong) key.
2. If I'm using the trackpad for 30 secs or so, it doesn't mean I'm done using the keyboard altogether.

Sean-StarLabs, 2020-11-20:
Nice idea. EC controls the backlight, so we just need to make the EC aware when the trackpad is used.

Sean-StarLabs, 2021-09-23:
Had a proper look at this, all the trackpad input is handled by Linux - so we'd need something in there to do the waking up.

I _think_ it could be achieved by sending a wake scan code `E0 5E` or an unassigned one `E0 7F`.

Or, we could add something into the EC memory, which could be written to from the Linux (that would be purged after the same amount of time as the keyboard backlight timeout).

closed 2021-11-30 by Sean-StarLabs