StarLabsLtd/firmware #5: Remember (or just lower) brightness for OEM logo

cassidyjames, 2020-11-15:
## Problem

When I boot my Star LabTop Mk IV up, it looks like the screen is initially at full brightness, and then once the OS loads, it restores my previous brightness. This means on initial boot—especially in a darker room, I see quite a bit of blacklight bleed. While this is normal on IPS displays, it is really exasperated by the full brightness and gives a worse first-impression.

## Proposal

Would it be possible to somehow load the boot screen at that saved brightness? If not, would it make sense to just lower the brightness of the screen during the boot logo? 

## Prior Art

Pixel phones remember he last-used light/dark style of the OS and load a light or dark variant of the boot splash. That's not exactly the same, but it's a similar idea.

Sean-StarLabs, 2020-11-20:
Default brightness has been changed from 255 to 100 in 1.5.0. Hoping to release it next week.

Sean-StarLabs, 2020-11-24:
Resolved with 4d9d71d526185473a38c761c28fb7ee4e63f3e79

closed 2020-11-24 by Sean-StarLabs