StarLabsLtd/firmware #8: Battery always reports it's charging

voider1, 2020-11-21:
Hi, after I updated to the latest firmware the battery began reporting it's always charging even when unplugged. ACPI tells me it's charging. This is what upower says when it's unplugged:

```
❯ upower -i /org/freedesktop/UPower/devices/battery_BAT0
  native-path:          BAT0
  vendor:               GDPT
  model:                CN6613-2S3P
  serial:               6UA3
  power supply:         yes
  updated:              Sat 21 Nov 2020 02:05:33 AM CET (9 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               charging
    warning-level:       none
    energy:              26.7596 Wh
    energy-empty:        0 Wh
    energy-full:         46.474 Wh
    energy-full-design:  45.6 Wh
    energy-rate:         0.2584 W
    voltage:             7.635 V
    percentage:          55%
    capacity:            100%
    icon-name:          'battery-good-charging-symbolic'
  History (charge):
    1605920733  55.000  charging
  History (rate):
    1605920733  0.258   charging

```

Sean-StarLabs, 2020-11-24:
Starting with the laptop turned off, can you press and hold `Ctrl` + `Alt` + `Del` + `Power` until the LED's on the left side of the laptop flash?

voider1, 2020-12-04:
That fixed it, perfect.

closed 2020-12-04 by voider1