StarLabsLtd/firmware #9: LabTop Mk IV: Power on when opening lid

cassidyjames, 2020-11-24:
I'm not sure this is possible solely in firmware, or if it is possible with the LabTop Mk IV hardware. But it's listed as a recommendation for elementary OS-optimized hardware:

>If possible, a laptop should automatically turn on when the lid is opened—Chromebooks do this, and it reduces the time from getting out the laptop and getting to work.

If there are concerns regarding a customer not wanting/expecting this (especially for certain specific workflows), it could potentially be added as an option in the BIOS settings.

Again, this is assuming it's even possible with this hardware. :)

Sean-StarLabs, 2020-12-15:
Unfortunately, this would *eat* the battery as it would be constantly on, waiting for the lid to be opened. The constant drain would also reduce the life-span considerably.

To properly implement this, you'd need a hardware switch that triggered when opening the laptop.

closed 2020-12-15 by Sean-StarLabs

cassidyjames, 2020-12-15:
@Sean-StarLabs makes sense! Thanks for considering it. Perhaps a future Star LabTop revision, then. ;)

CommonLoon102, 2022-12-10:
I wonder how often it happens that people missing deadlines in work because they had to push a button after opening the lid...