StarLabsLtd/firmware #13: Add option to stop charging battery

Cwmwd24, 2020-12-17:
If the hardware supports it it'd be nice if the BIOS could be configured to stop the charger when the battery reaches some percentage, this can help extend the battery life and save power when left "on charge". I think this would also be useful when using a dock that also charges the LabTop in a semi-permanent configuration.

The setting could be selectable from a defined list of values (75%; 85% or 90% say) and should have some hysteresis such that charging only restarts after the level has dropped back below the selected level by, for example, 3%. Obviously it should also be possible to disable the whole feature too.

I've seen this feature on some Lenovo laptops and I see that Tuxedo do something similar as well.

Sean-StarLabs, 2020-12-18:
That sounds like an old Clevo feature called "FlexiCharge(r?)". 

Our EC's do this by default, they reserve the last 5% of the battery, depending on the last charge cycle. (In turn, create #11).

I'll look into adding an option to adjust this to less than 95%.

Cwmwd24, 2020-12-18:
That's fair enough, didn't spot that it was already a feature of the EC. I can stop plugging and unplugging now then :).

closed 2020-12-18 by Cwmwd24