StarLabsLtd/firmware #18: Unable to install updates on Arch/Manjaro

Sean-StarLabs, 2021-08-09:
We've had several reports of users not being able to update their firmware or switch branches, for the LabTop Mk IV on Arch and Manjaro.

The error can vary slightly depending kernel but the most common one is:
```
failed to open device for prepare: flash probe failed: no chip was found

```

The issue seems to be with the version of flashrom that's installed. We're currently looking into a solution and will post once one is available.

Sean-StarLabs, 2021-08-20:
It turned out that this issue affected most distributions. 

Packages are now available for Ubuntu 20.04+, elementary OS 6+, Zorin OS 16+, Linux Mint 20.2+ etc. You should be able to update to these versions with:
```
sudo apt update
sudo apt full-upgrade
```
If you see any issues after this, please run:
```
sudo add-apt-repository ppa:starlabs/ppa
sudo apt install fwupd libfwupd2 fwupd-signed libflashrom1
```

Manjaro packages will follow shortly!

pivic, 2021-08-31:
Hey @Sean-StarLabs, thanks for the updates! Any ETA for when Manjaro packages follow?

Sean-StarLabs, 2021-09-01:
@pivic Just chasing the Manjaro guys!

Sean-StarLabs, 2021-09-23:
`flashrom-starlabs` and `starlabs-fwpd` are now available.

closed 2021-09-23 by Sean-StarLabs

barbastrejo7, 2021-10-04:
I am still having this same problem

Sean-StarLabs, 2021-10-05:
@barbastrejo7 Can you send an email to support@starlabs.systems with details of the problem and we'll get that sorted for you :)

andreas-jonsson, 2022-06-02:
Hi!

I'm having the same issue with my StarLite Mk IV on Fedora 36.

```
$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.7
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.17.11-300.fc36.x86_64
compile   com.hughsie.libjcat           0.1.10
compile   org.freedesktop.fwupd         1.7.7
runtime   org.freedesktop.gusb          0.3.10
```