StarLabsLtd/firmware #20: [coreboot] StarBook Mk V - Can't boot certain kernels

Sean-StarLabs, 2021-11-30:
BGRT will flicker and then hang indefinitely.

* 5.11.0-41
* 5.15.0-1115

Current workarounds are to use an older kernel or AMI firmware.

We should have a fix by the end of the day.

Sean-StarLabs, 2021-11-30:
We have found a solution to this problem, and are in the process of uploading this to the LVFS. This does usually take a few days, so if you'd like to install this now, please follow the below instructions.

There are no safeguards with this method, so please connect the charger and wait until the battery is charged to at least 30% before proceeding. Open a terminal window and enter the two commands below:

```
wget https://github.com/StarLabsLtd/firmware/raw/master/StarBook/MkV/coreboot/7.4/7.4.rom
sudo flashrom -p internal -w 7.4.rom -i bios --ifd -N
```

Once this command completes, please shutdown down your StarBook.

When you turn it on, it will take around 3-4 minutes to turn back on.

Sean-StarLabs, 2021-12-01:
The update is now in the LVFS. You can find our guide on how to install updates from the LVFS [here](https://support.starlabs.systems/kb/updates/installing-updates-from-the-lvfs).

Please use `fwupd refresh --force` if you don't see the update as it has only just been pushed to stable.

cassidyjames, 2021-12-01:
@Sean-StarLabs hm, when updating from the LVFS, I get:

```
$ fwupdmgr update 
Devices with the latest available firmware version:
 • Star Drive PCIe SSD
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade StarBook from 7 to 7.4?                                              ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release includes the following features:                         ║
║                                                                              ║
║ • Adjusted DDT to stop flickering issue on certain kernels                   ║
║                                                                              ║
║ StarBook and all connected devices may not be usable while updating.         ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
Downloading…             [***************************************]
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Decompressing…           [***************************************]
Firmware version formats were different, device was 'number' and release is 'plain'
```

I am willing to do a manual update, but figure it might be helpful to test the LVFS update as well. :)

Sean-StarLabs, 2021-12-01:
@cassidyjames Very helpful :)

I think that's outdated fwupd/flashrom - did you pick the ones up from our PPA?

cassidyjames, 2021-12-01:
@Sean-StarLabs I am currently on Fedora 35 for testing (I had a USB drive sitting around, ha!), so, no.

```
]$ fwupdmgr --version
client version:	1.7.2
compile-time dependency versions
	gusb:	0.3.8

daemon version:	1.7.2
```

```
]$ fwupdmgr --version
client version:	1.7.2
compile-time dependency versions
	gusb:	0.3.8

daemon version:	1.7.2
```

I can image back to an elementary OS or Ubuntu install and add the PPA if that's easier, though.

Sean-StarLabs, 2021-12-01:
@cassidyjames Ha!

The patches for TGL aren't merged into flashrom yet, so you can either build flashrom from our repo or switch, both will work

cassidyjames, 2021-12-02:
@Sean-StarLabs just to not leave you hanging, I reinstalled (an older ISO of) elementary OS, added the PPA, updated `fwupd`, and got the firmware update just fine. After that, I ran all the regular updates (including kernel) and everything seems to be working! Thanks!

jkibilds, 2021-12-15:
Hi

I have encountered both the hanging on flickering screen and the HDMI issue that should be fixed by this firmware update, according to [this Twitter post](https://twitter.com/starlabsltd/status/1465973554340458500). I have Pop!_OS 21.04 installed and its PPAs provided an older version of `fwupd` that seemed to not 'see' all of my hardware and a firmware update was impossible. I tried to update using `flashrom` but that didn't work. Then I managed to install `fwupd` 1.7.1 from StarLabs PPA and it seems like it is properly detecting my hardware and is able to retrieve update information from LVFS. However, it reports that my StarBook is already running firmware version 7.4 and that there are no updates to perform. But my laptop had already been shipped before this update was released so I see no way how it could actually be running the most recent firmware version (besides, I also observed the mentioned bugs).

Could it be that there is some bug in firmware version detection?

If that is not the case, would a reinstall of version 7.4 help? Anyhow, I would like to resolve the HDMI issue on my system.

Sean-StarLabs, 2021-12-15:
Could be, if you run `sudo dmidecode | grep Version | sed '1q;d'`, what version does it show?

jkibilds, 2021-12-15:
It gives me `Version: 7.4`

Sean-StarLabs, 2021-12-15:
How strange. Would you be able to send an email to support@starlabs.systems so I can check the build report? If possible, including a video of the behaviour that you're seeing would be very helpful.

jkibilds, 2021-12-17:
Ok, forgive my stupidity. :man_facepalming: I retried connecting an HDMI cable and it turned out that on the first attempt I had not plugged it in completely due to the socket being very tight. The HDMI works fine now. And I guess I had actually updated to firmware 7.4 on my first attempt with `flashrom`. During flashing, there were a few messages that led me to believe that the upgrade was not successful:
```
Enabling flash write... Warning: Setting BIOS Control at 0xdc from 0x8b to 0x89 failed.
```
and
```
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME.
```
(but Intel ME was already disabled in my system)

I'm sorry for bothering you with a false issue

Sean-StarLabs, 2022-01-06:
Solved with 6ca6cd17f73e998a8408cadea6fcd72c8fd2af80

closed 2022-01-06 by Sean-StarLabs