StarLabsLtd/firmware #21: Touchpad causing high IRQ on Starbook MKV

stan-janssen, 2021-12-14:
My new StarBook MKV running elementaryOS 6.0 is showing high IRQ on `irq/118-STAR000` (5 - 8% CPU usage):

```
$ ps aux | grep irq
root          12  0.0  0.0      0     0 ?        S    09:35   0:00 [ksoftirqd/0]
root          20  0.0  0.0      0     0 ?        S    09:35   0:00 [ksoftirqd/1]
root          26  0.0  0.0      0     0 ?        S    09:35   0:00 [ksoftirqd/2]
root          32  0.0  0.0      0     0 ?        S    09:35   0:00 [ksoftirqd/3]
root          38  0.0  0.0      0     0 ?        S    09:35   0:00 [ksoftirqd/4]
root          44  0.0  0.0      0     0 ?        S    09:35   0:00 [ksoftirqd/5]
root          50  0.0  0.0      0     0 ?        S    09:35   0:00 [ksoftirqd/6]
root          56  0.0  0.0      0     0 ?        S    09:35   0:00 [ksoftirqd/7]
root         133  0.0  0.0      0     0 ?        S    09:35   0:00 [irq/123-pciehp]
root         140  0.0  0.0      0     0 ?        I<   09:35   0:00 [vfio-irqfd-clea]

root         227  5.3  0.0      0     0 ?        D    09:35  14:40 [irq/118-STAR000]     <-------------------------

root         545  0.0  0.0      0     0 ?        S    09:35   0:02 [irq/153-iwlwifi]
root         560  0.0  0.0      0     0 ?        S    09:35   0:00 [irq/154-iwlwifi]
root         561  0.0  0.0      0     0 ?        S    09:35   0:00 [irq/155-iwlwifi]
root         562  0.0  0.0      0     0 ?        S    09:35   0:00 [irq/156-iwlwifi]
root         563  0.0  0.0      0     0 ?        S    09:35   0:00 [irq/157-iwlwifi]
root         564  0.0  0.0      0     0 ?        S    09:35   0:00 [irq/158-iwlwifi]
root         565  0.0  0.0      0     0 ?        S    09:35   0:00 [irq/159-iwlwifi]
root         566  0.0  0.0      0     0 ?        S    09:35   0:00 [irq/160-iwlwifi]
root         567  0.0  0.0      0     0 ?        S    09:35   0:00 [irq/161-iwlwifi]
root         572  0.0  0.0      0     0 ?        S    09:35   0:00 [irq/162-iwlwifi]
root         592  0.0  0.0  81940  3656 ?        Ssl  09:35   0:00 /usr/sbin/irqbalance --foreground
stan       12236  0.0  0.0  46952  2548 pts/0    S+   14:09   0:00 grep --color=auto irq
```

```
$ cat /proc/interrupts
            CPU0       CPU1       CPU2       CPU3       CPU4       CPU5       CPU6       CPU7       
   0:          8          0          0          0          0          0          0          0  IR-IO-APIC    2-edge      timer
   1:       3992          0          0          0          0        490          0          0  IR-IO-APIC    1-edge      i8042
   8:          0          0          0          0          0          0          1          0  IR-IO-APIC    8-edge      rtc0
   9:      12005      23112          0          0          0          0          0          0  IR-IO-APIC    9-fasteoi   acpi
  12:          0          0          0          0          0          0          0          0  IR-IO-APIC   12-edge      i8042
  14:          0          0          0          0          0          0          0          0  IR-IO-APIC   14-fasteoi   INT34C5:00
  28:    1007899     226461   39653728          0   66335638          0          0          0  IR-IO-APIC   28-fasteoi   idma64.0, i2c_designware.0
  32:          0          0          0          0          0          0          0          0  IR-IO-APIC   32-fasteoi   idma64.1, i2c_designware.1
  34:          0          0          0          0          0          0          0          0  IR-IO-APIC   34-fasteoi   idma64.2

 118:    4890108          0          0          0          0          0    2774247          0  IR-IO-APIC  118-fasteoi   STAR0001:00

 120:          0          0          0          0          0          0          0          0  DMAR-MSI    1-edge      dmar1
 121:          0          0          0          0          0          0          0          0  DMAR-MSI    0-edge      dmar0
 122:          0          0          0          0          0          0          0          0  DMAR-MSI    2-edge      dmar2
 123:          0          0          0          0          0          0          0          0  IR-PCI-MSI 114688-edge      PCIe PME, pciehp
 124:          0          0          0          0          0          0          0          0  IR-PCI-MSI 475136-edge      PCIe PME
 125:          8          0        106          0          0          0          6          2  IR-PCI-MSI 217088-edge      thunderbolt
 126:         25          5          0        107          0          0          0          0  IR-PCI-MSI 217089-edge      thunderbolt
 141:          0          0          0          0          0          0          0          0  IR-PCI-MSI 212992-edge      xhci_hcd
 142:      25267         49          0       1132       6413      11325          0          0  IR-PCI-MSI 327680-edge      xhci_hcd
 143:          0          0          0         20          0          0          0          0  IR-PCI-MSI 23068672-edge      nvme0q0
 144:          0          0        624          0          0          0          0          0  IR-PCI-MSI 514048-edge      snd_hda_intel:card0
 145:        669          0          0          0          0          0          0          0  IR-PCI-MSI 23068673-edge      nvme0q1
 146:          0        415          0          0          0          0          0          0  IR-PCI-MSI 23068674-edge      nvme0q2
 147:          0          0        642          0          0          0          0          0  IR-PCI-MSI 23068675-edge      nvme0q3
 148:          0          0          0        902          0          0          0          0  IR-PCI-MSI 23068676-edge      nvme0q4
 149:          0          0          0          0        746          0          0          0  IR-PCI-MSI 23068677-edge      nvme0q5
 150:          0          0          0          0          0        374          0          0  IR-PCI-MSI 23068678-edge      nvme0q6
 151:          0          0          0          0          0          0        996          0  IR-PCI-MSI 23068679-edge      nvme0q7
 152:      10442      12316          0     226854       2354          0     396039        466  IR-PCI-MSI 32768-edge      i915
 153:       1100       2375        268      23619       3814      15224      13220       7153  IR-PCI-MSI 333824-edge      iwlwifi: default queue
 154:        137         14        105          0       1265        755          2         46  IR-PCI-MSI 333825-edge      iwlwifi: queue 1
 155:         42         11         23         73        164        421        918          0  IR-PCI-MSI 333826-edge      iwlwifi: queue 2
 156:         23        257        185         82        224        181          6        528  IR-PCI-MSI 333827-edge      iwlwifi: queue 3
 157:         59          1         16          4       1563        700         40         28  IR-PCI-MSI 333828-edge      iwlwifi: queue 4
[... 19 lines cut ...]
 MCP:         48         48         48         48         48         48         48         48   Machine check polls
 ERR:          3
 MIS:          0
 PIN:          0          0          0          0          0          0          0          0   Posted-interrupt notification event
 NPI:          0          0          0          0          0          0          0          0   Nested posted-interrupt event
 PIW:          0          0          0          0          0          0          0          0   Posted-interrupt wakeup event
```

From dmesg I gather that this might be related to the touchpad:

```
$ dmesg | grep -i star000
[    1.536102] i2c_hid i2c-STAR0001:00: supply vdd not found, using dummy regulator
[    1.536124] i2c_hid i2c-STAR0001:00: supply vddl not found, using dummy regulator
[    1.708024] input: STAR0001:00 093A:0255 Mouse as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-STAR0001:00/0018:093A:0255.0001/input/input4
[    1.708080] input: STAR0001:00 093A:0255 Touchpad as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-STAR0001:00/0018:093A:0255.0001/input/input5
[    1.708109] hid-generic 0018:093A:0255.0001: input,hidraw0: I2C HID v1.00 Mouse [STAR0001:00 093A:0255] on i2c-STAR0001:00
[    3.361606] input: STAR0001:00 093A:0255 Mouse as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-STAR0001:00/0018:093A:0255.0001/input/input23
[    3.361742] input: STAR0001:00 093A:0255 Touchpad as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-STAR0001:00/0018:093A:0255.0001/input/input24
[    3.361858] hid-multitouch 0018:093A:0255.0001: input,hidraw0: I2C HID v1.00 Mouse [STAR0001:00 093A:0255] on i2c-STAR0001:00
```

Perhaps this is normal behavior; the trackpad works great otherwise; not sure if this is something I should be worried about regarding battery life.

What do you think?

Sean-StarLabs, 2021-12-14:
Can I ask which firmware are you using?

stan-janssen, 2021-12-14:
I believe I'm using 7.4 at the moment:

[image]

I wanted to update to 7.5 but it says cannot update while not on AC power. I'm currently charging it, will try again when battery reaches > 50%.

Sean-StarLabs, 2021-12-16:
Thank you. That's fixed in `7.6` which is available in the Testing remote. I'll get it pushed to stable ASAP
0a263a61ea399f4533602b6f805a7e2d8ae598d1

closed 2021-12-16 by Sean-StarLabs