StarLabsLtd/firmware #24: switch-branch: No supported devices

athyla, 2021-12-22:
Hello! 

I'm not sure if this is the right place to post this issue.
I've been trying to install coreboot onto a StarBook Mk V, following the official guide.
For some reason, I can't switch branch from AMI to coreboot.
```
user@starbook:~$ fwupdmgr switch-branch 
No supported devices
```
I am using Linux Mint 20.2 Cinnamon Edge, with the StarLabs PPA added.
I also tried the same steps using a live Ubuntu 21.10, the result is the same.

Here is the output of the commands I used:
```
user@starbook:~$ fwupdmgr --version 
client version:	1.7.4
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.7.4
```
```
user@starbook:~$ fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Successfully downloaded new metadata: 2 local devices supported
```
```
user@starbook:~$ fwupdmgr switch-branch 
No supported devices
```
```
user@starbook:~$ fwupdmgr update 
Devices with no available firmware updates: 
 • Star Drive PCIe SSD
 • StarBook
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • SuperIO FuSuperioIt89Device
 • System Firmware
```

Any help is appreciated!
Thanks!

Sean-StarLabs, 2021-12-22:
Hey

Can you let me know the output of these two commands please:
`upower -i /org/freedesktop/UPower/devices/battery_BAT0 | grep 'state\|percentage'`
`fwupdmgr security --force | grep BIOS`

athyla, 2021-12-22:
Here you go:
```
user@starbook:~$ upower -i /org/freedesktop/UPower/devices/battery_BAT0 | grep 'state\|percentage'
    state:               charging
    percentage:          97%
user@starbook:~$  fwupdmgr security --force | grep BIOS
✘ SPI BIOS region:               Unlocked
```

smiggiddy, 2021-12-22:
I'm having this issue as well:

    ~  upower -i /org/freedesktop/UPower/devices/battery_BAT0 | grep 'state\|percentage'
    state:               charging
    percentage:          88%
    ~  fwupdmgr security --force | grep BIOS                                      ✔ 
✘ SPI BIOS region:               Unlocked

Brudibert, 2021-12-26:
Same Issue here.
`fwupdmgr --version                                                                                                                                                                                  ✔ 
client version:	1.7.3-40-g935e92d0
compile-time dependency versions
	gusb:	0.3.8

daemon version:	1.7.3-40-g935e92d0`
`    ~   upower -i /org/freedesktop/UPower/devices/battery_BAT0 | grep 'state\|percentage'                                                                                                              INT ✘ 
    state:               discharging
    percentage:          98%
    ~   fwupdmgr security --force | grep BIOS                                                                                                                                                              ✔ 
✘ SPI BIOS Descriptor:           Invalid
✘ SPI BIOS region:               Unlocked
  2021-12-24 14:33:01:  ✘ SPI BIOS Descriptor appeared: Invalid`

athyla, 2022-01-05:
Hello again!

Since last time I have installed Fedora 35 Workstation. It has fwupdmgr version 1.7.3. The results are the same as reported above.

Sean-StarLabs, 2022-01-06:
If you re-run `fwupdmgr refresh --force`, it should now find a valid coreboot release.

@athyla For Fedora, you'll need to build flashrom from source as the packaged version isn't new enough to support the 11th Generation processors.

Brudibert, 2022-01-06:
On Linux Mint 20.2 I was able to update the Embedded Controller to 1.03 but it still says that Coreboot 7.4 is the latest release.

athyla, 2022-01-06:
> If you re-run `fwupdmgr refresh --force`, it should now find a valid coreboot release.
> 
> @athyla For Fedora, you'll need to build flashrom from source as the packaged version isn't new enough to support the 11th Generation processors.

Hey Sean, thanks for your reply!

On Fedora I have flashrom version 1.2-7 and fwupdmgr version 1.7.3.
I have the same result: No supported devices.

I have tried the process again using two live distros: Manjaro Gnome edition and Linux Mint Edge, both the latest version. 
On Manjaro I have installed the starlabs version for fwupd and flashrom, on Mint I have used the ppa.
Unfortunately I wasn't able to switch-branch, the message is the same for both: No supported devices.

Sean-StarLabs, 2022-01-07:
Could you let me know the output of `fwupdmgr get-releases --allow-branch-switch --force` for the `StarBook` device?

athyla, 2022-01-07:
```
StarBook
│
├─StarBook MkV:
│     New version:        2
│     Remote ID:          lvfs
│     Release ID:         10607
│     Summary:            AMI firmware for StarBook MkV
│     License:            Proprietary
│     Size:               11.5 MB
│     Created:            2021-11-08
│     Urgency:            Medium
│     Vendor:             Star Labs
│     Flags:              is-downgrade|blocked-version
│     Description:        
│     This stable release includes the following features:
│     
│     • Updated Verb Table
│     • New option to set the maximum battery charge level
│     • Fixed ACPI error
│     • Remember Fn Lock state between reboot
│   
└─StarBook MkV:
      New version:        1
      Remote ID:          lvfs
      Release ID:         10124
      Summary:            AMI firmware for StarBook MkV
      License:            Proprietary
      Size:               11.5 MB
      Created:            2021-09-27
      Urgency:            Medium
      Vendor:             Star Labs
      Flags:              is-downgrade|blocked-version
      Description:        
      This stable release includes the following features:
      
      • Initial release firmware for the StarBook Mk V
      • American Megatrends Inc:		5.29
      • (CS)Management Engine:		15.0.30.1776
      • Microcode:						0x88
      • Firmware Support Package:		A.0.51.31

```

Sean-StarLabs, 2022-01-07:
Thanks - so the issue is that the device isn't appearing (`System Firmware` is only used for AMI capsule updates).

Was that on Fedora? It might be that it doesn't have flashrom support, could you run `fwupdtool get-devices --plugins flashrom` and let me know the output?

athyla, 2022-01-07:
Yes, I'm on Fedora 35 now, but I can try with a live Linux Mint if needed, or some other live medium, just let me know.

```
user@starbook:~$ fwupdmgr get-devices --plugins flashrom
Failed to parse arguments: Unknown option --plugins
```
```
user@starbook:~$ fwupdmgr get-plugins 
acpi_dmar:
  Flags:                • Enabled

acpi_facp:
  Flags:                • Enabled

acpi_phat:
  Flags:                • Disabled

amt:
  Flags:                • Disabled

analogix:
  Flags:                • Enabled

ata:
  Flags:                • Enabled

bcm57xx:
  Flags:                • Enabled

bios:
  Flags:                • Disabled

ccgx:
  Flags:                • Enabled

colorhug:
  Flags:                • Enabled

cpu:
  Flags:                • Enabled

cros_ec:
  Flags:                • Enabled

dell_dock:
  Flags:                • Enabled

dell_esrt:
  Flags:                • Disabled
                        • Required hardware was not found

dfu:
  Flags:                • Enabled

dfu_csr:
  Flags:                • Enabled

ebitdo:
  Flags:                • Enabled

elanfp:
  Flags:                • Enabled

elantp:
  Flags:                • Enabled

emmc:
  Flags:                • Enabled

ep963x:
  Flags:                • Enabled

fastboot:
  Flags:                • Enabled

flashrom:
  Flags:                • Enabled

fresco_pd:
  Flags:                • Enabled

goodixmoc:
  Flags:                • Enabled

hailuck:
  Flags:                • Enabled

iommu:
  Flags:                • Enabled

jabra:
  Flags:                • Enabled

lenovo_thinklmi:
  Flags:                • Disabled
                        • Required hardware was not found

linux_lockdown:
  Flags:                • Enabled

linux_sleep:
  Flags:                • Enabled

linux_swap:
  Flags:                • Enabled

linux_tainted:
  Flags:                • Enabled

logind:
  Flags:                • Enabled

logitech_bulkcontroller:
  Flags:                • Enabled

logitech_hidpp:
  Flags:                • Enabled

modem_manager:
  Flags:                • Enabled

mtd:
  Flags:                • Enabled

nitrokey:
  Flags:                • Enabled

nordic_hid:
  Flags:                • Enabled

nvme:
  Flags:                • Enabled

optionrom:
  Flags:                • Enabled

parade_lspcon:
  Flags:                • Enabled

pci_bcr:
  Flags:                • Enabled

pci_mei:
  Flags:                • Enabled

pixart_rf:
  Flags:                • Enabled

realtek_mst:
  Flags:                • Enabled

redfish:
  Flags:                • Disabled

rts54hid:
  Flags:                • Enabled

rts54hub:
  Flags:                • Enabled

steelseries:
  Flags:                • Enabled

superio:
  Flags:                • Enabled

synaptics_cape:
  Flags:                • Enabled

synaptics_cxaudio:
  Flags:                • Enabled

synaptics_mst:
  Flags:                • Enabled

synaptics_prometheus:
  Flags:                • Enabled

synaptics_rmi:
  Flags:                • Enabled

system76_launch:
  Flags:                • Enabled

thelio_io:
  Flags:                • Enabled

thunderbolt:
  Flags:                • Enabled

tpm:
  Flags:                • Enabled

uefi_dbx:
  Flags:                • Enabled

uefi_pk:
  Flags:                • Enabled

upower:
  Flags:                • Enabled

vli:
  Flags:                • Enabled

wacom_raw:
  Flags:                • Enabled

wacom_usb:
  Flags:                • Enabled

msr:
  Flags:                • Enabled

uefi_capsule:
  Flags:                • Enabled

dell:
  Flags:                • Disabled
                        • Required hardware was not found

uefi_recovery:

```

Sean-StarLabs, 2022-01-07:
Thanks, and what does `sudo flashrom -p internal` show?

athyla, 2022-01-07:
```
user@starbook:~$ sudo flashrom -p internal
flashrom v1.2 on Linux 5.15.12-200.fc35.x86_64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
No DMI table found.
WARNING: No chipset found. Flash detection will most likely fail.
========================================================================
You may be running flashrom on an unknown laptop. We could not
detect this for sure because your vendor has not set up the SMBIOS
tables correctly. Some internal buses have been disabled for
safety reasons. You can enforce using all buses by adding
  -p internal:laptop=this_is_not_a_laptop
to the command line, but please read the following warning if you
are not sure.

Laptops, notebooks and netbooks are difficult to support and we
recommend to use the vendor flashing utility. The embedded controller
(EC) in these machines often interacts badly with flashing.
See the manpage and https://flashrom.org/Laptops for details.

If flash is shared with the EC, erase is guaranteed to brick your laptop
and write may brick your laptop.
Read and probe may irritate your EC and cause fan failure, backlight
failure and sudden poweroff.
You have been warned.
========================================================================
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.
```

Sean-StarLabs, 2022-01-07:
Thanks. That version of flashrom isn't new enough to support the 11th Generation processors; if you build the latest one from source, that should do it :)

smiggiddy, 2022-01-07:
Hey Sean, 

is that the issue with manjaro too?

    ~  sudo flashrom -p internal                                                  ✔ 
[sudo] password for smig: 
flashrom v1.2 on Linux 5.15.12-1-MANJARO (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Cannot open file stream for /dev/mtd0
coreboot table found at 0x76b33000.
Found chipset "Intel Tiger Lake U Premium".
Enabling flash write... Warning: Setting BIOS Control at 0xdc from 0x8b to 0x89 failed.
New value is 0x8b.
Error accessing SPIBAR, 0x1000 bytes at 0x00000000fe010000
/dev/mem mmap failed: Operation not permitted
FAILED!
FATAL ERROR!
Error: Programmer initialization failed.

Sean-StarLabs, 2022-01-07:
No, that just needs the `iomem=relaxed` kernel parameter. The two below commands will do this for you:

```
sudo sed -i 's/quiet/quiet iomem=relaxed/g' /etc/default/grub
sudo update-grub
```

athyla, 2022-01-07:
Alright, so I removed the default flashrom package and built the latest version from github.
Version is: `flashrom v1.2-566-gb5dc741 on Linux 5.15.12-200.fc35.x86_64 (x86_64)`

I did the refresh and then the switch-branch and it started to switch to coreboot, showing a warning and asking for confirmation. I have chosen No to bail out of the switch. I wanted to check the same on a live Linux Mint.

I checked the whole process on a live mint, updated the system, built flashrom from git, but it still did not work (No supported device).

I came back to Fedora to do the switch, but this time all the way through and to my surprise I get the initial error: No supported devices. :scream: 
I have rebuilt flashrom again, refreshed with force, same issue. This is very strange, is there a way to clean/reset fwupd and related things? :disappointed:

Sean-StarLabs, 2022-01-07:
fwupd doesn't have much that you can reset, the only thing that upsets it is using sudo.

Can you send me the output of `fwupdmgr get-devices` from Fedora, whilst it's saying no supported devices?

athyla, 2022-01-07:
```
user@starbook:~$ fwupdmgr get-devices 
StarBook
│
├─Star Drive PCIe SSD:
│     Device ID:          9aa3cbcc566d62d70c8e5aea851719da50f07ca8
│     Summary:            NVM Express solid state drive
│     Current version:    EIFM21.2
│     Vendor:             Phison Electronics Corporation (NVME:0x1987)
│     GUIDs:              c9465066-32bb-50c5-8338-6a750cae2a44
│                         b1fcba59-6797-58bd-af77-5ccb9be2493c
│                         fcc92585-8e8e-52a3-90cc-6c022b7f3843
│                         59639c7d-1b5d-5b7f-82da-9fd6b369a461
│                         12d61493-5ac4-57f7-a2c5-96878d115455
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs shutdown after installation
│                         • Device is usable for the duration of the update
│   
├─SuperIO FuSuperioIt89Device:
│     Device ID:          3ecf3fbe07ed8933226b294c8c19f4f9c8ebd71a
│     Summary:            Embedded controller
│     Current version:    01.01
│     Vendor:             DMI:Star Labs
│     Install Duration:   20 seconds
│     GUIDs:              049060b5-5010-5ca0-8002-07923951d15e
│                         d83a4cd0-fce6-58a8-93c7-bcee89b41dcf
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Update requires a reboot
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    33554432
│ │   Minimum Version:    33554432
│ │   Vendor:             Star Labs (DMI:American Megatrends International, LLC.)
│ │   Update State:       Success
│ │   GUIDs:              e27d9e98-cbb2-4975-a8f8-261de2adde05
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       51c09ad0-818b-5dc7-9a0f-12253ce64670
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  77
│       Minimum Version:  77
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Only version upgrades are allowed
│     
├─UEFI Device Firmware:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    1
│     Vendor:             DMI:American Megatrends International, LLC.
│     Update State:       Success
│     GUIDs:              eed54281-1c11-4358-bf5a-f64995fbf11b
│                         2b549001-ce65-5243-9056-7cb5214b9302
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─USB4 host controller:
      Device ID:          9f0039a80056f8e5d3da1adb71e273f11e4b9ecc
      Summary:            Unmatched performance for high-speed I/O
      Update Error:       No vendor ID set
      GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c
      Device Flags:       • Internal device
                          • System requires external power source

```

Sean-StarLabs, 2022-01-07:
How very stange. What does `sudo flashrom -p internal` say now?

athyla, 2022-01-07:
```
user@starbook:~$ sudo flashrom -p internal
flashrom v1.2-566-gb5dc741 on Linux 5.15.12-200.fc35.x86_64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
No DMI table found.
Found chipset "Intel Tiger Lake U Premium".
Enabling flash write... SPI Configuration is locked down.
FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG2: Management Engine region (0x00001000-0x004fffff) is read-only.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are write protected. For write operations,
you should use a flash layout and include only writable regions. See
manpage for more details.
Enabling hardware sequencing because some important opcode is locked.
OK.
Found Programmer flash chip "Opaque flash chip" (16384 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
No operations were specified.
```

Sean-StarLabs, 2022-01-07:
Thanks, last one: `sudo fwupdtool get-devices --verbose --plugins flashrom --allow-branch-switch`

athyla, 2022-01-07:
:sweat_smile: I just removed fwupd using dnf with all its dependencies and then installed it again and now I don't get the `No supported devices` message:
```
user@starbook:~$ fwupdmgr switch-branch 
Selected device: StarBook
╔══════════════════════════════════════════════════════════════════════════════╗
║ Switch branch from default to coreboot?                                      ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release includes the following features:                         ║
║                                                                              ║
║ • Update GPIOs                                                               ║
║ • Update CNVi GPIOs                                                          ║
║ • Update trackpad GPIO                                                       ║
║ • Enable TPM_IRQ GPIO                                                        ║
║ • Don't configure ESPI GPIOs                                                 ║
║ • Disconnect unused GPIO's                                                   ║
║ • Add comments for GPIOs                                                     ║
║ • Remove display from devicetree                                             ║
║ • Apply EC settings when suspending                                          ║
║ • Unify EC and CMOS names                                                    ║
║ • Adjust KBL configuration                                                   ║
║ • Enable I2C4                                                                ║
║ • Convert EC_GPE_SCI to Kconfig                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝

Do you understand the consequences of changing the firmware branch? [y|N]: 
Declined branch switch

```
I answered No again to bail out, ran the fwupd commands again, rebooted, ran the commands again and it is still offering to do the switch, so it seems to be okay now.

Before I do the switch all the way, here is the output of the `fwupdtool` command:

```
user@starbook:~$ sudo fwupdtool get-devices --verbose --plugins flashrom --allow-branch-switch
14:11:07:0793 FuDebug              Verbose debugging enabled (on console 1)
14:11:07:0794 FuMain               locked /run/lock/fwupdtool
14:11:07:0800 FuConfig             loading config values from /etc/fwupd/daemon.conf
14:11:07:0800 FuConfig             using autodetected max archive size 18.4 EB
14:11:07:0801 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
14:11:07:0801 FuRemoteList         loading /usr/share/fwupd/metainfo
14:11:07:0850 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
14:11:07:0850 XbSilo               file: b42e572c-f5e2-9544-989f-dc9ff77a2d8c, current:b42e572c-f5e2-9544-989f-dc9ff77a2d8c, cached: (null)
14:11:07:0850 XbSilo               loading silo with file contents
14:11:07:0850 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
14:11:07:0850 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
14:11:07:0851 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
14:11:07:0851 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
14:11:07:0851 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
14:11:07:0851 FuRemoteList         skipping invalid file /var/lib/fwupd/remotes.d/lvfs
14:11:07:0851 FuRemoteList         ordering lvfs-testing=lvfs+1
14:11:07:0851 FuRemoteList         ignoring unfound remote fwupd
14:11:07:0851 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
14:11:07:0851 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
14:11:07:0851 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
14:11:07:0851 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
14:11:07:0851 Jcat                 loaded 1 certificates
14:11:07:0851 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
14:11:07:0852 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
14:11:07:0852 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
14:11:07:0852 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
14:11:07:0852 Jcat                 loaded 1 certificates
14:11:07:0852 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
14:11:07:0852 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
14:11:07:0852 FuEngine             client certificate now exists: nothing to do
14:11:07:0852 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
14:11:07:0852 FuHistory            got schema version of 7
14:11:07:0852 FuSmbios             trying to read /sys/class/dmi/id
14:11:07:0853 FuHwids              loading HwId overrides from /etc/fwupd/daemon.conf
14:11:07:0853 FuHwids              not loading HwId overrides from /var/lib/fwupd/daemon.conf
14:11:07:0853 FuHwids              smbios property Manufacturer=Star Labs
[... 69 lines cut ...]
14:11:07:0894 FuEngine             requirement 1.7.0 ge 1.7.3 -> org.freedesktop.fwupd passed
[... cut ...]
  Guid:                 612d0af2-6b4b-5d1a-bd63-433022fdaefe ← Star Labs&B5&StarBook&B5&American Megatrends International, LLC.
  Plugin:               flashrom
  Protocol:             org.flashrom
  Flags:                internal|updatable|require-ac|supported|registered|needs-shutdown|has-multiple-branches
  Vendor:               Star Labs
  VendorId:             DMI:American Megatrends International, LLC.
  Version:              2.0.00
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2022-01-07
```
I'm not sure what was it, I checked for system updates, there were none.

athyla, 2022-01-07:
Alright, so I went ahead to do the switch, but it wasn't able to finish the process, here is the output:
```
user@starbook:~$ fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 3 local devices supported
```
```
user@starbook:~$ fwupdmgr switch-branch 
Selected device: StarBook
╔══════════════════════════════════════════════════════════════════════════════╗
║ Switch branch from default to coreboot?                                      ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release includes the following features:                         ║
║                                                                              ║
║ • Update GPIOs                                                               ║
║ • Update CNVi GPIOs                                                          ║
║ • Update trackpad GPIO                                                       ║
║ • Enable TPM_IRQ GPIO                                                        ║
║ • Don't configure ESPI GPIOs                                                 ║
║ • Disconnect unused GPIO's                                                   ║
║ • Add comments for GPIOs                                                     ║
║ • Remove display from devicetree                                             ║
║ • Apply EC settings when suspending                                          ║
║ • Unify EC and CMOS names                                                    ║
║ • Adjust KBL configuration                                                   ║
║ • Enable I2C4                                                                ║
║ • Convert EC_GPE_SCI to Kconfig                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝

Do you understand the consequences of changing the firmware branch? [y|N]: y
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade StarBook from 2.0.00 to 7.91?                                        ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release includes the following features:                         ║
║                                                                              ║
║ • Update GPIOs                                                               ║
║ • Update CNVi GPIOs                                                          ║
║ • Update trackpad GPIO                                                       ║
║ • Enable TPM_IRQ GPIO                                                        ║
║ • Don't configure ESPI GPIOs                                                 ║
║ • Disconnect unused GPIO's                                                   ║
║ • Add comments for GPIOs                                                     ║
║ • Remove display from devicetree                                             ║
║ • Apply EC settings when suspending                                          ║
║ • Unify EC and CMOS names                                                    ║
║ • Adjust KBL configuration                                                   ║
║ • Enable I2C4                                                                ║
║ • Convert EC_GPE_SCI to Kconfig                                              ║
║                                                                              ║
║ StarBook and all connected devices may not be usable while updating.         ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
Downloading…             [***************************************]
Downloading…             [**                                     ] Less than one Downloading…             [**                                     ] Less than one Downloading…             [***                                    ] Less than one Downloading…             [***                                    ] Less than one Downloading…             [***                                    ] Less than one Downloading…             [****                                   ] Less than one Downloading…             [****                                   ] Less than one Downloading…             [*****                                  ] Less than one Downloading…             [*****                                  ] Less than one Downloading…             [***************************************]
Decompressing…           [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
failed to open device for prepare: flash probe failed: no chip was found
```

kciredor, 2022-01-07:
That is - if I understand correctly - due to `sudo flashrom -p internal` outputting 'No chipset found', right?

What I don't get myself is how flashrom 1.2 is almost 2 years old and there's still differences in 1.2 versions across distro's?

athyla, 2022-01-07:
> That is - if I understand correctly - due to `sudo flashrom -p internal` outputting 'No chipset found', right?
> 
> What I don't get myself is how flashrom 1.2 is almost 2 years old and there's still differences in 1.2 versions across distro's?

Hey @kciredor!
Since I have built the latest version of `flashrom` from git, it finds the chipset:
```
user@starbook:~$ sudo flashrom -p internal
flashrom v1.2-566-gb5dc741 on Linux 5.15.12-200.fc35.x86_64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
No DMI table found.
Found chipset "Intel Tiger Lake U Premium".
Enabling flash write... SPI Configuration is locked down.
FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG2: Management Engine region (0x00001000-0x004fffff) is read-only.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are write protected. For write operations,
you should use a flash layout and include only writable regions. See
manpage for more details.
Enabling hardware sequencing because some important opcode is locked.
OK.
Found Programmer flash chip "Opaque flash chip" (16384 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
No operations were specified.
```

Sean-StarLabs, 2022-01-07:
@kciredor Yes, `no chip was found` is basically `flashrom is too old`.

flashrom hasn't tagged a release in about 2 years, the all the distros are just intermittently building versions from master to try and solve problems like this. I'll give them another nudge, see if they're planning to do it anytime soon...

@athyla Let me set one up with Fedora to try and figure out what's going on...

athyla, 2022-01-07:
Thanks, @Sean-StarLabs! 
Btw, what OS are you normally using with the StarLabs HW?

Sean-StarLabs, 2022-01-07:
Mostly Ubuntu, but I spend a lot of time with elementary and Manjaro

kciredor, 2022-01-07:
Would it be safe to use an Ubuntu live CD for firmware updates for me?

(Because NixOS is quite a battle to get fwupd/flashrom updated/packaged)

Sean-StarLabs, 2022-01-07:
Yup, 20.04 or 21.04, just run:
```
sudo add-apt-repository ppa:starlabs/ppa
sudo add-apt-repository universe
sudo apt update
sudo apt full-upgrade -y
sudo apt install fwupd
sudo apt -f install
fwupdmgr refresh --force
fwupdmgr switch-branch
```

kciredor, 2022-01-07:
Managed to build custom flashrom on NixOS, would you say this looks about right / safe to use?

```
sudo flashrom -p internal                                                               ☸ pinniped-azure-review-acc
flashrom 1.2-custom on Linux 5.10.89 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Cannot open file stream for /dev/mtd0
coreboot table found at 0x76b33000.
Found chipset "Intel Tiger Lake U Premium".
Enabling flash write... Warning: Setting BIOS Control at 0xdc from 0x8b to 0x89 failed.
New value is 0x8b.
SPI Configuration is locked down.
FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG2: Management Engine region (0x00001000-0x004fffff) is read-only.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are write protected. For write operations,
you should use a flash layout and include only writable regions. See
manpage for more details.
OK.
Found Programmer flash chip "Opaque flash chip" (16384 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
No operations were specified.
```

Sean-StarLabs, 2022-01-07:
Yes, looks good to me

kciredor, 2022-01-07:
This is what's used by fwupd to flash coreboot right? Meaning I could use it without fwupd and run it manually to flash coreboot upgrades for now?

Sean-StarLabs, 2022-01-07:
Yes, fwupd basically just runs a flashrom command. - `sudo flashrom -p internal -w coreboot.rom -i bios --ifd -N`

fwupd does also make sure it's done correctly i.e. power is connected, it'll shutdown after flashing and that it's the right file

smiggiddy, 2022-01-07:
What's the latest stable coreboot firmware?

kciredor, 2022-01-07:
Now that I'm thinking about it, can it be used to flash the NVME and other Chipset (I'm on 1.0.1 and there is 1.0.3 now) as well?

kciredor, 2022-01-07:
@smiggiddy 7.91 for StarBook Mk V -> https://github.com/StarLabsLtd/firmware

Sean-StarLabs, 2022-01-07:
No, both of those use other tools - flashing them manually is complicated and erroneous.

kciredor, 2022-01-07:
Thanks for everything @Sean-StarLabs. Just want to make sure one more thing: fwupd does not depend on anything else, to be able to flash those other things,  right? It is batteries included except for Coreboot flashing specifically where it needs flashrom?

Sean-StarLabs, 2022-01-07:
That's right :)

kciredor, 2022-01-07:
I've proceeded with an Ubuntu live cd following your instructions. Except for "fwupdmgr switch-branch" which I've replaced with "fwupdmgr update", because I was already on Coreboot (7.4) and trying to upgrade to 7.91. At the end of the updates it requests to shutdown the laptop which I did.

Then I turned it on again and now my laptop does not boot anymore saying:
```
Booting from 'NVMe: Star Drive PCIe SSD ' failed; verify it contains a 64-bit UEFI OS.

Press any key to continue booting...
```

After pressing enter I end up in UEFI interactive shell which I immediately 'exit' and then the UEFI options menu shows me FW: 7.91 01/04/2022 which means the upgrade has been completed I guess.

Looking through the boot manager items there is the StarBook NVMe and UEFI. The NVMe entry results in the boot error about a missing 64-bit UEFI OS.

I've now added my own boot entry pointing directly to the grub file in EFI on SSD and that allows me to boot my system again. Then I moved this up in the boot order so it's default now. Is this step in the process 'normal'?

Running fmupdmgr from NixOS quickly (not from live Ubuntu) shows me that the embedded controller is still at 1.01, even though the fwupd process showed me that it was upgraded to 1.03. What did I miss?

danth, 2022-01-08:
@kciredor NixOS adds a boot entry for GRUB when it is first installed. Updating Coreboot seems to reset the list of boot entries. I'm also running NixOS on the StarBook, and I've found that Systemd-Boot is picked up automatically even when the list is reset:

```nix
boot.loader = {
  efi.canTouchEfiVariables = true;
  systemd-boot.enable = true;
};
```

(You will need to remove GRUB from the EFI partition manually if you want to switch.)

As for the embedded controller update: NixOS refused to start up until I removed a symlink which fwupd had created at the filesystem root (had to be done from a live USB). After that, NixOS was booting again, but fwupd reported that the update had failed. I'm not sure exactly what the procedure for updating the embedded controller involves, so I can't say why this happens.

kciredor, 2022-01-08:
@danth Thanks for the information, makes sense. I'll probably switch to systemd-boot then.

I've run fwupdmgr update from NixOS now to retry upgrading the chipset from 1.01 to 1.03. It said it succeeded and so I rebooted. Then I ran into the same issue as you did so I had to use a live CD to remove the symlink from the root directory (/system-update pointing to /var/lib/fwupd).

I'm guessing something should be running to finalize the update upon boot, but instead it crashes NixOS.

Right now I can boot into NixOS again. My chipset firmware is still at 1.01. And... my trackpad is completely dead :-( It is working with a reboot into Ubuntu live cd though. Super strange.

athyla, 2022-01-08:
@kciredor I have tried out the chipset firmware update too, as it was 1.01 on my StarBook too. First I tried it from a live kubuntu and it went as you described, process finished with no error message, it offered to reboot, but after reboot the chipset version was still 1.01. I did this 2 times, same result.

Since then I installed kubuntu on an external drive, but I don't think this makes any difference, besides not needing to add the ppa and update the whole system after every reboot.
I tried the firmware update again, it went as before, no errors, but after reboot it was still the old version. Then when listing the devices I noticed, that I wasn't plugged in and updates cannot be done on battery. Strange that it let's you do the update without even a warning that it cannot be done on battery.

Anyways, I tried it again, this time plugged in, this time doint `fwupdmgr upgrade` instead of `fwupdmgr update` as I was doing before and after reboot I have had the latest version installed.
The help of fwupdmgr it says that `upgrade` is an alias to `update`, so I'm not sure what the problem with my first attempts.

kciredor, 2022-01-08:
Meanwhile reinstalled NixOS and my trackpad is still dead. I'm assuming something happened with the upgrade attempt to 1.03 which messed up my laptop.

athyla, 2022-01-08:
@Sean-StarLabs 
In the meantime I have tried out Kubuntu for a change (21.10), out of curiosity (I'm in a "let's try out a few more ditros" mode :grinning: ). I added the PPA and did the full update as described above. 
The `switch-branch` finished successfully this time. :partying_face: 
I'm not sure why it did not work on Mint.

The system booted normally. I played around a bit, I checked the coreboot menu options and `coreboot-configurator`. Everything works fine on Kubuntu.

Then I booted Fedora. After installing all the missing dependencies, building `coreboot-configurator` finished successfully, however the app does not run correctly, it opens and it complains about not finding `nvramtool`. 

I also tried switching back to AMI while on Fedora and it does the same as last time, the update fails with the `failed to open device for prepare: flash probe failed: no chip was found` message. So there is definitely a problem or something missing on Fedora.

Question: what is holding CTRL-ALT-DEL-POWER doing more exactly?

athyla, 2022-01-08:
@kciredor I thought the 5 minutes were only a one-time thing, on every boot it would be very annoying. I haven't experienced this at all, although I haven't tried NixOS, only: Fedora, OpenSuse, Linux Mint, Ubuntu, Kubuntu, Manjaro and EndeavorOS. Perhaps open a separate ticket with your problems, it might be clearer that way. Good luck!

grexe, 2022-01-08:
I have the same issue on Manjaro with Starlabs-fwupd, flashrom works (detects the flash chip and successfully flashes with the command posted above), but with fwupdmgr I get the "no devices detected" error.
However, with AMI BIOS (not with coreboot), `fwupdtool` sees 3 local devices and finds an update for the SuperIO device:
> Upgrade SuperIO FuSuperioIt89Device from 01.01 to 1.03? 

But after a successful update and reboot, it still proposes the same update again, so obviously it either has not really performed the operation, or it does not detect the version correctly.

I have previously flashed coreboot 7.11 because I thought it was newer than 7.91, but I was obviously wrong, so trying with 7.91 again now.

grexe, 2022-01-08:
so with coreboot 7.91 the SuperIO update worked, now correctly reported as latest firmware.
Still get the "no supported devices" error when trying to `sudo fwupdmgr switch-branch`

kciredor, 2022-01-09:
Update 1: the 'trackpad not working on boot, but after 5 minutes' was a configuration change I forgot about. Rookie mistake. Fixed it.

Update 2: I've given up on NixOS with regards to firmware upgrades and installed Ubuntu next to it. So I now have a dual boot setup, with Ubuntu as a dedicated firmware upgrade OS. And I've successfully updated the firmware from 1.01 to 1.03 that way.

So everything's good overhere ;-) Thanks everyone involved!

Sean-StarLabs, 2022-01-10:
> Question: what is holding CTRL-ALT-DEL-POWER doing more exactly?
Reset's the EC

> Is it normal for Coreboot not to boot my OS anymore without manual intervention after a firmware upgrade? And did I do the right thing?
No. coreboot doesn't have a mechanism to update, so when you install a new version with fwupd or flashrom, it's a "new install". That means it has to re-add boot entries, and for most OS's, these get re-added without an issue. From what's been said, it seems like coreboot is finding the systemd entry but not the grub one. If that's the case, it would need to be changed in NixOS.

> I'm guessing something should be running to finalize the update upon boot, but instead it crashes NixOS.
All firmware updates, apart from coreboot, are done on the next boot. Did the EC update not work with grub or systemd boot?

> it complains about not finding nvramtool.
You'll need to install nvramtool

@grexe Can you post the output of `sudo fwupdtool get-devices --verbose --plugins flashrom --allow-branch-switch` please?

athyla, 2022-01-10:
Thanks @Sean-StarLabs for all the help and patient answers :+1: 

The initially reported issue has been documented and solved, so from my point of view this ticket can be closed.
If you need me to test/assist with Fedora, let me know.

kciredor, 2022-01-10:
I agree, thanks @Sean-StarLabs. All good overhere now ;-)

closed 2022-01-11 by Sean-StarLabs

esad, 2023-03-03:
Hi, I am experiencing this issue on my Star Lite Mk III on Ubuntu 22.10. Here is it output of relevant commands:

```
me@lite:~$ fwupdmgr switch-branch
No supported devices
me@lite:~$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.0
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.8.10
runtime   org.kernel                    5.19.0-35-generic
me@lite:~$ fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 3 local devices supported
me@lite:~$ fwupdmgr switch-branch
No supported devices
me@lite:~$ fwupdmgr update
Devices with no available firmware updates: 
 • INTENSO SSD
 • Lite (bios)
Devices with the latest available firmware version:
 • SuperIO FuSuperioIt89Device
 • System Firmware
 • UEFI dbx
me@lite:~$ fwupdmgr security --force | grep BIOS
✘ SPI BIOS region:               Unlocked
me@lite:~$ fwupdmgr get-releases --allow-branch-switch --force
Choose a device:
0.	Cancel
1.	3ecf3fbe07ed8933226b294c8c19f4f9c8ebd71a (SuperIO FuSuperioIt89Device)
2.	a45df35ac0e948ee180fe216a5f703f32dda163f (System Firmware)
3.	362301da643102b9f38477387e2193e57abaa590 (UEFI dbx)
0
Request canceled
me@lite:~$ sudo fwupdtool get-devices --plugins flashrom
Loading…                 [ -                                     ]16:33:33.718 FuPluginFlashrom     At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
Loading…                 [-                                      ]
Star Labs Lite
│
└─Lite (bios):
      Device ID:          dbee8bd3b1ae0316ad143336155651eedb495a0e
      Current version:    2.1
      Vendor:             Star Labs (DMI:American Megatrends Inc.)
      GUIDs:              230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
                          048af916-1291-55aa-9ab7-e5e8018a1133 ← FLASHROM\VENDOR_Star Labs&PRODUCT_Lite&REGION_BIOS
                          9ea899e8-138e-5022-bab3-2cf2c91501ff ← FLASHROM\GUID_d5521faa-c50b-5d64-971d-8fd400030c51
                          792b324f-1e7b-5d65-ab49-643a64403a43 ← Star Labs&I3&Lite&I3&Star Labs Online Limited&Lite
                          b7af2189-7fc9-57c8-9549-106af3fa607d ← Star Labs&I3&Lite&I3
                          e234c53f-cf5e-537d-b41a-6016c7ec5cf4 ← Star Labs&I3&Lite
                          867749ec-7de0-5046-ab7d-48787144c358 ← Star Labs&I3&Star Labs Online Limited&Lite
                          ec375a72-9ed9-5a21-b1da-5e7f00dcada1 ← Star Labs&I3&Lite&I3&American Megatrends Inc.
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs shutdown after installation
                          • Cryptographic hash verification is available
    
me@lite:~$ sudo flashrom -p internal
flashrom unknown on Linux 5.19.0-35-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
No DMI table found.
Found chipset "Intel Gemini Lake".
Enabling flash write... SPI Configuration is locked down.
FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG1: BIOS region (0x00001000-0x0067efff) is read-write.
FREG5: Device Expansion region (0x0067f000-0x0077ffff) is locked.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
Enabling hardware sequencing because some important opcode is locked.
OK.
Found Programmer flash chip "Opaque flash chip" (8192 kB, Programmer-specific) on internal.
===
This flash part has status UNTESTED for operations: WP
The test status of this chip may have been updated in the latest development
version of flashrom. If you are running the latest development version,
please email a report to flashrom@flashrom.org if any of the above operations
work correctly for you with this flash chip. Please include the flashrom log
file for all operations you tested (see the man page for details), and mention
which mainboard or programmer you tested in the subject line.
Thanks for your help!
No operations were specified.
```

Sean-StarLabs, 2023-03-03:
coreboot for the Lite Mk III is in the testing remote as; it's a one-way switch, and it's still technically beta (S3 doesn't work).

esad, 2023-03-03:
> coreboot for the Lite Mk III is in the testing remote as; it's a one-way switch, and it's still technically beta (S3 doesn't work).

I see, thanks for the info! I guess I will hold off with the switch then :)