StarLabsLtd/firmware #25: Firmware security

vanntile, 2022-01-01:
We can see on [LVFS's archive](https://fwupd.org/lvfs/devices/com.starlabs.B5.coreboot.flashrom) that the latest coreboot firmware releases are not signed. Can this be rectified on future releases?

More so, when running `fwupdmgr security --force` with the latest version (7.8 from testing) we can see that the Host Security ID is reported as low. The security warnings (removed the specific ones due to privacy):

```
This system has a low HSI security level.
 » https://github.com/fwupd/fwupd/wiki/Low-host-security-level

This system has HSI runtime issues.
 » https://github.com/fwupd/fwupd/wiki/Host-security-ID-runtime-issues
```
Any changes an user can do to enforce a better state?

Sean-StarLabs, 2022-01-01:
Maybe in the future, coreboot can offer a better score and signing - but it won't be the next release.

Some of the points that coreboot falls short on are down to the way it is updated; you could get a better score by building a version of coreboot that wouldn't allow updates but that in itself has drawbacks. Some more info on that can be found [here](https://support.starlabs.systems/kb/faqs/ami-aptio-v-vs-coreboot)

If security is a priority, I'd recommend using AMI.

vanntile, 2022-01-01:
Understandable, I guess we'll just have to wait for when the versions will be signed.

closed 2022-01-01 by vanntile