StarLabsLtd/firmware #26: coreboot security features

athyla, 2022-01-10:
Hello!

I've had a short play with coreboot on StarBook Mk V. 
I looks cool and works fine, however it was too easy to switch branch form coreboot to AMI using a live medium.
So I was wondering are there or will there be security features implemented for coreboot, like:
- password protection for the settings interface
- boot protection
- upgrade/downgrade/fashing protection
- a lock protection similar to AMI BIOS lock, that prevents updates and switching branches
- other security features for various possible attacks

Thanks!

Sean-StarLabs, 2022-01-10:
We're working on BootGuard for coreboot, don't have an ETA yet though.

athyla, 2022-01-10:
No problem about ETA, I will stay on AMI for the moment, but it's very good to hear that there are plans to implement security feature for coreboot in the future.
What payload is used currently with coreboot?

Sean-StarLabs, 2022-01-10:
It's edk2, same as AMI

athyla, 2022-01-17:
Thanks, looking forward for new versions of coreboot!

closed 2022-01-17 by athyla