StarLabsLtd/firmware #27: (Starbook MK V) Power button event not handled

andrego, 2022-01-30:
Issue: Unable to use the power key to trigger the "power off" dialog usually present in most desktop environments

Expected behavior: when pressing power button, XF86PowerOff key code is sent and an event is recorded in evdev

Actual result: No key press is registered, no event is reported

Tested in: Arch Linux, Ubuntu 21.10

Steps to reproduce:
$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event2:	AT Translated Set 2 keyboard
...
/dev/input/event21:	USB 2.0 Camera: USB Camera
Select the device event number [0-21]: 1
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x0 product 0x1 version 0x0
Input device name: "Power Button"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 116 (KEY_POWER)
Properties:
Testing ... (interrupt to exit)

Sean-StarLabs, 2022-01-30:
Hey!

What firmware is that for?

andrego, 2022-01-30:
I'm running the coreboot firmware.  I've not tested using AMI, but am happy to do so if that helps isolate where the issue lies.

Sean-StarLabs, 2022-01-31:
Which version?

andrego, 2022-01-31:
coreboot: 7.91 (2022-01-06)

Sean-StarLabs, 2022-01-31:
Strange - works for me on Ubuntu 21.10. Can you try pressing the key for 3 seconds to see if that brings up the power dialogue?

andrego, 2022-02-01:
That does seem to work fairly consistently, thanks Sean.

I can see why a delay might be useful to prevent accidental shutdowns, but it was rather unintuitive given that holding a key forces an unsafe shutdown on ATX systems.  The default action can also be configured via systemd/logind.conf.
 
Any chance for a future option in corebook-configurator to reduce or remove the delay to treat as a normal key press?

Sean-StarLabs, 2022-02-01:
I'll add it to the list, should be possible :)

jozic, 2023-04-19:
Also, on a locked screen power button does nothing, Linux Mint 21

Sean-StarLabs, 2023-04-20:
> Also, on a locked screen power button does nothing, Linux Mint 21

The above won't change that, it's up to the display manager that the distro uses (LDM) as to what it does when the power button is pressed.

Sean-StarLabs, 2023-06-22:
Mapping to `E0 | 0x37` (power event) causes a fair few OS to just shut down, so not an option.

Delay has to be more than a second, so the lowest we could go is 2 seconds - not really worth an option for that. Source code for the EC will be published in the not too distant future, so building it with a 2 second delays will be possible then

closed 2023-06-22 by Sean-StarLabs