StarLabsLtd/firmware #29: updating fwupd on Mint 20.2 fails

mrberry, 2022-02-28:
After updating on Mint 20.2 fails with error:

E: /var/cache/apt/archives/libfwupdplugin5_1.7.7+f_amd64.deb: trying to overwrite '/usr/lib/x86_64-linux-gnu/libfwupdplugin.so.5.0.0', which is also in package libfwupdplugin4:amd64 1.7.4+f+3

Reading database ... 601521 files and directories currently installed.)
Preparing to unpack .../libfwupdplugin5_1.7.7+f_amd64.deb ...
Unpacking libfwupdplugin5:amd64 (1.7.7+f) ...
dpkg: error processing archive /var/cache/apt/archives/libfwupdplugin5_1.7.7+f_amd64.deb (--unpack):
trying to overwrite '/usr/lib/x86_64-linux-gnu/libfwupdplugin.so.5.0.0', which is also in package libfwupdplugin4:amd64 1.7.4+f+3
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
/var/cache/apt/archives/libfwupdplugin5_1.7.7+f_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
A package failed to install. Trying to recover:
dpkg: dependency problems prevent configuration of fwupd:
fwupd depends on libfwupdplugin5 (>= 1.7.7+f); however:
Package libfwupdplugin5:amd64 is not installed.

dpkg: error processing package fwupd (--configure):
dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of fwupd-signed:
fwupd-signed depends on fwupd (= 1.7.7+f); however:
Package fwupd is not configured yet.

dpkg: error processing package fwupd-signed (--configure):
dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.31-0ubuntu9.2) ...
Errors were encountered while processing:
fwupd

Unable to fix broken package.

fwupd version information

fwupd-gui/focal,focal,now 1 all [installed]
fwupd-signed/focal,now 1.62+f+1.7.7+f amd64 [installed]
fwupd-unsigned/focal,now 1.7.7+f amd64 [installed]
fwupd/focal,now 1.7.7+f amd64 [installed]
libfwupd2/focal,now 1.7.7+f amd64 [installed]
libfwupdplugin1/focal-updates,now 1.5.11-0ubuntu1~20.04.2 amd64 [installed]
libfwupdplugin4/focal,now 1.7.4+f+3 amd64 [installed]

fwupdmgr --version
fwupdmgr --version
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

Please note how you installed it (apt, dnf, pacman, source, etc):
apt (Synatptic package manager)

fwupd device information
Please provide the output of the fwupd devices recognized in your system.

fwupdmgr get-devices --show-all-devices
fwupdmgr get-devices --show-all-devices
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

Operating system and version: Mint 20.2

Sean-StarLabs, 2022-02-28:
Looks like you've got multiple versions of `libfwupdplugin1` installed.

If you remove the older ones with:
```
sudo dpkg --remove libfwupdplugin1 libfwupdplugin4
```
And then finish the install with:
```
sudo apt -f instal
```
That should do the trick :)

mrberry, 2022-02-28:
Sean, that solved the problem. This happened on two different laptops: a Starlite running Mint 20 and a Starbook running Mint 20.2 after running the Mint update manager. The Synaptic package manger was unable to fix it. Others using Mint might see this issue.

Thanks,

Mark

achou11, 2022-03-01:
Was experiencing a similar issue with Elementary. @Sean-StarLabs's suggestion worked for me :+1:

Sean-StarLabs, 2022-05-06:
The Manjaro package was updated yesterday, so I'd recommend checking that's up to date.

For the `Message recipient disconnected...` error, you need the `iomem=relaxed` kernel parameter - how to add [here](https://support.starlabs.systems/kb/updates/installing-updates-from-the-lvfs)

Jetlagger, 2022-05-06:
Thanks for the reply! I have set `iomem=relaxed`.

Nonetheless, I only get `Message recipient disconnected...` if I use fwupd instead of fwupd-starlabs.
With fwupd-starlabs installed, I don't even get that far, as the gnome-firmware GUI does not recognize any devices at all on my system.

The version of fwupd-starlabs I have is 1.7.6r321.g539413ca-1, the version of gnome-firmware is 42.0-1.

Jetlagger, 2022-05-06:
Haha, so sorry about the amount of work this is causing!

```
× fwupd.service - Firmware update daemon
     Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static)
     Active: failed (Result: exit-code) since Fri 2022-05-06 12:55:56 CEST; 11min ago
       Docs: https://fwupd.org/
    Process: 1775 ExecStart=/usr/lib/fwupd/fwupd (code=exited, status=1/FAILURE)
   Main PID: 1775 (code=exited, status=1/FAILURE)
        CPU: 10ms

Mai 06 12:55:56 manstaro systemd[1]: Starting Firmware update daemon...
Mai 06 12:55:56 manstaro fwupd[1775]: /usr/lib/fwupd/fwupd: /usr/lib/libjcat.so.1: version `LIBJCAT_0.1.11' not found (required by /usr/lib/fwupd/fwupd)
Mai 06 12:55:56 manstaro systemd[1]: fwupd.service: Main process exited, code=exited, status=1/FAILURE
Mai 06 12:55:56 manstaro systemd[1]: fwupd.service: Failed with result 'exit-code'.
Mai 06 12:55:56 manstaro systemd[1]: Failed to start Firmware update daemon.

```

Sean-StarLabs, 2022-05-06:
Thanks,

```
sudo pacman -Syu libjcat
sudo service fwupd restart
```

Should solve that :)

Sean-StarLabs, 2022-05-06:
Awesome, I'll let the Manjaro guys know so they can make `0.1.11` available :)

closed 2022-05-06 by Sean-StarLabs