StarLabsLtd/firmware #30: coreboot - Star Lite MKIII - 2.1.0 -> 4.15  "firmware is 3b0000 bytes larger..."

Gadgetoid, 2022-03-03:
Running into an(other) issue updating the Star Lite MKIII to Coreboot 4.15 on Linux Mint 20.3:

```
fwupdmgr switch-branch dbee8bd3b1ae0316ad143336155651eedb495a0e coreboot
<snip>
Perform operation? [Y|n]: y
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Decompressing…           [***************************************]
firmware is 3b0000 bytes larger than the allowed maximum size of 450000 bytes
```

Have tried `iomem=relaxed` in kernel command line.

Flashrom output:

```
phil@StarLite:~$ sudo flashrom -p internal
flashrom v1.2 on Linux 5.4.0-100-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
No DMI table found.
Found chipset "Intel Gemini Lake".
Enabling flash write... SPI Configuration is locked down.
FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG5: Device Expansion region (0x0067f000-0x0077ffff) is locked.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
Enabling hardware sequencing because some important opcode is locked.
OK.
Found Programmer flash chip "Opaque flash chip" (8192 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
No operations were specified.
```

I was previously running into "failed to open device for prepare: flash probe failed: no chip was found" which I fixed by purging "fwupd fwupd-gui fwupd-signed libfwupdplugin5 starlabs-lite-mk-iii" followed by an autoremove to grab any related packages and then reinstalling "starlabs-lite-mk-iii" and letting it bring its depends back.

Gadgetoid, 2022-03-03:
Trying `fwupdtool` yields worse results:

```
sudo fwupdtool switch-branch 
Loading…                 [-                                      ]20:19:32:0026 FuDevice             no FuContext assigned for FuFlashromInternalDevice:
  Protocol:             org.flashrom
  Flags:                internal|updatable|require-ac|needs-shutdown
  InternalFlags:        md-set-signed

Loading…                 [***************************************]
Choose a device:
0.	Cancel
1.	dbee8bd3b1ae0316ad143336155651eedb495a0e (Lite)
2.	c46944fe84f2b057692258cc87a812cc86364881 (Star Drive SATA SSD)
3.	3ecf3fbe07ed8933226b294c8c19f4f9c8ebd71a (SuperIO FuSuperioIt89Device)
4.	a45df35ac0e948ee180fe216a5f703f32dda163f (System Firmware)
5.	362301da643102b9f38477387e2193e57abaa590 (UEFI dbx)
1 

<snip>

Decompressing…           [***************************************]
20:19:42:0876 FuPluginFlashrom     At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
Restarting device…       [  -                                    ]20:19:42:0885 FuPluginFlashrom     At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
Writing…                 [   -                                   ]20:19:43:0137 FuPluginFlashrom     At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
Decompressing…           [    \                                  ]20:19:43:0386 FuPluginFlashrom     At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
20:19:43:0636 FuPluginFlashrom     At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
firmware is 3b0000 bytes larger than the allowed maximum size of 450000 bytes
```

The size error traces back to `libfwupdplugin/fu-device.c` which suggests this can be overridden in `/usr/share/fwupd/quirks.d/flashrom.quirk` with `FirmwareSizeMax = 0x800000` but that does not make a difference.

When I updated to 2.1.0 I ran into different problems with fwupd - which I cannot recall - and had to eventually use the EFI updater.

Sean-StarLabs, 2022-03-03:
Hmmm, interesting.

Just to check, have you turned off BIOS Lock?

Gadgetoid, 2022-03-03:
Yes, BIOS Lock is off. I think confirmed by:

```
sudo fwupdmgr get-devices
Lite
│
├─Lite:
│ │   Device ID:          dbee8bd3b1ae0316ad143336155651eedb495a0e
│ │   Current version:    2.1.0
│ │   Vendor:             Star Labs (DMI:American Megatrends Inc.)
│ │   GUIDs:              230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │                       792b324f-1e7b-5d65-ab49-643a64403a43 ← Star Labs&I3&Lite&I3&Star Labs Online Limited&Lite
│ │                       b7af2189-7fc9-57c8-9549-106af3fa607d ← Star Labs&I3&Lite&I3
│ │                       e234c53f-cf5e-537d-b41a-6016c7ec5cf4 ← Star Labs&I3&Lite
│ │                       867749ec-7de0-5046-ab7d-48787144c358 ← Star Labs&I3&Star Labs Online Limited&Lite
│ │                       ec375a72-9ed9-5a21-b1da-5e7f00dcada1 ← Star Labs&I3&Lite&I3&American Megatrends Inc.
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs shutdown after installation
│ │                       • Device supports switching to a different branch of firmware
```

Tempted to try flashrom directly, but I may be playing with fire there. Specially with the dance around the ME regions. (And the fact I have very little idea what I'm doing and just enough knowledge to be dangerous.)

Gadgetoid, 2022-03-03:
All this fancy firmware tooling ultimately just runs the following with extra steps, right?

```
flashrom -p internal -w 4.15.rom --ifd -i bios --noverify-all
```

Sean-StarLabs, 2022-03-03:
`fwupdmgr security --force | grep 'SPI write'` is the easiest way to check (get-devices sometimes wont say).

Hehe - pretty much, but it should figure out the switches, and for the Lite, it would just be
```
flashrom -p internal -w 4.15.rom
```
If `YOLO == true`, I'd give that a try, don't reboot and post the output. If something goes wrong, we can reverse it (until it's rebooted)

Gadgetoid, 2022-03-03:
SPI Write is- apparently- disabled.

```
fwupdmgr security --force
Host Security ID: HSI:0! (v1.7.7)

HSI-1
✔ Intel DCI debugger:            Disabled
✔ SPI write:                     Disabled
✔ TXE manufacturing mode:        Locked
✔ TXE override:                  Locked
✔ TXE v0:4.0.20.1310:            Valid
✘ SPI BIOS region:               Unlocked
✘ SPI lock:                      Disabled
✘ TPM v2.0:                      Not found
✘ UEFI platform key:             Invalid
```

No dice with flashrom still, baulks *reading* the old contents:

```
sudo flashrom -p internal -w 4.15.rom 
flashrom v1.2 on Linux 5.4.0-100-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
No DMI table found.
Found chipset "Intel Gemini Lake".
Enabling flash write... SPI Configuration is locked down.
FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG5: Device Expansion region (0x0067f000-0x0077ffff) is locked.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
Enabling hardware sequencing because some important opcode is locked.
OK.
Found Programmer flash chip "Opaque flash chip" (8192 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
Reading old flash chip contents... Transaction error between offset 0x0067f000 and 0x0067f03f (= 0x0067f000 + 63)!
FAILED.
```

I seem to have the weirdest system ever here.

How accessible is the flash chip on board if I toast this and need to program it manually? :laughing:

Sean-StarLabs, 2022-03-03:
Thanks - I'll do some testing tomorrow, see what I can find.

There's a 12pin FFC connector on the board under the headsink that can flash the BIOS, so if you have the right cable, easy ;)

Gadgetoid, 2022-03-03:
Cheers, let me know if you need anything else poked & prodded.

Okay so worst case I have to bodge a cable and spend six weeks trying to write an RP2040-based flasher because I'm too cheap to buy tools to save a *checks notes* £140 mainboard :laughing: 

In the interim I'll see if I can find time to boot something from USB and try again- lord only knows what I might have messed up by grabbing an early update to Mint 20.3.

Sean-StarLabs, 2022-03-03:
^ That output says nothing was written, so nothing is broken!

`flashrom` already has support for Pi's, £5 + Amazon would probably do it. You won't be able to solve it with a different distro though, it'll either be mods to flashrom or an adjustment to the AMI firmare.

Sean-StarLabs, 2022-03-04:
So `flashrom -p internal -w 4.15.rom --ifd -i bios -n` will get the job done, which doesn't quite make sense.

That wont work for getting back to AMI, so if you try it, it'll be a one way trip (for the moment)

Gadgetoid, 2022-03-09:
Still no dice, even with this, fails reading the old contents:

```
sudo flashrom -p internal -w 4.15.rom --ifd -i bios -n

flashrom v1.2 on Linux 5.4.0-104-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
No DMI table found.
Found chipset "Intel Gemini Lake".
Enabling flash write... SPI Configuration is locked down.
FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG5: Device Expansion region (0x0067f000-0x0077ffff) is locked.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
Enabling hardware sequencing because some important opcode is locked.
OK.
Found Programmer flash chip "Opaque flash chip" (8192 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
Reading ich descriptor... done.
Using region: "bios".
Reading old flash chip contents... Transaction error between offset 0x0067f000 and 0x0067f03f (= 0x0067f000 + 63)!
FAILED.
```

Which is completely balmy because I can *read* with:

```
sudo flashrom -p internal --ifd -i bios -n -r testing1.bin
```

Would dropping to EFI shell and using `AfuEfix64 4.15.bin /P /N` work?

Sean-StarLabs, 2022-03-09:
Hmm, I'll fire a Mint one up. Is that version of flashrom from our PPA or Ubuntu's?

Afu wouldn't work, Fpt might - but we need to avoid that at all costs as it's problematic to say the least!

Gadgetoid, 2022-03-09:
Latest from your PPA:

```
apt-cache policy flashrom
flashrom:
  Installed: 1.2-6+f
  Candidate: 1.2-6+f
  Version table:
 *** 1.2-6+f 500
        500 http://ppa.launchpad.net/starlabs/ppa/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status
     1.2-5 500
        500 http://mirror.bytemark.co.uk/ubuntu focal/universe amd64 Packages
        500 http://mirror.bytemark.co.uk/ubuntu focal-updates/universe amd64 Packages
```

Plus also the very latest built from their github mirror.

Chance that me using Afu to flash 2.1.0 caused this weirdness?

Sean-StarLabs, 2022-03-09:
Thanks

> Chance that me using Afu to flash 2.1.0 caused this weirdness?
None. Afu will do the same thing as fwupd, just in a more complicated way. It's just the security doing its job (a little too well!)

Gadgetoid, 2022-03-09:
Sheesh I should have spotted that! Here goes nothing :grimacing:

Gadgetoid, 2022-03-09:
We have lift off. Time to grit my teeth even harder and reboot.

```
sudo flashrom -p internal --ifd -i bios -N -w ~/4.15.rom
flashrom v1.2 on Linux 5.4.0-104-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
No DMI table found.
Found chipset "Intel Gemini Lake".
Enabling flash write... SPI Configuration is locked down.
FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG5: Device Expansion region (0x0067f000-0x0077ffff) is locked.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
Enabling hardware sequencing because some important opcode is locked.
OK.
Found Programmer flash chip "Opaque flash chip" (8192 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
Reading ich descriptor... done.
Using region: "bios".
Reading old flash chip contents... done.
Erasing and writing flash chip... Erase/write done.
Verifying flash... VERIFIED.
```

Sean-StarLabs, 2022-03-09:
Solved with f98acc89b39c45b84e7011473b8ad3e1f1b9eb8f

closed 2022-03-09 by Sean-StarLabs