StarLabsLtd/firmware #31: coreboot - 4.15 - Star Lite MKIII - keyboard backlight non-functional

Gadgetoid, 2022-03-09:
* Flashed coreboot 4.15
* Launched coreboot-configurator
* Set keyboard backlight timeout to 3 minutes
* Reboot
* No keyboard backlight when typing
* Pressing fn + f11 will intermittently cause the backlight to flash on, but not *stay* on
* Note it's obvious that brightness control is still working because pressing fn + f11 three times in a row will: "flash dim, flash bright, not flash at all" in a repeating pattern

Gadgetoid, 2022-03-09:
I may be a bit late to the party here? https://github.com/StarLabsLtd/coreboot/commit/a22847209f88f51c38a7cb90fd717107a58b4b5b

Happy to flash a bleeding-edge build, or build my own periodically from source if you've got a .config handy.

Sean-StarLabs, 2022-03-09:
That's a corrupt CMOS - can happen when switching. fwupd has a function to reset it, but you should you just need:
```
sudo nvramtool -w kbl_state=0x1
```

If not, I'll pull the reset function out of fwupd and send it over

Gadgetoid, 2022-03-09:
That doesn't seem to be enough- in fact it gets reset back to 0x0. Confirmed via coreboot-configurator.

Re-ran fwupd because it works now (of course it does) and it reinstalled coreboot (did an explicit install from the .cab file and it *seemed* to complete) and the issue persists.

Sean-StarLabs, 2022-03-09:
Gotcha. It might work writing directly to the EC, i.e. `sudo ectool -w 0x19 -z 0xdd`, but we can solve it in coreboot itself. I'll generate a build tomorrow that'll solve this and send it over

Gadgetoid, 2022-03-09:
Can confirm `sudo ./ectool -w 0x19 -z 0xdd` fixed it instantly. Thank you!

Sean-StarLabs, 2022-03-09:
Ha - yes, will probably reset on a reboot; EC memory and CMOS should stay in sync, and that's not possible to do with any tool that I know of.

As for breaking it, yes - writing any value other than `0xdd` or `0x00` to offset `0x19` would effectively break it.

Gadgetoid, 2022-03-09:
Indeed it does!

Found the offset that controls keyboard backlight intensity which is *very* interesting because I'm always finding the backlight turns off when I'm mousing for a while and have nefarious plans to poke it back to life on trackpad input if that's not something that could be baked into the firmware directly :thinking:

Sean-StarLabs, 2022-03-09:
It can't - the firmware doesn't know about cursor movement, so it would need help from the OS to do that.

Sean-StarLabs, 2022-03-14:
As always, took longer than expected but the new release is up - 862f6d39d970c18045a0c875def017430448afcc

Let me know how that goes :)

Sean-StarLabs, 2022-03-14:
Is that since the switch to coreboot?

Gadgetoid, 2022-03-14:
Back to being on topic- this new version has not fixed the keyboard backlight. Even if I run `sudo nvramtool -w kbl_state=0x1`. 

Flashed twice, first with fwupd and then with flashrom.

`sudo ./ectool -w 0x19 -z 0xdd` still gets it back up and running.

Sean-StarLabs, 2022-03-14:
If it's just S3 (suspend), dmesg is the first place to look (set it printing to a text file). Make things verbose by adding `acpi.debug_layer=0xffffffff acpi.debug_level=0x2`.

The vast majority of suspend issues we see are due to either snap, or misconfiguration.

Gadgetoid, 2022-03-30:
This is fixed in the 4.16 as of 6 days ago, which I managed to install via fwupd today. Needed the "--force" argument to reconcile the difference in version numbering schemes, but otherwise seems to have installed fine. Hooray.

Good grief does the Lite stay on a black screen for a while after an update though. *Just* long enough to make me start to worry :laughing:

closed 2022-03-30 by Gadgetoid

Sean-StarLabs, 2022-03-31:
Glad to hear :) The fwupd issue has been fixed, just needs to trickle down.