StarLabsLtd/firmware #33: Star Lite Mk IV does not charge via USB docking station

elagost, 2022-05-15:
When the Star Lite Mk IV is connected to a USB-C dongle/hub with power delivery in, it does not charge. /sys/class/power_supply/BAT0/status says 'charging', and GNOME reports charging. I'm running Fedora 36, but booted into a live Manjaro USB stick and saw the same behavior.

I tried a Brydge Stone II, a Vava USB-C hub with power delivery in, and a Dell monitor that has a built-in hub. All of these were able to connect to my wife's system76 lemur pro (also with coreboot) without issue. It was able to charge. My work macbook also works well with all these devices.

I've tried both AMI and Coreboot but strongly prefer the latter. On coreboot I get the following message logged to `dmesg` every couple seconds for a few minutes after plugging in: 

`usb usb2-port4: Cannot enable. Maybe the USB cable is bad?`

Please let me know if this is the wrong place to report this issue.

Sean-StarLabs, 2022-05-15:
Hey

Can you let me know:
What rate does it report?
What charger are you using with the hubs?
Do they support PD 3.0?
What's the model of the Dell monitor?

elagost, 2022-05-15:
1) Not sure on the rate - how would I obtain that? Is it in /sys too?
2) Both an apple 65w macbook charger and [this](https://www.amazon.com/ZMI-zPower-Turbo-Power-Adapter/dp/B07D64QLQ1) that I've had for years and used without issue.
3) They should - used them with multiple devices for a while without issue.
4) The Dell is a U2721DE.

Sean-StarLabs, 2022-05-16:
> Not sure on the rate - how would I obtain that? Is it in /sys too?

`upower -i /org/freedesktop/UPower/devices/battery_BAT0`

> Both an apple 65w macbook charger and [this](https://www.amazon.com/ZMI-zPower-Turbo-Power-Adapter/dp/B07D64QLQ1) that I've had for years and used without issue.

Most Apple chargers are PD 2.0, and the ZMI doesn't say which version it is. 

Can you post the output of `upower` command with the included Star Labs charger and then the ZMI one?

elagost, 2022-05-16:
Command: `upower -i /org/freedesktop/UPower/devices/battery_BAT0`

Charger 1: Star Labs USB-C charger:

    native-path:          BAT0
    vendor:               DGFGE
    model:                597077-3S
    serial:               3ICP6/70/77
    power supply:         yes
    updated:              Mon 16 May 2022 06:46:25 AM CDT (28 seconds ago)
    has history:          yes
    has statistics:       yes
    battery
      present:             yes
      rechargeable:        yes
      state:               charging
      warning-level:       none
      energy:              29.792 Wh
      energy-empty:        0 Wh
      energy-full:         30.4 Wh
      energy-full-design:  30.4 Wh
      energy-rate:         11.4 W
      voltage:             8.693 V
      charge-cycles:       N/A
      time to full:        3.2 minutes
      percentage:          98%
      capacity:            100%
      icon-name:          'battery-full-charging-symbolic'
    History (charge):
      1652701585	98.000	charging

Charger 2: Apple MacBook Pro (M1) included charger (65w?):

    native-path:          BAT0
    vendor:               DGFGE
    model:                597077-3S
    serial:               3ICP6/70/77
    power supply:         yes
    updated:              Mon 16 May 2022 07:07:04 AM CDT (5 seconds ago)
    has history:          yes
    has statistics:       yes
    battery
      present:             yes
      rechargeable:        yes
      state:               charging
      warning-level:       none
      energy:              28.88 Wh
      energy-empty:        0 Wh
      energy-full:         30.4 Wh
      energy-full-design:  30.4 Wh
      energy-rate:         11.4 W
      voltage:             8.688 V
      charge-cycles:       N/A
      time to full:        8.0 minutes
      percentage:          95%
      capacity:            100%
      icon-name:          'battery-full-charging-symbolic'
    History (charge):
      1652702824	95.000	charging
    History (rate):
      1652702824	11.400	charging

Charger 3: ["ZMI" charger (amazon something or other)](https://www.amazon.com/ZMI-zPower-Turbo-Power-Adapter/dp/B07D64QLQ1):

    native-path:          BAT0
    vendor:               DGFGE
    model:                597077-3S
    serial:               3ICP6/70/77
    power supply:         yes
    updated:              Mon 16 May 2022 07:09:13 AM CDT (1 seconds ago)
[... 97 lines cut ...]
    [ 3311.406682] usb 2-4: device not accepting address 45, error -71
    [ 3315.478927] usb usb2-port4: Cannot enable. Maybe the USB cable is bad?
    [ 3328.166799] usb usb2-port4: Cannot enable. Maybe the USB cable is bad?
    [ 3332.958768] usb usb2-port4: Cannot enable. Maybe the USB cable is bad?
    [ 3337.014756] usb usb2-port4: Cannot enable. Maybe the USB cable is bad?
    [ 3340.054280] usb 2-4: Device not responding to setup address.
    [ 3340.266285] usb 2-4: Device not responding to setup address.
    [ 3340.470599] usb 2-4: device not accepting address 47, error -71
    [ 3341.478281] usb 2-4: Device not responding to setup address.
    [ 3341.690088] usb 2-4: Device not responding to setup address.
    [ 3341.894581] usb 2-4: device not accepting address 48, error -71
    [ 3341.894779] usb usb2-port4: attempt power cycle
    [ 3342.586606] usb 2-4: new SuperSpeed USB device number 49 using xhci_hcd
    [ 3342.602321] usb 2-4: New USB device found, idVendor=0bda, idProduct=0423, bcdDevice= 0.04
    [ 3342.602404] usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
    [ 3342.602413] usb 2-4: Product: 4-Port USB 3.0 Hub
    [ 3342.602419] usb 2-4: Manufacturer: Generic
    [ 3347.854682] usb 2-4: Enable of device-initiated U1 failed.
    [ 3347.854780] usb 2-4: Set SEL for device-initiated U2 failed.
    [ 3347.854979] hub 2-4:1.0: USB hub found
    [ 3347.854990] hub 2-4:1.0: config failed, can't read hub descriptor (err -22)
    [ 3347.855225] usb 2-4: Failed to suspend device, error -19
    [ 3348.123332] usb 2-4: USB disconnect, device number 49

Probably more than you asked for. I tried out AMI, and I don't get that extra dmesg spam. I'm back on coreboot though and intend to stay there.

elagost, 2022-05-16:
Every charger plugged directly into the laptop charges the laptop.
Every charger plugged into a dock which is plugged into the laptop says it's charging the laptop, but the laptop battery percentage goes down over time.

elagost, 2022-05-16:
I found a combination that does charge the laptop!

Star Labs USB-C charger+cable plugged into [a vava usb-c hub](https://www.amazon.com/VAVA-Adapter-Ethernet-Charging-MacBook/dp/B07X1PR3PY):

```
  native-path:          BAT0
  vendor:               DGFGE
  model:                597077-3S
  serial:               3ICP6/70/77
  power supply:         yes
  updated:              Mon 16 May 2022 08:37:45 AM CDT (10 seconds ago)
  has history:          yes
  has statistics:	yes
  battery
    present:             yes
    rechargeable:        yes
    state:               charging
    warning-level:	 none
    energy:              26.752 Wh
    energy-empty:        0 Wh
    energy-full:         30.4 Wh
    energy-full-design:  30.4 Wh
    energy-rate:         11.4 W
    voltage:             8.677 V
    charge-cycles:	 N/A
    time to full: 	 19.2 minutes
    percentage:          88%
    capacity:            100%
    icon-name:          'battery-full-charging-symbolic'
  History (charge):
    1652708265  88.000  charging
  History (rate):
    1652708265  11.400  charging
```

elagost, 2022-05-20:
Also, sometimes I get the `usb usb2-port2: Cannot enable. Maybe the USB cable is bad?` dmesg error when plugging in a USB drive to the left-hand side USB port, and it doesn't recognize the disk until a usb-c cable is plugged in.

Is there maybe something wrong with my hardware?

Sean-StarLabs, 2022-05-20:
No, that's #35 - I've got a solution for that, will finish testing Monday and publish.

elagost, 2022-05-20:
Thank you; I will be patient.

Sean-StarLabs, 2022-06-01:
I've just added a new EC which solves this issue (019f195bed8e27895700dd31066328d483d768c8) which you can find [here](https://github.com/StarLabsLtd/firmware/raw/master/StarLite/MkIV/Nuvoton/1.01/efi-i4.zip).

This can be installed if using **AMI** firmware by following [this](https://support.starlabs.systems/kb/updates/installing-updates-with-the-efi-shell). 

For coreboot, we need to open a couple of registers to install correctly; that'll be done shortly.

elagost, 2022-06-01:
Thank you very much! I will wait for the coreboot release as that is what I'm running.

Sean-StarLabs, 2022-06-14:
`4.17` is now in LVFS testing. It'll automatically be pushed to stable on the 17th of June if no issues are reported. That will allow the EC to be updated following the instructions below:

> I've just added a new EC which solves this issue ([019f195](https://github.com/StarLabsLtd/firmware/commit/019f195bed8e27895700dd31066328d483d768c8)) which you can find [here](https://github.com/StarLabsLtd/firmware/raw/master/StarLite/MkIV/Nuvoton/1.01/efi-i4.zip).
> 
> This can be installed if using **AMI** firmware by following [this](https://support.starlabs.systems/kb/updates/installing-updates-with-the-efi-shell).
> 
> For coreboot, we need to open a couple of registers to install correctly; that'll be done shortly.

closed 2022-06-14 by Sean-StarLabs

elagost, 2022-06-17:
I was able to install the coreboot update and the EC update. Charging works as expected from the dock now. Thank you!
The EC update was a little fiddly to install, and had manually boot one of the efi files, but once I found the one that worked, it went well.

Sean-StarLabs, 2022-06-17:
Awesome :) I'll take a look at that it not starting automatically.