StarLabsLtd/firmware #34: Starlite MK IV "No devices found"

markmaas, 2022-05-18:
I was trying to see what I can do with this nifty little laptop, so I tried switching to coreboot / ami bios using fwupd from the starlabs ppa.

unfortunately, even when following https://support.starlabs.systems/kb/guides/installing-coreboot I still get this:

```
mark@starlite:~$ fwupdmgr switch-branch
No supported devices
```

also:
```
mark@starlite:~$ fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 0 local devices supported
```

This is Pop!_os 22.04
Any suggestions perhaps?

Sean-StarLabs, 2022-05-18:
It's most likely outdated fwupd or flashrom, new enough versions are in our PPA. You'll have to refer to the Pop documentation of installing from third-party repo's as they do a few things that make it different.

markmaas, 2022-05-18:
Ah I was indeed under the impression the upgrade had already been done. But I had to do:

```
sudo apt install libfwupd2=1.8.1+starlabs+1+j
sudo apt install fwupd=1.8.1+starlabs+1+j
```

before it would work. Now i'm off to find a better way, pinning probably.

echo-devnull, 2023-05-07:
For the next poor soul annoyed things have changed....:

```sh
sudo add-apt-repository ppa:starlabs/main
sudo add-apt-repository ppa:starlabs/ppa
```

`/etc/apt/preferences.d/pop-default-settings`

```
Package: *
Pin: release o=LP-PPA-starlabs-main
Pin-Priority: 2001

Package: *
Pin: release o=LP-PPA-starlabs-ppa
Pin-Priority: 2001
```

Yeah, they changed the names