StarLabsLtd/firmware #35: StarLite MkIV with coreboot: boot from usb not working

rinoandrejohnsen, 2022-05-19:
Hi, I would love to install another distro on the StarLite MkIV, but from what I can gather of information, it seems to not be possible? Can anyone confirm this?

Sean-StarLabs, 2022-05-19:
It's possible - there's a bug affecting the left port but the right should work okay. If that doesn't work, can you let me know the how and what please?

rinoandrejohnsen, 2022-05-19:
Just tried with the right port and the left port (and the usb-c port with an adapter). The usb pen is not discovered, so all the boot menus are missing it. Another observation is that the usb pen seems to hang. My usb pen has a light that turns on when it is read from and this light is on constantly while in boot menu.

Sean-StarLabs, 2022-05-19:
What distro are you trying to install?

rinoandrejohnsen, 2022-05-19:
I am trying to install openSUSE. I am not in a rush or anything, but it would be nice to have it work at some point.

murphym18, 2022-05-20:
@rinoandrejohnsen I ran into a similar problem earlier today. I tried to boot ventoy with a USB 3.0 flashdrive that runs with the `Driver=usb-storage` in Linux. ([this exact flash drive](https://www.amazon.com/gp/product/B07RWPF43G/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1)). I only tried the left USB-A port and it didn't show up in the boot menu (I'm also using coreboot).

As a work around, I burned the ISO file to a USB 2.0 Flash Drive ([this exact one](https://www.amazon.com/dp/B001XURP8G?ref=ppx_yo2ov_dt_b_product_details&th=1)) and tried the same port. This time it worked. I could boot from the drive and reach the live environment. I've run into this problem with other intel computers and a USB 2.0 drive seems to do the trick.

barbastrejo7, 2022-05-27:
Similar problem here on a lite mkIII, some pen drives, external ssd work only on the right port, some only  on the left(seems a political problem), on  the usb-c port some work some don't.

rinoandrejohnsen, 2022-05-29:
I was able to get hold of an older 4GB USB device and I did successfully boot from it. Very happy with this computer on openSUSE. @Sean-StarLabs Do you want me to close this issue or keep it up until further notice?

Sean-StarLabs, 2022-05-30:
Awesome. If we can leave it open, then I can close it linkinh in the firmware release that fixes it :)

Sean-StarLabs, 2022-05-30:
Fixed in 14405753d183ba2354b4f55632677ab581d117b1

If anyone wants to install the fix for the StarLite Mk IV right now, the below commands will do this (please connect the charger and ensure the battery is charged to 30%):
```
wget https://github.com/StarLabsLtd/firmware/raw/master/StarLite/MkIV/coreboot/4.16/4.16.rom
sudo flashrom -p internal -w 4.16.rom -i bios --ifd -N
```

coreboot 4.17 should be tagged later on today, so I'll do a "proper" 4.17 release with this fix once that happens to it'll come through via the LVFS.

closed 2022-05-30 by Sean-StarLabs