StarLabsLtd/firmware #36: Coreboot Source code

maryandmaxcakeyogurt, 2022-05-20:
I can find Coreboot binaries, but no source code.
Could you please release the source code?

Sean-StarLabs, 2022-05-20:
It's here - https://github.com/coreboot/coreboot

maryandmaxcakeyogurt, 2022-05-20:
Coreboot is GPL. If you distribute binaries, you must also distribute the source code.
Currently, this is a violation of the GPL.

maryandmaxcakeyogurt, 2022-05-20:
You must distribute the source code corresponding to the binary in your own repository (or on a non-GitHub web server), rather than pointing to someone else's repository.

Sean-StarLabs, 2022-05-20:
Not sure I follow - Our coreboot binaries are built from there?

maryandmaxcakeyogurt, 2022-05-20:
For example, the source code corresponding to [this binary](https://github.com/StarLabsLtd/firmware/tree/master/StarBook/MkV/coreboot/7.10) must be distributed (e.g., as a tarball) on the Star Labs server.
Linking to someone else's server will cause problems.

Sean-StarLabs, 2022-05-20:
Having a copy of it somewhere else, and adding a tarball will just complicate things. If you want to see exactly what's it that version, just use the commit hash

maryandmaxcakeyogurt, 2022-05-20:
Ah, I misunderstood. I am sorry.

I have a suggestion related to this, which I will raise in a separate Issue.
This Issue is closed.

closed 2022-05-20 by maryandmaxcakeyogurt