StarLabsLtd/firmware #40: Broken install on Debian

barbastrejo7, 2022-05-25:
I recently got a lite mkIII, and tried many distros like manjaro, ubuntu, ubuntu mate, fedora and mx linux, they all work fine except for Debian that after a normal install of a non-free live image I get the following error: SATA: Star Drives SATA SSD' failed: verify it contains a 64-bit UEFI OS
While running ubuntu I ran 'efibootmgr -v' and it gives as output:
BootCurrent: 0001
Timeout: 1 seconds
BootOrder: 0003,0000,0001,0002
Boot0000* UiApp	MemoryMapped(11,0x803860,0xb2885f)/FvFile(462caa21-7614-4503-836e-8ab6f4662331)
Boot0001* SATA: Star Drive SATA SSD 	PciRoot(0x0)/Pci(0x12,0x0)/Sata(0,65535,0)N.....YM....R,Y.
Boot0002* UEFI Shell	MemoryMapped(11,0x803860,0xb2885f)/FvFile(7c04a583-9e3e-4f1c-ad65-e05268d0b4d1)
Boot0003* ubuntu	HD(1,GPT,85704b1d-494c-41e7-8d45-b56270211742,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
 
Not sure if it may help you devs to reproduce the problem I have, let me know what else I can give as info if needed

Sean-StarLabs, 2022-05-25:
Sounds like the Debian installer didn't add a boot entry. Can you start it if you run the below commands in the EFI shell:
```
fs0:
cd efi
cd boot
grubx64.efi
```

barbastrejo7, 2022-05-27:
got this output:
'grubx64.efi' is not recognized as an internal or external command, operable program, or script file

Sean-StarLabs, 2022-05-27:
What files do you have in the directory?

barbastrejo7, 2022-05-28:
.
..
boot
Debian

I was able to boot it going into 'boot from file' then efi»Debian»grubx64.efi     
But I have to do it everytime to boot.

Also after installing the coreboot version you sent on Thursday, I have to plug the power cable every time I am going to boot the machine, otherwise there's that bug, that it flashes the power led without keeping it on and nothing happens.

barbastrejo7, 2022-05-28:
I was able to add the option to add the grubx64.efi as an option to boot from, naming it debian, than changed the order for it to boot first. It worked and I could see the option saved, but once I restarted the machine it was gone.It is clearly not keeping it during a reboot.

Sean-StarLabs, 2022-05-30:
edk2 can be picky about the entries that are added, so I'd recommend adding it from Linux:
```
sudo efibootmgr -c -d /dev/sda1 -p 1 -w -L Debian -l \EFI\Debian\grubx64.efi
```

barbastrejo7, 2022-05-30:
I ran that command, but apparently nothing happened, I also tried adding the option to the boot menu, but still does not hold after the next boot.

Sean-StarLabs, 2022-05-30:
What does `efibootmgr` show?

barbastrejo7, 2022-05-30:
BootCurrent: 0003
Timeout: 1 seconds
BootOrder: 0000,0001,0002
Boot0000* UiApp
Boot0001* SATA: Star Drive SATA SSD 
Boot0002* UEFI Shell
Boot0003* shimx64.efi

barbastrejo7, 2022-05-30:
In the Debian forum, someone told me to run: 
grub-install --force-extra-removable

It worked!

barbastrejo7, 2022-05-30:
I also asked about in the Debian forum, and someone asked to run: grub-install --force-extra-removable
It worked, now  efibootmgr outputs:
BootCurrent: 0001
Timeout: 1 seconds
BootOrder: 0003,0000,0001,0002
Boot0000* UiApp
Boot0001* SATA: Star Drive SATA SSD
Boot0002* UEFI Shell
Boot0003* debian

closed 2022-05-30 by Sean-StarLabs