StarLabsLtd/firmware #50: [StarBook MkV] can't upgrade to coreboot 8.02

twiztedtbone, 2022-07-07:
Howdy!

Having a weird issue trying to update coreboot.

OS: Arch
Current coreboot: 8.00

`[xxxx@starpod ~]$ fwupdmgr update
Reading…                 [-                                      ]Devices with no available firmware updates: 
 • 0000:00:1f.5
 • Star Drive PCIe SSD
**StarBook (bios) is not currently updatable: failed to open device for prepare: invalid region name**
`
I've tried searching for the issue, but the phrase "invalid region name" doesn't actually return any search results that I can find, so I'm at a bit of a loss for next steps.

Lemme know if any additional info would help.
T

Sean-StarLabs, 2022-07-07:
Can you let me know:
* kernel version
* flashrom version
* fwupd version
* output of `sudo flashrom -p internal`

Sean-StarLabs, 2022-07-07:
No, that's happy.

Did you refresh? i.e. `fwupdmgr refresh --force`

twiztedtbone, 2022-07-07:
I did before opening the issue, yea. Here's what that looks like in full:
```
[xxxx@starpod ~]$ fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 2 local devices supported
[tpautsch@starpod ~]$ fwupdmgr update
Devices with no available firmware updates: 
 • 0000:00:1f.5
 • Star Drive PCIe SSD
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade StarBook (bios) from 8.00 to 8.02?                                   ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release includes the following features:                         ║
║                                                                              ║
║ • Disabled edk2 serial debug support for faster boot times                   ║
║ • Configured PCIe Base Address in edk2 to avoid auto-config                  ║
║ • Increase setup resolution to full screen                                   ║
║ • Adjusted spacing for boot prompt                                           ║
║                                                                              ║
║ StarBook (bios) and all connected devices may not be usable while updating.  ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Waiting…                 [   -                                   ]
failed to open device for prepare: invalid region name
```

Sean-StarLabs, 2022-07-07:
Okay, so that's being returned from flashrom - I guess the version of the flashrom might be really old and have a different API (version number is flashrom are weird and quite useless).

Manjaro package [here](https://software.manjaro.org/package/flashrom-starlabs) - might be worth trying that or building flashrom from source?

twiztedtbone, 2022-07-07:
> Manjaro package [here](https://software.manjaro.org/package/flashrom-starlabs) - might be worth trying that [...]

Addressing this first for visibility (good news later); this was the version I pointed to in an earlier reply as not existing. The page you linked to exists, but if you follow **those** links to get to the source on Gitlab, you're greeted with an empty page (meaning I couldn't acquire a PKGBUILD). The [only package in Marjaro with starlabs in the name](https://gitlab.manjaro.org/search?utf8=%E2%9C%93&search=starlabs) appears to be fwupd-starlabs (which I currently have installed); [flashrom doesn't seem to exist in any form](https://gitlab.manjaro.org/search?utf8=%E2%9C%93&search=flashrom), unless I'm severely misunderstanding how the Manjaro repos work.

Out of curiosity, could I just be using the latest Git masters for flashrom/fwupd and be good to go? Or does fwupd-starlabs (and flashrom, for that matter) have system-specific patches? Probably a dumb question, but I don't understand this firmware-level stuff very well.

> Okay, so that's being returned from flashrom - I guess the version of the flashrom might be really old and have a different API (version number is flashrom are weird and quite useless).
>
> [...] or building flashrom from source?

And there we go! Building the latest flashrom from Git (the last official release being from over two years ago) did the trick. It's not too often you have to worry about this sort of thing on a rolling distro, but there it is.

Thanks Sean!

closed 2022-07-07 by twiztedtbone

Sean-StarLabs, 2022-07-07:
Awesome :)

Git masters would do the job - we don't do forks, `flashrom-starlabs` is just newer.