StarLabsLtd/firmware #53: Star Lite IV: cannot update to lastest release via fwupd

elagost, 2022-08-05:
Using a Linux Mint 21 Live USB and the instructions here: https://support.starlabs.systems/kb/updates/installing-updates-from-the-lvfs

running `fwupdmgr get-devices` showd in red text (BIOS Locked), and `fupdmgr upgrade` shows a line telling me I can't update due to a locked BIOS. I'm currently on coreboot 4.17. Could be similar to https://github.com/StarLabsLtd/firmware/issues/50 ?

Sean-StarLabs, 2022-08-05:
No, that's a known bug in fwupd. Building the new version for Ubuntu now, should be done by the end of the day. They're already built for Manjaro and Debian.

elagost, 2022-08-05:
Thank you, I appreciate it. I will test sometime this weekend, and if it's still not working, use a debian live usb.

Sean-StarLabs, 2022-08-05:
Packages have been published for focal and jammy: https://launchpad.net/~starlabs/+archive/ubuntu/main/+packages

closed 2022-08-05 by Sean-StarLabs

elagost, 2022-08-05:
Thank you! Updated to 8.05 now.