StarLabsLtd/firmware #54: [StarLite Mk IV] After upgrade to coreboot 8.05 it wont turn off anymore

echo-devnull, 2022-08-21:
I have to keep the power button pressed for ten seconds before it actually turns off completely.

So from gnome: menu, turn computer off, everything goes down and the even the screen is off, not just black, the backlight is off; but the blue led's stay on.

It did not do this before.

Sean-StarLabs, 2022-08-21:
That's something we're looking into - could you let me know which EC you have installed? You can check with:
`cat /sys/class/dmi/id/ec_firmware_release`

echo-devnull, 2022-08-21:
That would be `1.0`

Sean-StarLabs, 2022-08-21:
I'd recommend updating that; the latest is `1.04` and solves the issue for most distros (Arch is the only one known that it doesn't solve it for)

echo-devnull, 2022-08-21:
Ok, but since `fwupdmgr upgrade` shows that everything has the latest updates, how do I update the EC?

Sean-StarLabs, 2022-08-21:
Via the EFI Shell:
https://support.starlabs.systems/kb/updates/installing-updates-with-the-efi-shell
https://github.com/StarLabsLtd/firmware/raw/master/StarLite/MkIV/Nuvoton/1.04/efi-i4.zip

echo-devnull, 2022-08-22:
So I just upgraded the EC to 1.4, unfortunately, the problem persists ;-)

madmo, 2022-08-24:
It seems that the problem is related to a change in linux 5.18.8 (It also happens on fedora):

https://bugzilla.kernel.org/show_bug.cgi?id=216243

echo-devnull, 2022-08-24:
hmmm, I'm on Pop!_os: 

```sh
Linux starlite 5.19.0-76051900-generic #202207312230~1660780566~22.04~9d60db1 SMP PREEMPT_DYNAMIC Thu A x86_64 x86_64 x86_64 GNU/Linux
```

would that still be the case?

madmo, 2022-08-25:
Yes, probably. You could try and build/run a custom kernel with the mentioned patch reverted.

NHellFire, 2022-08-26:
> It seems that the problem is related to a change in linux 5.18.8 (It also happens on fedora):
> 
> https://bugzilla.kernel.org/show_bug.cgi?id=216243

I don't think it is. If I boot mine directly to the EFI shell, `reset -s` should make it shutdown, but it doesn't (lights still on, responds to function lock).

NHellFire, 2022-08-26:
> It seems to be both firmware and kernel (or maybe two different things).
> Kernel 4.17 on something like Ubuntu will shut down correctly, but 4.18
> or later won't.
>
> Regardless, EFI Shell should shut it dow; I'm halfway through a patch
> should fix that, and hopefully, that'll get it working with kernel 4.18.

It doesn't shut down correctly for me on Xubuntu 22.04 with kernel 5.17.

elkoniu, 2022-08-28:
Hi

Just tested this with Fedora 36 on both: Gnome and KDE (spin). My impression is that after initial OS install from USB drive everything was fine. Then, after first packages update using `sudo dnf update` and reboot - this problem started to be presented. I think this may correspond to kernel change from `5.17.5-300.fc36.x86_64` to `5.18.19-200.fc36.x86_64` during system update process.

I also have another observation:
1) I have Fedora 36 running (Gnome)
2) I connect USB-C charger
3) Laptop is charging
4) I trigger system shutdown from Gnome menu
5) Laptop is turning off
6) The first blue LED is still on
7) I disconnect the charger
8) The LED is still on
9) I cant turn on laptop, power button is not working
10) I keep power button pressed for few seconds
11) The LED is off
12) Now power button is working and I can turn laptop on

I switched from coreboot to AMI following https://support.starlabs.systems/kb/guides/installing-coreboot and now everything looks to work fine. I can shut down the laptop, and I cant reproduce issue witch charger anymore.

Sean-StarLabs, 2022-08-31:
`8.09` pushed to testing, which fixes the issue.

NHellFire, 2022-08-31:
8.09 also stays running after shutdown, both when shutting down from UEFI
and Linux 5.17 (Xubuntu 22.04).

Sean-StarLabs, 2022-08-31:
> from UEFI 

Do you mean the EFI Shell? Is EC 1.04 installed?

NHellFire, 2022-08-31:
> > from UEFI
> 
> Do you mean the EFI Shell? Is EC 1.04 installed?

Yes, and yes:

```
$ grep . /sys/class/dmi/id/bios_* /sys/class/dmi/id/ec_firmware_release 
/sys/class/dmi/id/bios_date:08/31/2022
/sys/class/dmi/id/bios_release:4.17
/sys/class/dmi/id/bios_vendor:coreboot
/sys/class/dmi/id/bios_version:8.09
/sys/class/dmi/id/ec_firmware_release:1.4
```

Sean-StarLabs, 2022-08-31:
Hmm, would you be able to dump your CMOS and EC Space?

```
sudo nvramtool -a
sudo ectool -d
```

nvramtool and ectool live in the coreboot repo.

NHellFire, 2022-08-31:
```
$ sudo ectool -d
EC RAM:

00: 01 04 01 00 16 01 00 00 01 01 06 00 00 00 00 00 
10: 00 00 00 00 00 00 dd aa 00 00 00 00 00 00 00 00 
20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 27 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 00 26 27 00 00 00 00 00 00 00 00 00 00 01 01 
80: 02 01 15 20 a0 0f b0 1d a0 0f 00 00 01 14 03 60 
90: 0e 69 20 5c 00 00 00 00 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

$ sudo nvramtool -a
boot_option = Fallback
reboot_counter = 0x0
debug_level = Debug
vtd = Enable
power_profile = Balanced
wireless = Enable
webcam = Enable
legacy_8254_timer = Disable
kbl_timeout = 30 seconds
fn_ctrl_swap = Disable
fast_charge = Normal
power_on_after_fail = Disable
fn_lock_state = 0x0
trackpad_state = 0x0
kbl_brightness = 0x3
kbl_state = 0x1
```

Sean-StarLabs, 2022-09-16:
`8.12` added to lvfs-testing, solves the shutdown problem in EFI Shell and recent kernels

elagost, 2022-09-17:
With EC 1.4 and Coreboot 8.12, I'm able to shut down again. The BIOS was not able to detect a drive created from the instructions to update the EC, so I dumped the contents of the .zip into /boot/efi/ on my main drive.

lelledaniele, 2022-09-19:
Hi, just I wanted to share my experience for who, like me, did not perform a manual firmware update yet.
StarLite
Using PopOS 22.04 with SSD encrypted so I could not detect the coreboot with fwupdmgr.
Even live PopOS could not detect the firmware, also I couldn't add PPAs.
So I tried Ubuntu 22.04 live and and everything was detected.
fwupdmgr was saying I was already on the latest version.
So I did the install with local file option and I downloaded from this repository the 8.12 version for the StarLite.
Rebooted from the live Ubuntu and back to PopOS. Tried to perform a shutdown and everything worked fine.
Amazing, thanks!
Sources that helped me a bit
https://support.starlabs.systems/kb/guides/installing-coreboot
https://support.starlabs.systems/kb/updates/installing-updates-from-the-lvfs

madmo, 2022-09-19:
After updating to 8.12 poweroff works fine on Fedora. Thanks!

Sean-StarLabs, 2022-09-20:
Awesome. `8.13` added to testing to solve the issues with USBs. Will push to stable after a full round of testing :)

closed 2022-09-20 by Sean-StarLabs

NHellFire, 2022-10-11:
Yes, and yes.

Sean-StarLabs, 2022-10-11:
> Yes, and yes.
> […](#)

Have you tried with `8.14`?

NHellFire, 2022-11-08:
@Sean-StarLabs Apologies for the delay on this. Yes, 8.14 shuts down properly from Linux. Thanks for fixing!