StarLabsLtd/firmware #55: [StarBook MkV] Touchpad stopped working

danth, 2022-09-03:
Hello,

The touchpad on my StarBook MkV (Intel version) has mysteriously stopped working.

Using an external mouse works fine and booting into a live USB doesn't help, which gives me the idea that something has gone wrong in the firmware.

I found this in the system journal which looks related:

```
Sep 03 15:57:17 StarBook-MkV kernel: i2c_hid_acpi i2c-STAR0001:00: failed to reset device: -61
Sep 03 15:57:23 StarBook-MkV kernel: i2c_hid_acpi i2c-STAR0001:00: failed to reset device: -61
Sep 03 15:57:29 StarBook-MkV kernel: i2c_hid_acpi i2c-STAR0001:00: failed to reset device: -61
Sep 03 15:57:35 StarBook-MkV kernel: i2c_hid_acpi i2c-STAR0001:00: failed to reset device: -61
Sep 03 15:57:36 StarBook-MkV kernel: i2c_hid_acpi i2c-STAR0001:00: can't add hid device: -61
Sep 03 15:57:36 StarBook-MkV kernel: i2c_hid_acpi: probe of i2c-STAR0001:00 failed with error -61
```

Output of `nvramtool -a`:

```
boot_option = Normal
reboot_counter = 0x0
debug_level = Warning
hyper_threading = Enable
vtd = Enable
power_profile = Balanced
me_state = Disable
me_state_counter = 0x0
wireless = Enable
webcam = Enable
microphone = Enable
legacy_8254_timer = Disable
thunderbolt = Disable
kbl_timeout = 30 seconds
fn_ctrl_swap = Disable
max_charge = 80%
fan_mode = Normal
power_on_after_fail = Disable
fn_lock_state = 0x0
trackpad_state = 0x0
kbl_brightness = 0x2
kbl_state = 0x1
```

danth, 2022-09-03:
Sorry, it's working now!

I'd rebooted multiple times to no avail - turns out leaving the laptop fully powered off for a while fixed it.

It'd be interesting to know what caused this though...

closed 2022-09-03 by danth

Sean-StarLabs, 2022-09-05:
It's a kernel bug - I'm not sure and the ins and outs of it, but is fixed upstream.