StarLabsLtd/firmware #57: [StarLite Mk IV] Volume keys don't work on Ubuntu

CommonLoon102, 2022-12-10:
Fn+F1, Fn+F2, Fn+F3 are not working. Fn+F8 and Fn+F9 are working fine though.
OS: Ubuntu 22.04.1, with dist-upgrade ran

CommonLoon102, 2022-12-10:
OK, I have figured out that something is mixed up. There are three blue LEDs above the keyboard. About the first one I have no idea what it is supposed to indicate. The second is for the caps lock and the third one is for the "lock". There is a "lock" button on the keyboard. I don't know what it locks but when it is "locked" then I can use the F1-F12 keys normally, like when I press F1 then it is pressing F1, usually opens help. If the "lock" is off, then the volume keys are working.

When I press the "lock", sometimes the 3rd LED changes instantly, sometimes I have to wait 200ms.

So here are some combinations while using gedit.
Lock is ON:
F1-F12: keys are working as they should on any computer.

Lock is OFF:
Keys are doing what it is painted onto them (volume, brightness, etc.)

Lock is ON:
F1 = Opens help
Fn+F1 = Mute/Unmute
F9 = Opens document list sidebar
Fn+F9 = Increasing brightness

Lock is OFF:
Fn+F1 = Opens help
F1: Mute/Unmute
Fn+F9 = Increasing brightness (bug: should open document sidebar instead)
F9 = Increasing brightness

Fn+F9 should open the document list sidebar in gedit instead, like sending a normal F9 button event.

I can't get my head around this. The behaviour is not consistent. Why there is a "lock" button at all? Would be much simpler if F1-F12 buttons would just send F1-F12. Except if you hold down the Fn key, then it would do what is painted on them (volume, brightness, etc.). Instead of the "lock" key, there could be something more useful. "Insert" for example. If someone has only 1 hand without a thumb, then the "lock" button makes sense of course. When the "lock" is on, it should behave like the "Fn" key is constantly pushed. But now it locks the Fn when it is turned OFF. Why?

Sean-StarLabs, 2022-12-12:
LEDs left to right: Power Indicator, Caps Lock, Function Lock.
Function Lock locks the function keys on, not the media keys.
`Fn` + `Del` will perform an insert.

There are only 2 scan codes in the firmware, so if the behaviour of some keys changes beyond the function key, it's not down to the firmware. 

>  Except if you hold down the Fn key, then it would do what is painted on them (volume, brightness, etc.). 
Just leave function lock on, and don't use the lock key

closed 2022-12-12 by CommonLoon102