StarLabsLtd/firmware #58: StatLite III coreboot

ams-123, 2022-12-15:
Hi,

I'm attempting to switch the StarLite III to coreboot, but running into a similar issue as has been reported here:

`sudo fwupdmgr switch-branch` 
`No supported devices`

I've been through the previous issue thread, but am unable to get this working on the StarLite III. I'm running Fedora 27 and have built flashrom from source, so am now able to see the following:

`sudo flashrom -p internal`

flashrom v1.2-1127-g77fe2663 on Linux 6.0.12-300.fc37.x86_64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
No DMI table found.
Found chipset "Intel Gemini Lake".
Enabling flash write... SPI Configuration is locked down.
FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG1: BIOS region (0x00001000-0x0067efff) is read-write.
FREG5: Device Expansion region (0x0067f000-0x0077ffff) is locked.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
Enabling hardware sequencing because some important opcode is locked.
OK.
Found Programmer flash chip "Opaque flash chip" (8192 kB, Programmer-specific) on internal.
No operations were specified.

At this point in the previous issue, switch-branch looked to work, but on this laptop I still see No supported devices.

Another command for debug, again from the previous issue:

`sudo fwupdtool get-devices --verbose --plugins flashrom --allow-branch-switch`

_snip as lots of output_
06:10:20.650 FuEngine             enabling flashrom due to HwId d5521faa-c50b-5d64-971d-8fd400030c51
06:10:20.653 FuEngine             enabling flashrom due to HwId d5521faa-c50b-5d64-971d-8fd400030c51
Loading…                 [***                                    ]06:10:20.655 FuContext            battery threshold now 30
Loading…                 [ -                                     ]06:10:20.655 FuPlugin             startup(flashrom)
06:10:20.660 FuPluginFlashrom     No DMI table found.
06:10:20.662 FuPluginFlashrom     WARNING: No chipset found. Flash detection will most likely fail.
06:10:20.662 FuPluginFlashrom     ========================================================================
06:10:20.662 FuPluginFlashrom     You may be running flashrom on an unknown laptop. We could not
detect this for sure because your vendor has not set up the SMBIOS
tables correctly. Some internal buses have been disabled for
safety reasons. You can enforce using all buses by adding
  -p internal:laptop=this_is_not_a_laptop
to the command line, but please read the following warning if you
are not sure.

Not sure how to proceed from here. As others have done, I have tried a couple of live distro's, including fedora rawhide which should be bleeding edge for latest software. No luck.

Sean-StarLabs, 2022-12-15:
That all looks good; did you enable the testing remote? coreboot for the Lite Mk III is in beta (S3 doesn't work) so it's only available there.

ams-123, 2022-12-15:
Hi Sean, thanks - how do I enable testing remote?

Sean-StarLabs, 2022-12-15:
`fwupdmgr enable-remote lvfs-testing`

Just FYI, you can't switch back to AMI without a hardware programmer.

ams-123, 2022-12-15:
Thanks. Have enabled the testing remote, but still getting the same result:

[ams@fedora ~]$ fwupdmgr refresh --force
Updating lvfs-testing
Downloading…             [***************************************]
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Successfully downloaded new metadata: 3 local devices supported
[ams@fedora ~]$ sudo fwupdmgr switch-branch 
No supported devices

Sean-StarLabs, 2022-12-15:
Can you post the output of:
```
fwupdmgr refresh --force
fwupdmgr get-releases --verbose --plugins flashrom --allow-branch-switch

```

Sean-StarLabs, 2022-12-15:
Sorry, should be:
`fwupdtool get-releases --verbose --plugins flashrom --allow-branch-switch`

ams-123, 2022-12-15:
```
[ams@fedora ~]$ sudo fwupdtool get-releases --verbose --plugins flashrom --allow-branch-switch
09:58:04.074 FuDebug              Verbose debugging enabled (on console 1)
Command not found

Use fwupdtool --help for help
```

Sean-StarLabs, 2022-12-15:
It's probably in `/usr/bin`

ams-123, 2022-12-15:
I'm not sure what it's not finding:

```
[ams@fedora ~]$ which flashrom
/usr/local/sbin/flashrom
[ams@fedora ~]$ sudo fwupdtool get-releases --verbose --plugins /usr/local/sbin/flashrom --allow-branch-switch
10:00:34.290 FuDebug              Verbose debugging enabled (on console 1)
Command not found

Use fwupdtool --help for help
```

Sean-StarLabs, 2022-12-15:
Okay, just use:
`sudo fwupdmgr get-releases --verbose --allow-branch-switch`

ams-123, 2022-12-15:
```[ams@fedora ~]$ sudo fwupdmgr get-releases --verbose --allow-branch-switch
(fwupdmgr:8820): GLib-DEBUG: 18:05:17.828: setenv()/putenv() are not thread-safe and should not be used after threads are created
(pkttyagent:8822): GLib-GIO-DEBUG: 18:05:17.837: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:8820): GLib-GIO-DEBUG: 18:05:17.865: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:8820): dconf-DEBUG: 18:05:17.865: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:8820): dconf-DEBUG: 18:05:17.866: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:8820): dconf-DEBUG: 18:05:17.867: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:8820): dconf-DEBUG: 18:05:17.867: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:8820): dconf-DEBUG: 18:05:17.867: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:8820): dconf-DEBUG: 18:05:17.867: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:8820): dconf-DEBUG: 18:05:17.867: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:8820): dconf-DEBUG: 18:05:17.867: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:8820): dconf-DEBUG: 18:05:17.868: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:8820): dconf-DEBUG: 18:05:17.868: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:8820): GLib-GIO-DEBUG: 18:05:17.873: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:8820): GLib-GIO-DEBUG: 18:05:17.889: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:8820): GLib-GIO-DEBUG: 18:05:17.890: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:8820): dconf-DEBUG: 18:05:17.904: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:8820): dconf-DEBUG: 18:05:17.905: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:8820): dconf-DEBUG: 18:05:17.906: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:8820): dconf-DEBUG: 18:05:17.907: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:8820): dconf-DEBUG: 18:05:17.908: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:8820): Fwupd-DEBUG: 18:05:17.914: Emitting ::status-changed() [idle]
Choose a device:
0.	Cancel
1.	3ecf3fbe07ed8933226b294c8c19f4f9c8ebd71a (SuperIO FuSuperioIt89Device)
2.	a45df35ac0e948ee180fe216a5f703f32dda163f (System Firmware)
3.	362301da643102b9f38477387e2193e57abaa590 (UEFI dbx)
2
  AppstreamId:          com.starlabs.I3.AMI.capsule
  ReleaseId:            10126
  RemoteId:             lvfs
  Summary:              AMI firmware for StarLite MkIII
  Description:          <p>This stable release includes the following features:</p><ul><li>Updated SGX allocations</li><li>Updated Microcode</li><li>Improved Charger configuration</li><li>Updated SGX configuration</li><li>Improved DPTF configuration</li><li>Enabled dito</li><li>Updated CSE</li></ul>
  Version:              2.1.0
  Filename:             uefi-I3.cab
  Protocol:             org.uefi.capsule
  Categories:           X-Device
  Checksum:             SHA1(423796dce4876ba83545088cddf2682378c9772e)
  Checksum:             SHA256(3c8f0f1d9d6627d0d05cfc82ee0a01fa3cbefff43bc4082c3383297751123705)
  License:              LicenseRef-proprietary
  Size:                 8.4 MB
  Created:              2021-09-22
  Uri:                  https://fwupd.org/downloads/3c8f0f1d9d6627d0d05cfc82ee0a01fa3cbefff43bc4082c3383297751123705-uefi-I3.cab
  Homepage:             https://starlabs.systems
  Urgency:              medium
  Vendor:               Star Labs
  Flags:                none

[ams@fedora ~]$ sudo fwupdmgr switch-branch 
No supported devices
```

Is option 2 the correct option?

Sean-StarLabs, 2022-12-15:
No, you're missing the correct option. What version of `fwupd` are you using?

ams-123, 2022-12-17:
After this, I booted back into rawhide to use what should be a more recent fwupd, but still no luck. As it happens, Fedora 37 updated fwupd/flashrom over the weekend - the current version is:

[ams@fedora ~]$ rpm -qa | grep fwupd
fwupd-efi-1.3-1.fc37.x86_64
fwupd-plugin-flashrom-1.8.8-1.fc37.x86_64
fwupd-plugin-modem-manager-1.8.8-1.fc37.x86_64
fwupd-plugin-uefi-capsule-data-1.8.8-1.fc37.x86_64
fwupd-1.8.8-1.fc37.x86_64

I still get the exact same result.

Is there a live CD that is known to have all dependencies to update the Lite III to coreboot?

Sean-StarLabs, 2022-12-19:
No - but on Ubuntu you can run:
```
sudo add-apt-repository ppa:starlabs/main
sudo add-apt-repository universe
sudo apt update
sudo apt install fwupd libflashrom1
```

Sean-StarLabs, 2023-02-03:
New build of flashrom in the copr which will do the trick :)

closed 2023-02-08 by Sean-StarLabs