StarLabsLtd/firmware #59: suspend resume on starbook

netmonk, 2023-01-19:
received my starbook yesterday, updated firwmare to 8.22, running ubuntu 22.04 preinstalled. 

Suspend/Resume is not working at all, each time i close the lid and reopen it few time later i have to power down and power up again while laptop rebooting from 0. 

Is there any solutions for this issue ?

Sean-StarLabs, 2023-01-19:
Interesting - what happens when you open the lid? Blank screen?

Can you let me know your kernel version (`uname -r`)?

netmonk, 2023-01-19:
plonky@nano:~$ uname -r 
5.15.0-58-generic

the backlight of the keyboard switch on, but screen stays blank, nothing happens.

netmonk, 2023-01-24:
Good evening, I just installed 8.23 firmware, it doesnt seems to fix the issue, still need to force shutdown after suspend to have the screen powering on and system booting for zero again. 
The Resume step is not working (may be cause the suspend step is in failure i dont know )

Sean-StarLabs, 2023-01-25:
Hmm, are you able to switch to the AMI firmware and see if you can reproduce it there? I haven't been able to replicate it on `8.23`

netmonk, 2023-01-25:
where can i find th AMI firmware ?

Sean-StarLabs, 2023-01-27:
https://support.starlabs.systems/kb/guides/installing-coreboot

Sean-StarLabs, 2023-02-03:
Can no longer replicate with e93a096e9da512371e8c5eec8ff03884487aabfa, will leave testing over the weekend and then push

netmonk, 2023-02-05:
well the problem is not replicable after endeavouros installation. I have another laptop with ubuntu and suffer same issue. So it' more tied to ubuntu and kernel than hardware firmware from my perspective. 

Also, can i use upstream flashrom to flash firmware or do i need to use starlabs fork ?

commit b50523c3 (2023-02-06): Added StarBook MkVI-Intel coreboot 8.27

Fixes #70
Fixes #59
Fixed #63
Fixes #60
closed 2023-02-06 by Sean-StarLabs