StarLabsLtd/firmware #62: [StarBook MK VI] Adjust fan curve

Sean-StarLabs, 2023-01-23:
Multiple users reported that the fan starts too early.

Need to port the fan curve from L4

3lpsy, 2023-02-19:
It seems like this may have been fixed recently. It doesn't seem like lm_sensors/fancontrol is supported (no sensors detected via sensors-detect). Can you elaborate on whether there are any utilities for monitoring / controlling fan speed at runtime? Thanks!

cichy1173, 2023-02-19:
> It seems like this may have been fixed recently

@3lpsy I can be wrong, but only AMD version received update that changed fan curve.

3lpsy, 2023-02-20:
@cichy1173 Explains why the issue isn't closed. But yeah, I am using the AMD variant.

garybowers, 2023-02-21:
I've noticed since updating to 1.2.0 with the new fan curves, the system under heavy load doesn't ramp the fans up so throttles a lot.  Under heavy load but not 100% e.g. running a VM with just a few cores - the system lags a LOT.

OS: Arch / Gnome tried reinstalling - I may downgrade the bios to test the theory.

cichy1173, 2023-02-21:
> I've noticed since updating to 1.2.0 with the new fan curves, the system under heavy load doesn't ramp the fans up so throttles a lot. Under heavy load but not 100% e.g. running a VM with just a few [...]

1.3.0 is available. Try new version - maybe it is better :-).

garybowers, 2023-02-21:
These guys work too fast :-P going to try that now.

Sean-StarLabs, 2023-02-21:
> I've noticed since updating to 1.2.0 with the new fan curves, the system under heavy load doesn't ramp the fans up so throttles a lot. Under heavy load but not 100% e.g. running a VM with just a few [...]

Nothing in 1.3.0 that'll change the fan, but which curve were you using?

Sean-StarLabs, 2023-02-22:
Sorry, do you mean at all?

garybowers, 2023-02-22:
I've not had the fans ramp up to 1.1.0 levels :) It's REALLY quiet now.

garybowers, 2023-02-23:
So I'm also struggling to monitor the fanspeed (it's missing in acpi), I did run a watch on /proc/cpuinfo to try and capture the issue.... When under heavy load the cpu is clocking down to 399mhz and then going up and back again,  when the heavy load goes I can see the CPU clocking correctly.

[image]
[image]
[image]
[image]

Sean-StarLabs, 2023-02-23:
I think it's the multiplier not being set correctly on the aggressive profile; I'll adjust that in the next release but normal or quiet might actually be better for now.

garybowers, 2023-02-24:
Normal does indeed feel better, I've noticed it still happening but far less.

garybowers, 2023-02-24:
[image]

esimopoulo, 2023-02-26:
Can also confirm that super simple VM stuff like clicking on a Next button or scrolling through a drop-down list completely choke the system for a few seconds. Plenty of unused RAM both in host & VM, that's not an issue.

Pressing the Super key and trying to open up System Monitor (just to understand what is going on) renders the transition effect in 4 FPS, as if the GPU is struggling too.

Have tried Gnome Boxes, VirtualBox and VMWare Workstation, all with similar behaviour.

Setup: StarBook MK VI AMD, 16GB RAM, Fedora 37, Bios 1.2.0 (Normal Profile)

garybowers, 2023-03-01:
Is there any update on this? VM's are basically unusable at the moment on this machine.

Sean-StarLabs, 2023-03-01:
Not yet - don't hold me to it, but aiming for fixing #62 #66 #75 by the end of the week.

You can downgrade it via the EFI Shell in the meantime.

garybowers, 2023-03-01:
awesome thanks Sean!

garybowers, 2023-03-06:
v1.1.0 seems to be missing the .bin file, i see a .cap file - had a look to see if this can be converted but not a bios expert :)

Sean-StarLabs, 2023-03-06:
Fixed the zip file

garybowers, 2023-03-06:
Yeah that's bricked my laptop :(

cichy1173, 2023-03-06:
> Not yet - don't hold me to it, but aiming for fixing #62 #66 #75 by the end of the week.
> 
> You can downgrade it via the EFI Shell in the meantime.

Will the update also include changes for Intel-Coreboot?

garybowers, 2023-03-17:
Any further updates? I appreciate it's a complicated issue :)

Sean-StarLabs, 2023-03-17:
Being tested at the moment

garybowers, 2023-04-05:
Any further updates on this? the laptop is unable to be used for anything cpu hungry compiling / docker / vm's all choke the CPU, downgrading was causing the machine to be bricked.

garybowers, 2023-04-06:
Working PERFECTLY in my testing, Agressive v's Normal seems to work really well, both don't show any signs of thermal throttling under heavy load ... Normal hovers around 90degC at full pelt and agressive sits around 80 and the fans ramp up but not too noisy!

Thanks so much.

garybowers, 2023-04-06:
> Huh - I thought you were AMD? I haven't uploaded it yet!

Yep and I updated the ite firmware published 6hrs ago

Sean-StarLabs, 2023-04-06:
Of course (it's been a long week!). Cool

Croissong, 2023-04-10:
Hey,

I just got my StarBook :)
To be honest, it's not clear to me what the current fan curve is supposed to look like on the latest `MkVI-Intel coreboot 8.36` firmware.
Are these values still valid?
```c++
const struct fan_curve code quiet_mode = {
	//				1	2	3	4	5	6	7	8	9	10	11
	/* .start_temp	= */	{	0,	55,	60,	65,	70,	75,	80,	85,	90,	95,	100},
	/* .duty	= */	{	0,	0,	0,	20,	25,	30,	35,	40,	50,	60,	70}
};
```

I'm asking because for me, on `fan_mode = Quiet` the fan still regularly starts during light usage, at `coretemp`s < 50C°.

Sean-StarLabs, 2023-04-10:
> Could you maybe provide some insights whether these values are still valid?

They are, but they're not in coreboot. Any EC version later than `1.03` uses them.

`sensors` will use the Package Temp, which is pretty much just the highest temp of a core. The EC's driver is a weighted average, so it won't match entirely.

Sean-StarLabs, 2023-04-21:
Making a solved with EC `1.12` for the Intel variant and `2.0.0` (bundled in AMI `1.7.0`)

closed 2023-04-21 by Sean-StarLabs

cichy1173, 2023-04-21:
@Sean-StarLabs Does Coreboot 8.40 bring EC 1.12? Is EC 1.12 stable?

cichy1173, 2023-04-23:
> EC 1.12 wasn't included in 8.40. Sean will always list it as included in the release notes if it includes an EC update. I too am waiting for the fixed fan curve

That's what I guessed, but I saw that the issue was closed, so I preferred to ask. 

Also, I'm all for releasing EC 1.02 (or under another name) with just the new fan curve. This is the feature I've been waiting for the longest, as the noisy fan is a bit annoying, especially when I'm in public places like a cafeteria or a university.

Sean-StarLabs, 2023-04-23:
> Does Coreboot 8.40 bring EC 1.12? Is EC 1.12 stable?

`8.40` contains EC `1.12` but the mirror flag isn't enabled (it wont auto install). It has the PD update in it so that will only work for people us DC chargers.

We're considering `1.12` stable; we'll soon have a mechanism to push these out via the LVFS, but for now, the EFI Shell is the only option. EC `1.11` is likely also good; the only difference is that EC `1.11` supports a higher charging rate. We had one issue reported with it, but that's assumed to be a problem with the charger. We'll be testing the higher charging rate en-masse soon though.

> Or am I being too naive and not really understanding the complexities? 

The reason that this took a while is that on Intel platforms, the fan is tied to a bunch of other things. I quickly pushed an update out for the AMD ones, as that's a lot simpler - one temperature reading. With the Intel ones, it's now got a new PECI driver as before; the driver worked similarly to `lm-sensors` in Linux, where it would report the temperature as the highest that any core is. It now knows the temperature of each individual core, so it will account for scenarios like 1 little core being at 80 degrees. (Also, rewriting it as I go so it'll work with SDCC, so we can published the source code. It's a slow process as SDCC has ... quirks!)

hkdb, 2023-04-24:
@Sean-StarLabs Thanks for the clarification and the hard work on the firmware! Off I go to try to update via EFI shell.

@cichy1173 Proves your theory right. Good to ask! ;)

hkdb, 2023-04-24:
Just wanted to report back to say that the update went through seamlessly. When the fans are spun up from higher loads, there's still a little bit of the high pitch noise but in general, it's a massive improvement. No more sly comments from my co-workers about how "my machine is taking off!". ;)  Thanks again @Sean-StarLabs !

hkdb, 2023-04-24:
@Sean-StarLabs  Hum... I think I spoke too soon. So I closed my laptop to let it go to sleep when I went out for dinner. When I came back, I wrote the above comment while letting one of my VMs update and it all of a sudden froze. So I turned it off by holding the power button and then tried to turn it back on again. Then it sat in the Starlabs logo for much longer than usual and then goes to EFI Shell. I tried flashing 1.01 back on via EFI shell but still the same. So I guess my machine is in a bricked state. Any advice?

Sean-StarLabs, 2023-04-24:
That's not related to the EC update (the EC doesn't even know what a file system is).

> I assume the reason why it's not booting anymore is because of coreboot not seeing the stardrive?

edk2 will just search any known filesystems for EFI loaders, usually, that's a FAT32 partition with `grubx64.efi`. If there's a problem with the data on the drive, it won't be able to find anything. 

> LUKS is enabled

LUKS is only a factor for installing EC updates, the tool that performs the update can't decrypt a LUKS Volume so can't perform an update.

hkdb, 2023-04-24:
I am thinking to use live usb of the starlabs provided ubuntu to downgrade coreboot to see if that helps.

Sean-StarLabs, 2023-04-24:
> But it seems weird that boot manager doesn't see the stardrive as a boot option at all doesn't it?

No, if there's no FAT32 partition (you said it was one big unknown volume), it wont. Different versions of coreboot, or even AMI, wont change that (they work in exactly the same way).

hkdb, 2023-04-24:
Hum... then I am not sure what happened cause everything was working fine all along.... Hope it's not that the stardrive is dying or something along the lines of that? Should we maybe switch to support to take a deeper look into this? Thx!

Sean-StarLabs, 2023-04-24:
Usually, when SSDs fail, they disappear - if it's still seeing a partition, it sounds software related. If you want to recover it, `testdisk` will likely be the best bet. Feel free to ping support if you want though

hkdb, 2023-04-24:
@Sean-StarLabs ,

Well, I just booted with the starlabs provided Ubuntu iso on usb and the drive is not showing there. In dmesg, it shows the following error:

`nvme nvme0: Device not ready; aborting initialisation, CSTS=0x0`

Does that sound like something you are familiar with? Again, I am happy to move this conversation to support instead if more appropriate. I already left a message with support.

Thanks!