StarLabsLtd/firmware #65: [Starbook V/VI - AMD] Can't Install Coreboot / No Supported Device

3lpsy, 2023-01-29:
## Background

I ordered a Starbook MK V last may so I'm not sure if I own a V or VI (it has a fingerprint reader if that helps). While I ordered it with Coreboot, it came with American Megatrends. 

### Other Info
CPU: AMD Ryzen 7 5800U with Radeon Graphics
OS: Arch Linux

## First Attempt
When attempting to switch branches, I first performed a BIOS update and rebooted. Then when running `switch-branch`, i received a No Supported Devices message. I am running Arch Linux with the standard `linux` kernel (I had different issues under `linux-hardened` but I believe those are expected). I first attempted to use the Manjaro `fwupd-starlabs` and f`lashrom-starlabs` packages. I've included the details below

### Arch Linux + Standard Kernel + Starlabs fwupdmgr + Starlabs Flashrom
```
$ sudo fwupdmgr switch-branch
No supported devices

$ sudo flashrom -p internal
flashrom unknown on Linux 6.1.8-arch1-1 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
No DMI table found.
Found chipset "AMD FP4".
Enabling flash write... FCH device found but SMBus revision 0x51 does not match known values.
Please report this to flashrom@flashrom.org and include this log and
the output of lspci -nnvx, thanks!.
Could not determine chipset generation.PROBLEMS, continuing anyway
========================================================================
You may be running flashrom on an unknown laptop. We could not
detect this for sure because your vendor has not set up the SMBIOS
tables correctly. Some internal buses have been disabled for
safety reasons. You can enforce using all buses by adding
  -p internal:laptop=this_is_not_a_laptop
to the command line, but please read the following warning if you
are not sure.

Laptops, notebooks and netbooks are difficult to support and we
recommend to use the vendor flashing utility. The embedded controller
(EC) in these machines often interacts badly with flashing.
See the manpage and https://flashrom.org/Laptops for details.

If flash is shared with the EC, erase is guaranteed to brick your laptop
and write may brick your laptop.
Read and probe may irritate your EC and cause fan failure, backlight
failure and sudden poweroff.
You have been warned.
========================================================================
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.
```

## Second Attempt
I also installed the `flashrom-git` package from the Arch AUR and attempted to switch to coreboot.

### Arch Linux + Standard Kernel + Starlabs fwupdmgr + flashrom-git
```
$ sudo fwupdmgr switch-branch
No supported devices

$ sudo flashrom -p internal
flashrom unknown on Linux 6.1.8-arch1-1 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
No DMI table found.
Found chipset "AMD FP4".
Enabling flash write... FCH device found but SMBus revision 0x51 does not match known values.
Please report this to flashrom@flashrom.org and include this log and
the output of lspci -nnvx, thanks!.
Could not determine chipset generation.PROBLEMS, continuing anyway
========================================================================
You may be running flashrom on an unknown laptop. We could not
detect this for sure because your vendor has not set up the SMBIOS
tables correctly. Some internal buses have been disabled for
safety reasons. You can enforce using all buses by adding
  -p internal:laptop=this_is_not_a_laptop
to the command line, but please read the following warning if you
are not sure.

Laptops, notebooks and netbooks are difficult to support and we
recommend to use the vendor flashing utility. The embedded controller
(EC) in these machines often interacts badly with flashing.
See the manpage and https://flashrom.org/Laptops for details.

If flash is shared with the EC, erase is guaranteed to brick your laptop
and write may brick your laptop.
Read and probe may irritate your EC and cause fan failure, backlight
failure and sudden poweroff.
You have been warned.
========================================================================
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.
```

## Third Attempt
Finally, I installed Manjaro on a USB device, booted into it and tried performing the same action. I of course used the starlabs fwupd and flashrom packages. I don't have the raw results for the flashrom command, but the switch-branch command had the same output. 

Is Coreboot supported on the AMD variant? And if possible, can you please advise me on what I'm missing

rmsthebest, 2023-01-31:
They are waiting for an FSP fix from AMD before they upload coreboot for the AMD version. Support has said everyone who bought the coreboot amd laptops will be notified when it is available.

Imho it is strange that each customer has to discover for themselves that they didnt receive what they purchased, but it is what it is I guess. To be clear, I'm glad they sent them even if they havent got coreboot working yet, but the delivery receipt should not have said that it contained a laptop with coreboot.

3lpsy, 2023-01-31:
Yeah, I'm just happy to have my laptop. Fine being patient and glad support will eventually get added :)

neeshy, 2023-06-07:
Any news on this? It's been over 5 months. Have you guys received word from AMD about that FSP fix? On our end, is it currently possible to build coreboot from source and flash it manually w/ flashrom?

Sean-StarLabs, 2023-06-07:
Yes, still waiting for a new build though. The latest emails hint at getting something soon.

> On our end, is it currently possible to build coreboot from source and flash it manually w/ flashrom?

You can build and flash anything you like; it wouldn't circumvent the problem and would introduce others that exist in the publicly available blobs.

> As far as I can tell, only Intel support for the StarBooks has been merged into upstream coreboot. Likewise, I only see Intel related commits on StarLab's coreboot fork.

Patches won't be merged until the problem is fixed.

neeshy, 2023-06-07:
> The latest emails hint at getting something soon.

That sounds promising :)

sysnux, 2023-07-19:
> Yes, still waiting for a new build though. The latest emails hint at getting something soon.

Any update?

Sean-StarLabs, 2023-07-21:
It's basically being held up by internal AMD stuff (NDA's prevent too much detail there) - I'll give them another nudge!

sysnux, 2023-09-21:
Still waiting for the coreboot I ordered more than one year ago...  fortunately, the StarBook AMD arrived sooner ;)

flokaiser, 2023-11-03:
Do you have any update?

flokaiser, 2023-11-30:
I think we were ghosted by a company :(

Sean-StarLabs, 2023-11-30:
Sorry, not sure how I missed that - nothing yet. I'll give them another nudge.

cyb3rmonk, 2024-01-23:
@Sean-StarLabs did you get an answer?

Sean-StarLabs, 2024-01-23:
Yeah, so, paraphrasing - official channels are proving useless, so the engineer is going to try to reach someone from the PSP team directly. And, to be fair, that's how things get done with Intel - the only difference is "we" know the Intel devs.

> Also, I hope you guys review this mess internally. 

We're not doing any more AMDs until they fix it (for consumers).

>  I understand you depend on AMD, but if that's the case then why advertise coreboot compatibility for a setup before you get everything you need from them?

Because the code is 99% there. The way we usually do it, make the AMI firmware as it's way, way more time consuming (code base is about 20x the size of coreboot), and then do coreboot with essentially the same code. The same happened here, and it was a small details than turned into a big blocker. And, getting zero support from AMD honestly wasn't anticipated. 

> At some point you have to give up on the problem and embrace the fact that you have dozens of new friends

I like the idea, but no, I don't like giving up. AMD are slowly making progress with their open source code, and there is one other thread I'm pulling on at the moment - long shot, but worth a try.

Either way, it'll happen - the variable is when.

hicaru, 2024-02-22:
> Yeah, so, paraphrasing - official channels are proving useless, so the engineer is going to try to reach someone from the PSP team directly. And, to be fair, that's how things get done with Intel - [...]

[image]

You don't need to lie! You still have them on sale!

Sean-StarLabs, 2024-02-22:
> You don't need to lie! You still have them on sale!

No, sorry, no idea how you got that from what I said. To clarify, we're not going to stop selling the ones we've already made because 100K of code doesn't have the license we want!

hicaru, 2024-02-22:
> > You don't need to lie! You still have them on sale!
> 
> No, sorry, no idea how you got that from what I said. To clarify, we're not going to stop selling the ones we've already made because 100K of code doesn't have the license we want!

Could you publish it? We will build and flash it directly to chip.

Sean-StarLabs, 2024-02-22:
Publish the coreboot patch? It's in Gerrit, but the problem isn't with building or flashing it - it's with the PSP not allowing the Ap base to be set, which stops FSP from running.

hicaru, 2024-02-22:
> Publish the coreboot patch? It's in Gerrit, but the problem isn't with building or flashing it - it's with the PSP not allowing the Ap base to be set, which stops FSP from running.

have we any chance to have `coreboot` then in near time?

sjlongland, 2024-02-22:
> > Publish the coreboot patch? It's in Gerrit, but the problem isn't with building or flashing it - it's with the PSP not allowing the Ap base to be set, which stops FSP from running.
> 
> have we any chance to have `coreboot` then in near time?

It's frustrating, but this is one of these situations where we're all going to have to be patient.  Good things come to those who wait and all that.

AMD's a big company, likely there's a lot of layers of bureaucracy that need to be worked around… and this takes time.  Starlabs is a UK-based company if I'm not mistaken, and AMD US-based, so time-zones further extend how long these negotiations will take.

Other equipment has been made in the past running CoreBoot on AMD hardware, so I don't think the issue is necessarily impossible to fix technically.

It'd be nice if the website had flagged that AMD + Coreboot was still being worked on (with no hard ETA known) and that the product would ship with AMI UEFI instead, but I am certain it'll happen eventually.  I'm fine waiting as this machine has been a much better out-of-box experience running Linux than most machines with similar-age hardware.  The machine still boots running UEFI firmware, and once `grub` loads the kernel and hands over, UEFI quickly becomes irrelevant anyway.

Let's give them the time they need before stomping our feet. :-)

hicaru, 2024-02-22:
> It's frustrating, but this is one of these situations where we're all going to have to be patient. Good things come to those who wait and all that.
>
> AMD's a big company, likely there's a lot of [...]

have to sit with a AMD backdoor and wait for until someone reverse it or official AMD answer.

Sean-StarLabs, 2024-02-22:
> AMD backdoor

What do you mean?

hicaru, 2024-02-22:
> > AMD backdoor
> 
> What do you mean?

i mean: AMD PSP  is a backdoor like Intel ME

Sean-StarLabs, 2024-02-22:
Okay. coreboot doesn't do anything different with the PSP than AMI does.

sjlongland, 2024-02-22:
> > > AMD backdoor
> > 
> > What do you mean?
> 
> i mean: AMD PSP is a backdoor like Intel ME

Unless you've got the equipment to non-destructively inspect every integrated circuit on your computer, I'd wager that any computer you buy or make yourself could theoretically have a backdoor you don't know about or control.

The only way around that would be to use such low-level building blocks (TTL chips, CPLDs, maybe FPGAs), that making a practical backdoor becomes impossible.  Modern off-the-shelf x86 is not for you.

If that's not appealing, then we have to accept that we're not going to "see" everything that goes on inside every IC, and some of them have "features" we may not like.  It's a package deal.  That's life.

hicaru, 2024-02-23:
> Unless you've got the equipment to non-destructively inspect every integrated circuit on your computer, I'd wager that any computer you buy or make yourself could theoretically have a backdoor you [...]

To my deepest regret, I agree. There are indeed some open-source chips like RISC-V that are sufficiently open to be utilized. However, they are too weak for everyday use. Additionally, there is Elbrus, but it is overly proprietary and likely contains backdoors.

sjlongland, 2024-02-23:
> To my deepest regret, I agree. There are indeed some open-source chips like RISC-V that are sufficiently open to be utilized.

Even if they were, you don't know that the specimen you hold in your hand, is not backdoored for sure without (possibly destructive) inspection.

Something like a logic gate package: it's so generic it'd be almost impossible to practically back-door it.
A CPLD or FPGA: could be more easily back-doored, but again, you'd maybe spot it in an X-ray photo since it'd be a "different" looking block inside a very regular and repetitive grid of blocks that have a very generic function and behaviour.
An ASIC like an off-the-shelf CPU (any architecture you like): all bets are off.

If you want fast: go x86, go modern ARM… maybe look at RISC-V → accept there may be backdoors.
If you want back-door free: get out your breadboards and start wiring TTL chips together → accept 8-bit or 16-bit CPUs with clock speeds in the kilohertz.

rmsthebest, 2024-02-24:
Please stay on topic, I want to stay subscribed to this issue so I get notified when coreboot is available.

hicaru, 2024-03-10:
> Okay. coreboot doesn't do anything different with the PSP than AMI does.

These are just words, and you've already misled me once. How am I supposed to believe in this now?  Т_Т

Sean-StarLabs, 2024-03-14:
> How am I supposed to believe in this now?

Do you own research, PSP runs before the bootloader - AMD don't try and hide that

hicaru, 2024-03-15:
> > How am I supposed to believe in this now?
> 
> Do you own research, PSP runs before the bootloader - AMD don't try and hide that

Friend, I understand all of this, and my point is really about something else. You could have mentioned that you support Coreboot so that people know what they're buying. My case is a prime example. I purchased something that didn't meet my expectations, and part of the mistake is mine—I could have checked the repository myself to see all the supported models! So, I'm heartbroken and left with no choice but to wait.

hicaru, 2024-04-08:
Hello team. i see you still have there wrong info! please remove it! people will buy your laptop and get no coreboot! 
[image]

Sean-StarLabs, 2024-04-08:
"No firmware is available for this device." - you don't think that covers it?

hicaru, 2024-04-08:
> "No firmware is available for this device." - you don't think that covers it?

ahh so i can download it and install directly SPI?

Sean-StarLabs, 2024-04-08:
> ahh so can i download it and install directly SPI? could you share me a link for rom? or source code

I don't follow?

hicaru, 2024-04-08:
> > ahh so can i download it and install directly SPI? could you share me a link for rom? or source code
> 
> I don't follow?

i mean if you have coreboot for this device, how can i build it? i want to flash SPI on board.

Sean-StarLabs, 2024-04-08:
Click on the link - it shows "No firmware is available for this device."

hicaru, 2024-04-08:
> Click on the link - it shows "No firmware is available for this device."

ok, i though i didn't get something, because there check the box opposite coreboot(LVFS).

hicaru, 2024-04-08:
> Click on the link - it shows "No firmware is available for this device."

Please could you explain current stage, coreboot for VI AMD? sorry but we wait so many time  T_T

Sean-StarLabs, 2024-04-08:
Working on it

hicaru, 2024-04-08:
> on

you have an estimated release date?

Sean-StarLabs, 2024-04-08:
I do not

hicaru, 2024-04-27:
hello dear developers starlabs, i just have learned this page https://doc.coreboot.org/soc/amd/family17h.html, and i got a question! Why you cannot use open source `AGESA` which called `Picasso` in `soc/amd/picasso`? As i understand the Official AMD `AGESA` need NDA, you will wait NDA files endlessly, because they will never open them in near future.

Sean-StarLabs, 2024-04-27:
That's a different processor, and would still have the same problem.

NDAs aren't in the equation either.

hicaru, 2024-04-27:
> That's a different processor, and would still have the same problem.
> 
> NDAs aren't in the equation either.

yes, you are right((( it so sad

cyb3rmonk, 2024-06-06:
I just realized I'm unable to update the system firmware to the latest version (i.e. `fwupdmgr refresh --force && fwupdmgr update`). I get this error in dmsg:

```
[   15.913653] systemd[1]: Starting Firmware update daemon...
[   15.994985] fwupd[1944]: 03:12:52.955 FuPluginFlashrom     FCH device found but SMBus revision 0x51 does not match known values.
[   15.994985] fwupd[1944]: Please report this to flashrom@flashrom.org and include this log and
[   15.994985] cyb3rlab2305 fwupd[1944]: the output of lspci -nnvx, thanks!.
[   15.994985] cyb3rlab2305 fwupd[1944]: 03:12:52.955 FuPluginFlashrom     Could not determine chipset generation.
[   15.995368] cyb3rlab2305 fwupd[1944]: 03:12:52.955 FuPluginFlashrom     Laptops, notebooks and netbooks are difficult to support and we
[   15.995368] fwupd[1944]: recommend to use the vendor flashing utility. The embedded controller
[   15.995368] fwupd[1944]: (EC) in these machines often interacts badly with flashing.
[   15.995368] fwupd[1944]: See the manpage and https://flashrom.org/Laptops for details.
[   15.995368] fwupd[1944]: If flash is shared with the EC, erase is guaranteed to brick your laptop
[   15.995368] fwupd[1944]: and write may brick your laptop.
[   15.995368] fwupd[1944]: Read and probe may irritate your EC and cause fan failure, backlight
[   15.995368] fwupd[1944]: failure and sudden poweroff.
[   15.995368] fwupd[1944]: You have been warned.
```

Is this caused by the the same PSP / Ap base issue? Should I open a new issue for this?

Sean-StarLabs, 2024-06-06:
> Is this caused by the the same PSP / Ap base issue?

No, doesn't look like anything is wrong - what firmware are you running?

cyb3rmonk, 2024-06-06:
```
# fwupdmgr get-devices
├─System Firmware:
│ │   Device ID:          c738f5c2f089f407525d70d40c11ec7dbeb1e687
│ │   Summary:            UEFI System Resource Table device (updated via NVRAM)
│ │   Current version:    1.2.0
│ │   Minimum Version:    1.2.0

# fwupdmgr refresh --force && fwupdmgr update
Updating lvfs
Downloading…
Successfully downloaded new metadata: 2 local devices supported
Devices with no available firmware updates: 
 • Star Drive PCIe SSD
╔════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.2.0 to 24.3.0?                  ║
╠════════════════════════════════════════════════════════════════╣
║ This release includes the following features:                  ║
║                                                                ║
║ • Fixed an issue that stopped the backlight working after S3   ║
║ • Enabled support for PD 2.0 chargers                          ║
║ • Disable USB power in S3                                      ║
║ • Improved battery reporting                                   ║
║                                                                ║
║ StarBook must remain plugged into a power source for the       ║
║ duration of the update to avoid damage.                        ║
╚════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Waiting…
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Devices with the latest available firmware version:
 • UEFI dbx
An update requires a reboot to complete. Restart now? [y|N]: y
```

Then I reboot and I get the logs above

cyb3rmonk, 2024-06-09:
I don't want to spam everyone who's subscribed to this thread, so since you've said this wasn't related to the PSP / Ap base issue I opened a separate issue:

https://github.com/StarLabsLtd/firmware/issues/165

Sean-StarLabs, 2024-06-13:
[image]

Geting close...

sysnux, 2024-06-13:
Great, thanks for your commitment to that issue!

hicaru, 2024-07-13:
hello dear team, any news?

Sean-StarLabs, 2024-07-13:
Nothing yet

lil0r, 2024-08-17:
With the new starbook being released, does that mean we can finally wave good bye the hope to any descend support for the AMD machines?

Sean-StarLabs, 2024-08-17:
No - they're completely unrelated..

hicaru, 2024-08-21:
> No - they're completely unrelated..

When coreboot for AMD bro?

lil0r, 2024-08-31:
yes, when coreboot bro? Nearing fast two year wait for your supporters.

samueldr, 2024-08-31:
I know I'm not a "stakeholder" in this organization or repository. But let's act civil. Being rude is unlikely to help things get going, or worse: will annoy or anger the other party into not responding.

I know that, as an OSS developer, this kind of nagging might even make me de-prioritize the issue, if it's not one for me personally.

* * *

Here's my guess of how things are going. Anyone that knows: please correct me.

One is to assume they have a limited set of developers working on platform firmware [BIOS, UEFI, whatever you want to call it]. After all, only three contributors in this repo.

Couple that with the recent starlite launch (#184), development time was invested there in the past two months.

So, *even if there weren't any more blockers*, it's a good guesstimation that no time was available for this.

* * *

Here's how I'd approach it if I wanted to know about the state of things here:

> Hi!
>
> Since June/July 13th, has there been any change on things?
>
> While not asking for dates or time estimate, is there a list of things you can share that still needs to happen for Coreboot on the AMD variant? Of special interest is if there is anything blocked outside of StarLabs' control, to help better understand.
>
> Is there anything that can be done to help with this process?
>
> I am eager to make use of the Coreboot release to customize my personal system in a deeper manner.
>
> Cheers!

... or thereabout.

Which uh, I guess this is me asking, since I do have an AMD variant needing its Coreboot fulfilled :).

Cheers!

hicaru, 2024-09-04:
> I know I'm not a "stakeholder" in this organization or repository. But let's act civil. Being rude is unlikely to help things get going, or worse: will annoy or anger the other party into not responding.
> 
> I know that, as an OSS developer, this kind of nagging might even make me de-prioritize the issue, if it's not one for me personally.
> 
> Here's my guess of how things are going. Anyone that knows: please correct me.
> 
> One is to assume they have a limited set of developers working on platform firmware [BIOS, UEFI, whatever you want to call it]. After all, only three contributors in this repo.
> 
> Couple that with the recent starlite launch (#184), development time was _likely_ invested there in the past two months.
> 
> So, _even if there weren't any more blockers_, it's a good guesstimation that no time was available for this.
> 
> Here's how I'd approach it if I wanted to know about the state of things here:
> 
> > Hi!
> > Since June/July 13th, has there been any change on things?
> > While not asking for dates or time estimate, is there a list of things you can share that still needs to happen for Coreboot on the AMD variant? Of special interest is if there is anything blocked outside of StarLabs' control, to help better understand.
> > Is there anything that can be done to help with this process?
> > I am eager to make use of the Coreboot release to customize my personal system in a deeper manner.
> > Cheers!
> 
> ... or thereabout.
> 
> Which uh, I guess this is me asking, since I do have an AMD variant needing its Coreboot fulfilled :).
> 
> Cheers!

it all sounds great, but in 2 years we only got a screenshot, I bought this laptop where there was a statement that it would work on coreboot, but it turned out not to be so! I feel cheated and if things are somehow difficult, I think the team could unsubscribe and maybe someone from the community will help in any way they can

sjlongland, 2024-09-05:
If you can do it faster, knock yourself out… but bellyaching here is not going to make it happen any sooner.

hamishfor, 2024-09-08:
Hi @Sean-StarLabs et all, is there any additional pressure you can apply to AMD? Sometimes the squeaky wheel... Either trying to get the official product manager on the phone, or another attempt at contacting the engineering team directly. It's really disappointing, I have a lot of AMD hardware.

[27 more replies]