StarLabsLtd/firmware #67: [StarBook MK VI - Intel] Experiencing sudden black screen

drainboy, 2023-01-31:
Occasionally, the screen will turn black. The backlight is definitely still on but I'm unable to do anything but force shut down. I'm on Manjaro GNOME and would be willing to provide additional details to debug this issue

lemonconsumer, 2023-02-15:
Same boat here seems to be BIOS issue

mirdono, 2023-02-19:
I am experiencing a similar issue, for me it seems to be correlated with switching between laptop screen and  an external screen (it also happened twice when waking the laptop up from suspend-to-ram). I am using coreboot and Arch Linux.

lemonconsumer, 2023-02-19:
I had the fingerprint cable catch on fire removing it caused my laptop to fix completely id test and see if it's something with the fingerprint cable on yours

drainboy, 2023-03-01:
Have been experiencing this a lot more after switching from `AMI 1.2.0` to `coreboot 8.31`. I had theorized that it was because I was using too many applications at a time on my 8GB RAM. However, it has happened immediately after logging in and a few times with only firefox open.

Really hope this issue gets fixed soon. It currently feels like I'm threading on thin ice whenever I'm doing work on my StarBook

closed 2023-03-13 by Sean-StarLabs

drainboy, 2023-03-20:
Tested out the new `coreboot 8.32` for the past few days

the first 2 days was fine with no black screen crashes HOWEVER suspending and turning the laptop on again causes an issue where some of the icons and other things to not load in the application menu requiring a force shutdown.

after that first 2 days, the black screen issue has continued to occasionally happen. Will experiment again by reinstalling a fresh install of `coreboot 8.32` again

Sean-StarLabs, 2023-03-21:
> HOWEVER

Should be fixed in `8.33`

drainboy, 2023-03-26:
Back to report after testing out the new `coreboot 8.33` for the past few days. The first 3 days was surprisingly fine with very rare black screen crashes even after suspending. Even managed to use the laptop for a full day without any issue

However, after that 3 days, the black screen crashes has continued to happen. Today alone, it happened every time after suspending + a random black screen crash.

drainboy, 2023-08-22:
I know this thread has been closed for awhile but i have been trying to debug the issue so that i can use the Starbook as my daily. I initially reseated all of the ribbon cables connected to the motherboard and it was working for a short while before the issue happened again :(

I believed it was a hardware issue so I decided to test disconnecting some of the hardware (etc. keyboard backlight, daughter board, fingerprint, touchpad) from the motherboard. I remembered @lemonconsumer comment and disconnected the fingerprint ribbon cable.

> I had the fingerprint cable catch on fire removing it caused my laptop to fix completely id test and see if it's something with the fingerprint cable on yours

I can confidentially say that the black screen has stopped as I have been using the Starbook for the past few weeks

Sean-StarLabs, 2023-08-22:
Interesting - can you send your details over to support@starlabs.systems?

lemonconsumer, 2023-10-14:
I was given half a refund after I was told it was my fault due to custom ram slots so to have someone validate it is helpful thank you @drainboy

drainboy, 2023-10-19:
@lemonconsumer what do you mean by custom ram slots? I had bought extra RAM but experimented and still experience black screen with the RAM I got with my StarBook

lemonconsumer, 2023-10-20:
My own ram sticks and they gave me  I dont believe even a half of a refund  unless  I waited a long time to take it to a factory and  prove its broken :/

commit 9944895d (2023-03-13): Added StarBook MkVI-Intel coreboot 8.32

Fixes #67
Fixes #66
Fixes #61