StarLabsLtd/firmware #69: [Starbook VI] AMI Bios - Network Boot not working

garybowers, 2023-02-02:
I'm trying to pxe boot off my Starbook VI AMD Variant using a USB network adaptor (tested and working on other laptops) including a Dell USB-C to Ethernet and a Anker USB-A to Ethernet. 

Network Stack is enabled in the bios, however the boot option never appears, despite it being selected as a boot option in the bios settings.

jhvst, 2023-02-02:
I discovered the same, one option which I haven't looked into yet since the AMD variant does not support coreboot yet is to compile iPXE and coreboot manually and include the WiFi driver (not sure is it included in iPXE or not), flash it, and see what happens.

Sean-StarLabs, 2023-02-03:
We can add them - who makes the chips in those adapters? I think Dell use Realktek, not sure on Anker. 
`lsusb | grep Ethernet` will say

garybowers, 2023-02-03:
RTL8153 for all of them (Dell + x2 Anker) seems a pretty popular chip for usb(c/a) ethernet adapters

commit da7e759b (2023-02-08): Added StarBook MkVI-AMD ami 1.2.0

Fixes #69
closed 2023-02-08 by Sean-StarLabs