StarLabsLtd/firmware #71: Coreboot secure boot support

R-VdP, 2023-02-05:
Is there a way to do secure boot with coreboot on the starbook mk VI?
I searched a bit online, but the documentation on the topic seems to be quite sparse.

`bootctl` currently indicates that secure boot is unsupported. Is that because it's fundamentally unsupported by coreboot, or because it's not included in the current coreboot build/config?

I would basically like to:
1. Enable secure boot setup mode so that I can enrol my own keys
2. Enable secure boot

Thanks!

Sean-StarLabs, 2023-02-06:
You will find little on coreboot + Secure Boot; it's edk2 that needs to provide that. Right now, it can't do it. There are patches around to add it, but how they do it basically makes it pointless.

We've been working on using a different type of edk2 for a while, and once that's done, it'll offer Secure Boot, TPM control, variable store and a tonne of other things.

R-VdP, 2023-02-09:
Ok, glad to know that this is in the works! Do you have any idea whether we can expect this anytime soon, or will it be a while still?
Not to put pressure, but just to know the state of affairs.

Can we follow the progress on this somewhere?

Sean-StarLabs, 2023-02-17:
It should be possible to put a beta version out early next month. The way it's being done, no real way to track progress but I'll update here.

R-VdP, 2023-06-11:
Just checking if there are any updates on the progress on this? Thanks!

Sean-StarLabs, 2023-06-12:
Yes, almost all of it is done - CFR builds are in testing. The last component is just having a way to make the variables only accessible from edk2, and that'll give us "Secure" Secure Boot and BIOS Lock. Hard to say timings, as it is resisting :)

r-vdp, 2024-01-19:
@Sean-StarLabs any updates on this?

Sean-StarLabs, 2024-01-19:
UPL - no, moving SMM which in turns gives us secureboot will happen first. No ETA, but things are moving

Sean-StarLabs, 2024-11-04:
Fixed with 1509c8ea3fa92db26bb2d5700274b75d2949da26

NHellFire, 2024-11-04:
> Fixed with [1509c8e](https://github.com/StarLabsLtd/firmware/commit/1509c8ea3fa92db26bb2d5700274b75d2949da26)

Can we also get it enabled on the Lite MkIV?
Glad to see it supported on some though.

Sean-StarLabs, 2024-11-04:
Yeah, it'll get it in the next release