StarLabsLtd/firmware #72: [starbook mkvi intel] unable to flash

netmonk, 2023-02-08:
This weekend i installed endeavouros on the laptop, Since then i wanted to install latest firmware (8.27). 

I installed flashrom from pacman arch repo, in latest version 
`[root@nano ~]# flashrom --version
flashrom v1.2 on Linux 6.1.9-arch1-2 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

[root@nano ~]# flashrom -p internal -w /home/plonky/Downloads/8.27.rom -i bios --ifd -N
flashrom v1.2 on Linux 6.1.9-arch1-2 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Opened /dev/mtd0 successfully
Found Programmer flash chip "Opaque flash chip" (32768 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
Error accessing Opaque flash chip, 0x2000000 bytes at 0x00000000fe000000
/dev/mem mmap failed: Operation not permitted
Could not map flash chip Opaque flash chip at 0x00000000fe000000.
`

It's not working as expected, which flashroom do i need to use ? 
Do you distribute custom flashrom package for other distro than ubuntu/debian ?

Sean-StarLabs, 2023-02-08:
`v1.2` is almost 3 years old; you need `1.2-10` or later. For Endeavour, you could borrow the PKGBUILD from Manjaro - https://gitlab.manjaro.org/packages/community/flashrom-starlabs/-/blob/master/PKGBUILD

Sean-StarLabs, 2023-02-08:
You can check [here](https://fwupd.org/lvfs/devices/com.starlabs.B6-I.coreboot), but it's much easier to do with fwupd. It'll default only to use stable, but testing can be enabled with:
```
fwupdmgr enable-remote lvfs-testing
```
1.8-10 should work flashrom is installed, but if it isn't, `sudo fwupdtool get-devices --verbose --plugins flashrom` will say why not.

cichy1173, 2023-02-08:
> You can check [here](https://fwupd.org/lvfs/devices/com.starlabs.B6-I.coreboot), but it's much easier to do with fwupd. It'll default only to use stable, but testing can be enabled with:
>
> ``` [...]

Enabling `lvfs-enabling` nothing changed. On my first Fedora 37 installation I was using fwupd and flashrom from starlabs copr and it was working, but now I have second installation and I have flashrom from starlabs copr and fwupd from Fedora repo. 

output from `sudo fwupdtool get-devices --verbose --plugins flashrom`:

```bash
20:41:13.744 FuDebug              Verbose debugging enabled (on console 1)
20:41:13.744 FuMain               locked /run/lock/fwupdtool
20:41:13.750 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
20:41:13.750 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
Wczytywanie…             [-                                      ]20:41:13.756 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
20:41:13.756 FuCommon             reading /etc/fwupd/daemon.conf with 2494 bytes
20:41:13.756 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
20:41:13.757 FuConfig             using autodetected max archive size 4,3 GB
20:41:13.757 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
Wczytywanie…             [ -                                     ]20:41:13.758 FuRemoteList         loading /usr/share/fwupd/metainfo
20:41:13.758 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
20:41:13.758 XbSilo               file: e19bfd82-f29f-e20b-9164-55dc25a3709a, current:e19bfd82-f29f-e20b-9164-55dc25a3709a, cached: (null)
20:41:13.758 XbSilo               loading silo with file contents
20:41:13.758 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
20:41:13.758 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
20:41:13.758 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
20:41:13.758 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
20:41:13.758 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
20:41:13.758 FuRemoteList         ignoring unfound remote fwupd
20:41:13.758 FuRemoteList         ordering lvfs-testing=lvfs+1
Wczytywanie…             [                                       ]20:41:13.758 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
20:41:13.758 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
20:41:13.758 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
20:41:13.759 Jcat                 loaded 1 certificates
20:41:13.759 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
20:41:13.759 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
20:41:13.759 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
20:41:13.759 Jcat                 loaded 1 certificates
20:41:13.759 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
20:41:13.759 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
20:41:13.759 FuEngine             client certificate now exists: nothing to do
Wczytywanie…             [                                       ]20:41:13.759 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
20:41:13.759 FuHistory            got schema version of 8
Wczytywanie…             [*                                      ]20:41:13.763 FuPlugin             load(/usr/lib64/fwupd-1.8.10/libfu_plugin_modem_manager.so)
20:41:13.763 FuContext            added udev subsystem watch of platform
20:41:13.763 FuContext            added udev subsystem watch of block
20:41:13.763 FuContext            added udev subsystem watch of pci
20:41:13.763 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
20:41:13.763 FuPluginDell         Using libsmbios 2.4
20:41:13.763 FuContext            added udev subsystem watch of i2c
20:41:13.763 FuContext            added udev subsystem watch of i2c-dev
20:41:13.763 FuContext            added udev subsystem watch of hidraw
20:41:13.763 FuContext            added udev subsystem watch of gpio
20:41:13.763 FuContext            added udev subsystem watch of mei
20:41:13.764 FuContext            added udev subsystem watch of iommu
20:41:13.764 FuContext            added udev subsystem watch of video4linux
20:41:13.764 FuContext            added udev subsystem watch of msr
20:41:13.764 FuContext            added udev subsystem watch of mtd
20:41:13.764 FuContext            added udev subsystem watch of nvme
20:41:13.765 FuContext            added udev subsystem watch of drm
20:41:13.765 FuContext            added udev subsystem watch of drm_dp_aux_dev
20:41:13.765 FuContext            added udev subsystem watch of serio
20:41:13.765 FuPluginTest         init
20:41:13.765 FuContext            added udev subsystem watch of thunderbolt
20:41:13.765 FuContext            added udev subsystem watch of tpm
20:41:13.765 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
[... cut ...]
20:41:13.785 FuEngine             no flashrom plugin for HwId 5c917039-d938-5c9a-b22a-9c392b1534f3: no plugin flashrom found
Wczytywanie…             [***                                    ]20:41:13.785 FuContext            battery threshold now 30
Wczytywanie…             [-                                      ]
20:41:13.858 FuCommon             creating path /var/cache/fwupd/motd.d
20:41:13.858 FuEngine             writing motd target /var/cache/fwupd/motd.d/85-fwupd
No detected devices
20:41:13.860 FuMain               No detected devices

20:41:13.861 FuPluginTest         destroy

```

Sean-StarLabs, 2023-02-08:
> Enabling lvfs-enabling nothing changed. 

It will have enabled the testing remote, but that'll only find updates if its working first.

Whats the package version of that fwupd?

cichy1173, 2023-02-08:
> Whats the package version of that fwupd?

```bash
[cichy1173@starbook ~]$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.10
```

Sean-StarLabs, 2023-02-08:
Thanks, that's odd - as the hwid is in that version (https://github.com/fwupd/fwupd/blob/44137021f26f69d04d5ae250adc06a52d0412bcf/plugins/flashrom/flashrom.quirk#L43). Will ask the man who will know!

Sean-StarLabs, 2023-02-08:
Is `fwupd-plugin-flashrom` installed?

cichy1173, 2023-02-10:
> Is `fwupd-plugin-flashrom` installed?

Ok. I installed this and rebooted and it is now OK :-)

Screenshot from GNOME firmware app:

[image]

Sean-StarLabs, 2023-02-10:
Awesome - guide updated :)