StarLabsLtd/firmware #76: [StarBook MkVI - AMD] Trouble with s2idle

tsmithe, 2023-02-15:
The platform reports `s2idle` support in `/sys/power/mem_sleep`, but attempting to enter suspend causes the system to lock up.  Moreover, the `amd_pmc` module does not load correctly, and the ACPI FADT doesn't seem to report s2idle support.  See the comments around here for a little more information:  https://gitlab.freedesktop.org/drm/amd/-/issues/2403#note_1774129

Sean-StarLabs, 2023-03-23:
Working on it regardless - about 80% of the way through the dsdt :/

superm1, 2023-03-28:
There's an issue that some newer Intel hardware doesn't support lane switch when transitioning to L0s, so if disabling aspm helps try to just disable L0s to narrow down.

commit 52d92e17 (2023-02-20): Added StarBook MkVI-AMD ami 1.3.0

Fixes #76
Fixes #77
closed 2023-04-11 by Sean-StarLabs