StarLabsLtd/firmware #78: [StarBook Mk VI - Intel - AMI] Firmware update error

winterknife, 2023-02-19:
A few days ago, I switched branch from `Coreboot` to `AMI Aptio V` `BIOS` and apparently the transition was successful, but now `fwupd` keeps offering to update the System firmware even though I'm already on the latest version(`1.2.0`) and what's more is that when I try to proceed with the update, it throws a "BIOS locked" error upon reboot.

Things that I have already tried:
1) Ensuring that `UEFI` `Secure Boot` is disabled
2) Ensuring that "boot order lock" is not set
2) Moving `Linux-Firmware-Updater` entry to the top of the boot order

Here are some command outputs to provide more insights into the issue:
```console
[cci-dev06@starbook Desktop]$ fwupdmgr --version | grep fwupd
compile   org.freedesktop.fwupd         1.8.10
runtime   org.freedesktop.fwupd-efi     1.4
runtime   org.freedesktop.fwupd         1.8.10
```

```console
[cci-dev06@starbook Desktop]$ sudo dmidecode --type bios
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.5.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: American Megatrends International, LLC.
	Version: 1.2.0
	Release Date: 01/11/2023
	Address: 0xF0000
	Runtime Size: 64 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		BIOS ROM is socketed
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		Print screen service is supported (int 5h)
		Serial services are supported (int 14h)
		Printer services are supported (int 17h)
		CGA/mono video services are supported (int 10h)
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 5.26
	Firmware Revision: 1.1

Handle 0x0064, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 1
		en|US|iso8859-1
	Currently Installed Language: en|US|iso8859-1
```

```console
[cci-dev06@starbook Desktop]$ fwupdmgr get-devices
Star Labs StarBook
│
├─12th Gen Intel Core™ i7-1260P:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000421
│     Vendor:             Intel
[... 245 lines cut ...]
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─StarBook MkVI-Intel ami:
        New version:      1.2.0
        Remote ID:        lvfs
        Release ID:       17911
        Summary:          ami firmware for StarBook MkVI-Intel
        Licence:          Proprietary
        Size:             16.8 MB
        Created:          2023-01-13
        Urgency:          Medium
        Vendor:           Star Labs
        Release Flags:    • Is upgrade
        Description:      
        This stable release includes the following features:
        
        • Updated default configuration
        • Enabled mirror flag to update EC along with BIOS
```

winterknife, 2023-03-03:
Here is an error report generated from a failed update attempt:
```console
Target:                  https://fwupd.org/lvfs/firmware/report
Payload:                 {
                           "ReportVersion" : 2,
                           "MachineId" : "13ef742aebbc213ad55e1065cde6394fdfb1e8170e0c4f350166f9436937aec7",
                           "Metadata" : {
                             "DistroId" : "ubuntu",
                             "DistroVersion" : "22.04"
                           },
                           "Reports" : [
                             {
                               "Checksum" : "a779a2533e0632ce7190106866da83d44d63b29b",
                               "ReleaseId" : null,
                               "Protocol" : "org.uefi.capsule",
                               "UpdateState" : 3,
                               "UpdateError" : "failed to run update on reboot",
                               "Guid" : [
                                 "02e64049-abcc-44a5-af9a-fafc390dd844"
                               ],
                               "Plugin" : "uefi_capsule",
                               "VersionOld" : "1.1.0",
                               "VersionNew" : "1.2.0",
                               "Flags" : 574619947,
                               "Created" : 1677783343,
                               "Modified" : 1677783447,
                               "Metadata" : {
                                 "LastAttemptVersion" : "0x1010000",
                                 "HostEnclosureKind" : "9",
                                 "TpmFamily" : "2.0",
                                 "RuntimeVersion(org.kernel)" : "5.19.0-35-generic",
                                 "SystemIntegrityNew" : "UEFI:Boot0003=4d3ee0baa484576d412ad9258fdc909e91924a4900da8e1f6a03eb574361ed5d\nUEFI:PK=033950aff7d7722022803dbd1fa90bff5e0cf61d4660ab6293369dc916f949a5\nUEFI:Boot0006=fdff09a931ed879fd863c3ff737920106c8844b575304b9c9bd4a7636b92c5b9\nUEFI:BootOrder=3a6ecf411e0c62c035e3102e390e87c7bf685aea87fcb4f1c82bfd552e23e059\nUEFI:SignatureSupport=f8b3f359e3073393d6acc06daff98b5600fc096fe2b75994cb1136c501245fb5\nUEFI:KEKDefault=8599624905e4fa11b379471f80f870369cc046d1ed45fefe540072a6784934bf\nUEFI:Boot0001=647b7721d0ccb7f5dacbb0c2a611ffa2d147e93a649c8fca5c0b33640967b7d2\nUEFI:SetupMode=6e340b9cffb37a989ca544e6bb780a2c78901d3fb33738768511a30617afa01d\nUEFI:OsIndicationsSupported=7c9fa136d4413fa6173637e883b6998d32e1d675f88cddff9dcbcf331820f4b8\nUEFI:db=5dbc0253a9bb2a84e03522cfc37a3b5be3b1088ecfcd753d08f1425cdef2b5c1\nUEFI:SecureBoot=6e340b9cffb37a989ca544e6bb780a2c78901d3fb33738768511a30617afa01d\nUEFI:VendorKeys=4bf5122f344554c53bde2ebb8cd2b7e3d1600ad631c385a5d7cce23c7785459a\nUEFI:Boot0004=396b18aad4fd46827a59ee1ef8944c9a92d0a2889dc553e6a739504732677d35\nUEFI:BootCurrent=99be5efb88ca2013bd8e4eb035fd42d5245468fe9afa70d8ba9c1c419a48c4e8\nUEFI:KEK=8599624905e4fa11b379471f80f870369cc046d1ed45fefe540072a6784934bf\nUEFI:dbx=1da86b89493c35e72a8a1bf64d67f88224e3030dcdb2dcadfd9780e27a28881d\nUEFI:Boot0002=4e4e368106f59ca1f86e986a5dd025baf493a2d783e13a26d88f2e7766926362\nUEFI:Boot0005=50353cfffc3700d131398887941667db2c188a79425875dd77375a12ea023abb\nACPI:TPM2=a921ce3b853defab0b85990952a9b0e6e232cb53cb87a75a2fcf871255d9a4cc\nUEFI:PKDefault=033950aff7d7722022803dbd1fa90bff5e0cf61d4660ab6293369dc916f949a5",
                                 "DistroVersion" : "22.04",
                                 "RuntimeVersion(org.freedesktop.fwupd)" : "1.8.10",
                                 "HostSku" : "B6-I",
                                 "UEFIUXCapsule" : "Enabled",
                                 "CpuArchitecture" : "x86_64",
                                 "SecureBoot" : "Disabled",
                                 "HostFamily" : "B6",
                                 "HostBiosVersion" : "1.2.0",
                                 "HostVendor" : "Star Labs",
                                 "RuntimeVersion(org.freedesktop.gusb)" : "0.3.10",
                                 "CompileVersion(org.freedesktop.gusb)" : "0.3.10",
                                 "KernelName" : "Linux",
                                 "DistroPrettyName" : "Ubuntu 22.04.2 LTS",
                                 "RuntimeVersion(com.dell.libsmbios)" : "2.4",
                                 "MissingCapsuleHeader" : "False",
                                 "DistroName" : "Ubuntu",
                                 "SystemIntegrityOld" : "UEFI:Boot0003=4d3ee0baa484576d412ad9258fdc909e91924a4900da8e1f6a03eb574361ed5d\nUEFI:PK=033950aff7d7722022803dbd1fa90bff5e0cf61d4660ab6293369dc916f949a5\nUEFI:Boot0006=fdff09a931ed879fd863c3ff737920106c8844b575304b9c9bd4a7636b92c5b9\nUEFI:BootOrder=a8db6e879b7a6854f319c0bd7ef4e0be174251d941fa48b3e87bdfdf592c25d0\nUEFI:SignatureSupport=f8b3f359e3073393d6acc06daff98b5600fc096fe2b75994cb1136c501245fb5\nUEFI:KEKDefault=8599624905e4fa11b379471f80f870369cc046d1ed45fefe540072a6784934bf\nUEFI:Boot0001=647b7721d0ccb7f5dacbb0c2a611ffa2d147e93a649c8fca5c0b33640967b7d2\nUEFI:SetupMode=6e340b9cffb37a989ca544e6bb780a2c78901d3fb33738768511a30617afa01d\nUEFI:OsIndicationsSupported=7c9fa136d4413fa6173637e883b6998d32e1d675f88cddff9dcbcf331820f4b8\nUEFI:db=5dbc0253a9bb2a84e03522cfc37a3b5be3b1088ecfcd753d08f1425cdef2b5c1\nUEFI:SecureBoot=6e340b9cffb37a989ca544e6bb780a2c78901d3fb33738768511a30617afa01d\nUEFI:VendorKeys=4bf5122f344554c53bde2ebb8cd2b7e3d1600ad631c385a5d7cce23c7785459a\nUEFI:Boot0004=396b18aad4fd46827a5 [... cut ...]
                                 "EfivarNvramUsed" : "77059",
                                 "DistroId" : "ubuntu",
                                 "HostFirmwareMinorRelease" : "01",
                                 "CompileVersion(org.freedesktop.fwupd)" : "1.8.10",
                                 "LastAttemptStatus" : "0x0",
                                 "HostBiosMinorRelease" : "1a"
                               }
                             }
                           ]
                         }
```

Sean-StarLabs, 2023-03-03:
I know what the issue is, will work a fix into the next release that includes other fixed

commit 1069ed9f (2023-03-27): Added StarBook MkVI-Intel ami 1.3.0

Fixes #78
Fixes #79