StarLabsLtd/firmware #80: [StarBook MK VI - Intel - AMI] switch-branch: image write failed, err=2

drainboy, 2023-02-22:
Since coreboot has been updated quite a bit (since i last had it), I wanted to try switching back to it but I kept getting this error

```
Perform operation? [Y|n]: 
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Waiting…                 [***************************************]
Writing…                 [***************************************]
Writing…                 [                                       ]image write failed, err=2
```

Sean-StarLabs, 2023-02-22:
That's an old fwupd bug - you need a newer version

drainboy, 2023-02-22:
I have installed the following packages

`Packages (2) flashrom-starlabs-1.3.0rc2-4  fwupd-starlabs-1.8.10-4`

are these the latest versions?

Sean-StarLabs, 2023-02-22:
Is that Manjaro? If so, replace `fwupd-starlabs` with `fwupd`

drainboy, 2023-02-22:
Yes i'm using Manjaro. I installed `fwupd` but i'm still experiencing the same error

Here's the logs when I used the `fwupdmgr --version` command
```
compile   org.freedesktop.fwupd         1.8.10
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.3
runtime   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.8.10
runtime   org.kernel                    6.1.12-1-MANJARO
```

Sean-StarLabs, 2023-02-22:
What does `sudo flashrom -p internal -r foo.bar -i bios --ifd -N` show?

drainboy, 2023-02-22:
```
flashrom unknown on Linux 6.1.12-1-MANJARO (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Opened /dev/mtd0 successfully
Found Programmer flash chip "Opaque flash chip" (32768 kB, Programmer-specific) on internal.
Reading ich descriptor... done.
Using region: "bios".
Reading flash... done.
```

Sean-StarLabs, 2023-02-22:
Okay, might be kermel - can you let me know the version you're using? And are you able to try an older one?

drainboy, 2023-02-22:
I'm using Manjaro `6.1.12-1` and I have tried `fwupdmgr switch-branch` on the following kernel versions:

- `6.0.19 - 4`
- `5.15.94 - 1`

drainboy, 2023-02-23:
Managed to install `coreboot 8.31` by live booting to Linux Mint and following the instructions from the [support page](https://support.starlabs.systems/kb/updates/installing-updates-from-the-lvfs)

closed 2023-02-23 by drainboy