StarLabsLtd/firmware #81: Inconsistent memory map preventing hibernation

R-VdP, 2023-02-25:
Hibernation of my StarBook Mk VI (Intel, coreboot) fails because the memory map passed to the kernel seems to be different on every boot. Below are a couple of memory maps that I captured, you can notice a difference for instance in the fourth region,  which has a different ending address in all three mappings.

Is there any way to deal with this so that hibernation can be used?

```
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x0000000000000000-0x0000000000000fff] reserved
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x0000000000001000-0x000000000009ffff] usable
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x0000000000100000-0x000000007252dfff] usable
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x000000007252e000-0x000000007252ffff] ACPI data
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x0000000072530000-0x00000000755aafff] usable
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x00000000755ab000-0x00000000755abfff] ACPI data
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x00000000755ac000-0x0000000076613fff] usable
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x0000000076614000-0x0000000076617fff] ACPI data
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x0000000076618000-0x000000007661cfff] usable
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x000000007661d000-0x0000000076621fff] ACPI data
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x0000000076622000-0x0000000076635fff] usable
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x0000000076636000-0x0000000076637fff] ACPI data
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x0000000076638000-0x0000000076809fff] usable
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x000000007680a000-0x000000007694dfff] reserved
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x000000007694e000-0x0000000076951fff] ACPI NVS
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x0000000076952000-0x0000000076959fff] ACPI data
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x000000007695a000-0x000000007695afff] usable
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x000000007695b000-0x00000000803fffff] reserved
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x00000000ff030000-0x00000000ff06ffff] reserved
fév 21 23:02:31 starbook kernel: BIOS-e820: [mem 0x0000000100000000-0x000000087fbfffff] usable

fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x0000000000000000-0x0000000000000fff] reserved
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x0000000000001000-0x000000000009ffff] usable
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x0000000000100000-0x00000000750a0fff] usable
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x00000000750a1000-0x00000000750a3fff] ACPI data
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x00000000750a4000-0x00000000755d7fff] usable
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x00000000755d8000-0x00000000755d8fff] ACPI data
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x00000000755d9000-0x0000000076613fff] usable
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x0000000076614000-0x0000000076617fff] ACPI data
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x0000000076618000-0x000000007661cfff] usable
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x000000007661d000-0x0000000076621fff] ACPI data
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x0000000076622000-0x0000000076635fff] usable
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x0000000076636000-0x0000000076637fff] ACPI data
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x0000000076638000-0x0000000076809fff] usable
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x000000007680a000-0x000000007694dfff] reserved
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x000000007694e000-0x0000000076951fff] ACPI NVS
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x0000000076952000-0x0000000076959fff] ACPI data
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x000000007695a000-0x000000007695afff] usable
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x000000007695b000-0x00000000803fffff] reserved
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x00000000ff030000-0x00000000ff06ffff] reserved
fév 25 00:46:55 starbook kernel: BIOS-e820: [mem 0x0000000100000000-0x000000087fbfffff] usable

fév 25 11:04:07 starbook kernel: BIOS-provided physical RAM map:
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x0000000000000000-0x0000000000000fff] reserved
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x0000000000001000-0x000000000009ffff] usable
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x0000000000100000-0x000000007255bfff] usable
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x000000007255c000-0x000000007255dfff] ACPI data
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x000000007255e000-0x00000000750abfff] usable
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x00000000750ac000-0x00000000750acfff] ACPI data
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x00000000750ad000-0x00000000755d7fff] usable
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x00000000755d8000-0x00000000755d8fff] ACPI data
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x00000000755d9000-0x0000000076613fff] usable
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x0000000076614000-0x0000000076617fff] ACPI data
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x0000000076618000-0x000000007661cfff] usable
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x000000007661d000-0x0000000076621fff] ACPI data
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x0000000076622000-0x0000000076635fff] usable
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x0000000076636000-0x0000000076637fff] ACPI data
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x0000000076638000-0x0000000076809fff] usable
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x000000007680a000-0x000000007694dfff] reserved
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x000000007694e000-0x0000000076951fff] ACPI NVS
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x0000000076952000-0x0000000076959fff] ACPI data
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x000000007695a000-0x000000007695afff] usable
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x000000007695b000-0x00000000803fffff] reserved
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x00000000ff030000-0x00000000ff06ffff] reserved
fév 25 11:04:07 starbook kernel: BIOS-e820: [mem 0x0000000100000000-0x000000087fbfffff] usable
```

Sean-StarLabs, 2023-03-01:
Short answer is no, UPL mentioned in #71 might actually fix it as it avoids the resource allocator script and uses hobs.

closed 2023-03-01 by Sean-StarLabs

divico, 2025-04-30:
I'm having the same issue, resuming from hibernation fails often with
```
PM: hibernation: Image mismatch: architecture specific data
PM: hibernation: Failed to load image, recovering.
```
Could you tell me if this can be fixed with a firmware update, if so which version should I use on my Starbook MkVIr2-Intel?

Sean-StarLabs, 2025-04-30:
What's "often"?
What firmware are you running?

divico, 2025-08-14:
> What kernel you using?

Using `6.15.9` (arch linux-lts)

Sean-StarLabs, 2025-11-11:
> Can we help fix this by providing some more logs or testing?

Afraid not

sim31, 2025-11-18:
Experiencing this as well. But for me it started happening fairly recently, since 25.10 version I think. And I was using hibernation before then a lot. Kernel: 6.17.5-arch1-1.

divico, 2025-11-18:
> Experiencing this as well. But for me it started happening fairly recently, since 25.10 version I think. And I was using hibernation before then a lot. Kernel: 6.17.5-arch1-1.

I also did not experience this in the first years of using my Starbook. For sure I did not have it on firmware versions <`24`. I'm not sure when it started. 

I don't understand why more people aren't reporting this issue. It's such a significant regression for laptops IMHO.

divico, 2025-12-18:
Is this fix part of coreboot `25.12` firmware release? The changelog states  "Fixed resuming from hibernation (S4)"