StarLabsLtd/firmware #83: Error when updating: failed to open device for prepare: failed to read layout from Intel ICH descriptor

sim31, 2023-03-09:
Output from `fwupdmgr report-history`:
```
Target:                  https://fwupd.org/lvfs/firmware/report
Payload:                 {
                           "ReportVersion" : 2,
                           "MachineId" : "afecf0c45d852258b60addd29fbe0dc6c0e7e08d485cce24fdecffe152d2be3c",
                           "Metadata" : {
                             "DistroId" : "arch"
                           },
                           "Reports" : [
                             {
                               "Checksum" : "2da33b6f9c0b25c4af8a8655b338826f43b75932",
                               "ReleaseId" : null,
                               "Protocol" : "org.flashrom",
                               "UpdateState" : 3,
                               "UpdateError" : "failed to open device for prepare: failed to read layout from Intel ICH descriptor",
                               "Guid" : [
                                 "230c8b18-8d9b-53ec-838b-6cfc0383493a"
                               ],
                               "Plugin" : "flashrom",
                               "VersionOld" : "8.22",
                               "VersionNew" : "8.31",
                               "Flags" : 687299756073,
                               "Created" : 1678340053,
                               "Modified" : 1678340584,
                               "Metadata" : {
                                 "HostEnclosureKind" : "9",
                                 "RuntimeVersion(org.kernel)" : "6.2.1-arch1-1",
                                 "RuntimeVersion(org.freedesktop.fwupd)" : "1.8.12",
                                 "HostSku" : "B6-I",
                                 "CpuArchitecture" : "x86_64",
                                 "HostFamily" : "B6-I",
                                 "HostBiosVersion" : "8.22",
                                 "HostVendor" : "Star Labs",
                                 "RuntimeVersion(org.freedesktop.gusb)" : "0.4.5",
                                 "KernelName" : "Linux",
                                 "DistroPrettyName" : "Arch Linux",
                                 "RuntimeVersion(com.dell.libsmbios)" : "2.4",
                                 "RuntimeVersion(com.hughsie.libjcat)" : "0.1.12",
                                 "DistroName" : "Arch Linux",
                                 "SystemIntegrityOld" : "UEFI:BootOrder=93f238b8a0e6780007f67739b547206ea24769ed49b3eff7395930097cd371d8\nUEFI:Boot0002=5cea51f03b0ad88743dce6770a9ebc4b96fc86866b2a8b459abea08d0ebc5948\nUEFI:Boot0001=3a757cae8c3ad33c0f9105f7124c9d56a4bce5cdaf625ee88436bf8b10c44490\nUEFI:Boot0000=88047d5437636859eee3716e2ca0851eab4cb4615c84a6e2b57c694bc084cf1d\nUEFI:BootCurrent=2921a11f25dadaa24aa79a548e4e81508c2e5e56af2d833d65e2bcce448ce2f5\nUEFI:Boot0005=573ba587bc3c3f75affcdbc03cefb9ce74a1faf7821287d6b1b7ee81bb1eba3e\nUEFI:OsIndicationsSupported=14e4b8ac16877a5d25661d84f8558c1481bf795a165d92feee46191b3224cde5\nUEFI:Boot0003=e19705946a08a2305f8fc84dae26f1704c56104cd15d1b135a9d6272cb2f3ff3\nACPI:TPM2=81ffd55b51e7d2a789c67d65dcde84d5727377a7d3b642e69dc2d60542d9dfbc",
                                 "FwupdSupported" : "True",
                                 "KernelVersion" : "6.2.1-arch1-1",
                                 "CompileVersion(org.freedesktop.gusb)" : "0.4.5",
                                 "HostFirmwareMajorRelease" : "01",
                                 "HostProduct" : "StarBook",
                                 "BootTime" : "1678340036",
                                 "RuntimeVersion(org.freedesktop.fwupd-efi)" : "1.4",
                                 "CpuModel" : "Intel 12th Gen Intel Core™ i7-1260P",
                                 "HostBaseboardManufacturer" : "Star Labs",
                                 "HostBiosMajorRelease" : "04",
                                 "HostBiosVendor" : "coreboot",
                                 "LinuxLockdown" : "none",
                                 "CompileVersion(com.hughsie.libjcat)" : "0.1.12",
                                 "CompileVersion(com.hughsie.libxmlb)" : "0.3.11",
                                 "HostBaseboardProduct" : "StarBook",
                                 "DistroId" : "arch",
                                 "CompileVersion(org.freedesktop.fwupd)" : "1.8.12",
                                 "HostFirmwareMinorRelease" : "01",
                                 "HostBiosMinorRelease" : "12"
                               }
                             }
                           ]
                         }
```

Relevant journal entries (`journalctl -u fwupd`):
```
-- Boot a6582d3961d6439d81c816449ae6198d --
Mar 07 14:56:35 sbook systemd[1]: Starting Firmware update daemon...
[... 19 lines cut ...]
Mar 07 16:29:39 sbook systemd[1]: fwupd.service: Consumed 1.717s CPU time.
-- Boot 687705eee169400e9216de6fcabdcc49 --
Mar 07 16:30:51 sbook systemd[1]: Starting Firmware update daemon...
Mar 07 16:30:51 sbook fwupd[1006]: 14:30:51.442 FuPluginUefiCapsule  SMBIOS BIOS Characteristics Extension Byte 2 is invalid -- UEFI Specification is unsupported, but /sys/firmware/efi exists: System does not support UEFI mode
Mar 07 16:30:52 sbook systemd[1]: Started Firmware update daemon.
Mar 07 16:31:21 sbook fwupd[1006]: 14:31:21.628 FuPluginFlashrom     Error accessing Opaque flash chip, 0x2000000 bytes at 0x00000000fe000000
Mar 07 16:31:21 sbook fwupd[1006]: 14:31:21.628 FuPluginFlashrom     /dev/mem mmap failed: Operation not permitted
Mar 07 16:31:21 sbook fwupd[1006]: 14:31:21.628 FuPluginFlashrom     Could not map flash chip Opaque flash chip at 0x00000000fe000000.
Mar 08 08:07:46 sbook systemd[1]: fwupd.service: Deactivated successfully.
Mar 08 08:07:46 sbook systemd[1]: fwupd.service: Consumed 2.120s CPU time.
Mar 08 10:28:15 sbook systemd[1]: Starting Firmware update daemon...
Mar 08 10:28:15 sbook fwupd[40677]: 08:28:15.102 FuPluginUefiCapsule  SMBIOS BIOS Characteristics Extension Byte 2 is invalid -- UEFI Specification is unsupported, but /sys/firmware/efi exists: System does not support UEFI mode
Mar 08 10:28:16 sbook systemd[1]: Started Firmware update daemon.
Mar 08 12:30:21 sbook systemd[1]: fwupd.service: Deactivated successfully.
Mar 09 06:27:32 sbook systemd[1]: Starting Firmware update daemon...
Mar 09 06:27:32 sbook fwupd[109193]: 04:27:32.797 FuPluginUefiCapsule  SMBIOS BIOS Characteristics Extension Byte 2 is invalid -- UEFI Specification is unsupported, but /sys/firmware/efi exists: System does not support UEFI mode
Mar 09 06:27:33 sbook systemd[1]: Started Firmware update daemon.
Mar 09 06:28:18 sbook fwupd[109193]: 04:28:18.132 FuPluginFlashrom     Error accessing Opaque flash chip, 0x2000000 bytes at 0x00000000fe000000
Mar 09 06:28:18 sbook fwupd[109193]: 04:28:18.132 FuPluginFlashrom     /dev/mem mmap failed: Operation not permitted
Mar 09 06:28:18 sbook fwupd[109193]: 04:28:18.132 FuPluginFlashrom     Could not map flash chip Opaque flash chip at 0x00000000fe000000.
```

I run arch linux and installed fwupd using [gnome-firmware](https://archlinux.org/packages/community/x86_64/gnome-firmware/) package.

Not sure if this is the best place to report it, so let me know.

sim31, 2023-03-09:
Just tried that. Now I get a slightly different error in journal:
```
ar 09 16:16:04 sbook systemd[1]: Starting Firmware update daemon...
Mar 09 16:16:04 sbook fwupd[1654]: 14:16:04.137 FuPluginUefiCapsule  SMBIOS BIOS Characteristics Extension Byte 2 is invalid -- >
Mar 09 16:16:05 sbook systemd[1]: Started Firmware update daemon.
Mar 09 16:16:37 sbook fwupd[1654]: 14:16:37.640 FuPluginFlashrom     Error accessing Opaque flash chip, 0x2000000 bytes at 0x000>
Mar 09 16:16:37 sbook fwupd[1654]: 14:16:37.640 FuPluginFlashrom     /dev/mem mmap failed: Resource temporarily unavailable
Mar 09 16:16:37 sbook fwupd[1654]: 14:16:37.640 FuPluginFlashrom     Could not map flash chip Opaque flash chip at 0x00000000fe0>
Mar 09 16:17:18 sbook fwupd[1654]: 14:17:18.621 FuPluginUpower       failed to query lid state
Mar 09 16:17:18 sbook fwupd[1654]: 14:17:18.621 FuPluginUpower       failed to query lid state
Mar 09 16:17:19 sbook systemd[1]: Stopping Firmware update daemon...
Mar 09 16:17:19 sbook fwupd[1654]: 14:17:19.425 FuMain               Received SIGTERM
Mar 09 16:17:19 sbook systemd[1]: fwupd.service: Deactivated successfully.
Mar 09 16:17:19 sbook systemd[1]: Stopped Firmware update daemon.
Mar 09 16:17:19 sbook systemd[1]: fwupd.service: Consumed 1.122s CPU time.
```

Tried the suggestion I found [here](https://www.flashrom.org/FAQ) to set nopat kernel parameter. Then the computer does not boot. It gets stuck at: `Triggerring uevents...`. Normally I see this in the beginning of the boot.

Sean-StarLabs, 2023-03-09:
You don't need `nopat` but that does suggest an old version of flashrom - what version do you have installed?

sim31, 2023-03-10:
1.2-4. Just saw that it is flagged as out of date in arch linux repos. Do you think 1.3 would fix this?

Sean-StarLabs, 2023-03-10:
Yes it would

sim31, 2023-03-11:
Built and installed 1.3.0 flashrom version. Weird - now it does even start the update - says: "no updateable devices".
This is the output from `fwupdmgr refresh --force`:
```
Reading…                 [ -                                     ]
Updating lvfs
Downloading…             [      \                                ]
Successfully downloaded new metadata: 0 local devices supported
```
This is the output from `fwupdmgr update`:
```
Reading…                 [ -                                     ]
Devices with no available firmware updates: 
 • 0000:00:1f.5
 • Star Drive PCIe SSD
 • SuperIO FuSuperioIt89Device
No updatable devices
```
Output from `fwupdmgr get-devices`:
```
Reading…                 [ -                                     ]Star Labs StarBook
│

├─0000:00:1f.5:
│     Device ID:          b04e387fb80d2b91f37a4d0c7b21461c451775e1
│     Summary:            Memory Technology Device
│     Vendor:             DMI:Star Labs
│     GUIDs:              5f93d7e7-e282-59b9-b663-0146e382f8f6 ← MTD\NAME_0000:00:1f.5
│                         8332559d-efc1-5670-9baa-e06a25556a96 ← MTD\VENDOR_Star-Labs&NAME_0000:00:1f.5
│                         ca2344e2-00a1-509a-9062-031db63bb375 ← MTD\VENDOR_Star-Labs&PRODUCT_StarBook&NAME_0000:00:1f.5
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│   
├─12th Gen Intel Core™ i7-1260P:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000429
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         ab855c04-4ff6-54af-8a8a-d8193daa0cd8 ← CPUID\PRO_0&FAM_06&MOD_9A
│                         3ebbde86-d03e-549a-a8fd-02ebf9aa537a ← CPUID\PRO_0&FAM_06&MOD_9A&STP_3
│     Device Flags:       • Internal device
│   
├─Alder Lake-P Integrated Graphics Controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    0c
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              eaad9970-8e4d-56da-88ab-41a8c1e2811f ← PCI\VEN_8086&DEV_46A6
│                         ed0b9458-c2f1-54c5-9063-dea8f75b4039 ← PCI\VEN_8086&DEV_46A6&REV_0C
│                         226cbe0a-2ccd-5e12-8a5a-60aa1e438894 ← PCI\VEN_8086&DEV_46A6&SUBSYS_808646A6
│                         c78c6be8-ff08-5945-bb0b-92f25dca2b68 ← PCI\VEN_8086&DEV_46A6&SUBSYS_808646A6&REV_0C
│                         c4625510-a985-517c-8800-0ecfc6f68c8f ← PCI\VEN_8086&DEV_46A6&REV_00
│                         25449cf0-7ce3-5bdd-a343-e9f719a164d2 ← PCI\VEN_8086&DEV_46A6&SUBSYS_808646A6&REV_00
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Star Drive PCIe SSD:
│     Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:            NVM Express solid state drive
│     Current version:    EIFM31.4
│     Vendor:             Phison Electronics Corporation (NVME:0x1987)
│     Serial Number:      1348072C06C700242034
│     GUIDs:              59639c7d-1b5d-5b7f-82da-9fd6b369a461 ← NVME\VEN_1987&DEV_5018
│                         fcc92585-8e8e-52a3-90cc-6c022b7f3843 ← NVME\VEN_1987&DEV_5018&REV_01
│                         b1fcba59-6797-58bd-af77-5ccb9be2493c ← NVME\VEN_1987&DEV_5018&SUBSYS_19875018
│                         c9465066-32bb-50c5-8338-6a750cae2a44 ← NVME\VEN_1987&DEV_5018&SUBSYS_19875018&REV_01
│                         12d61493-5ac4-57f7-a2c5-96878d115455 ← Star Drive PCIe SSD
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs shutdown after installation
│                         • Device is usable for the duration of the update
│   
├─SuperIO FuSuperioIt89Device:
│     Device ID:          3ecf3fbe07ed8933226b294c8c19f4f9c8ebd71a
│     Summary:            Embedded controller
│     Current version:    01.01
│     Vendor:             DMI:Star Labs
│     Install Duration:   20 seconds
│     GUIDs:              55a8a029-ca9f-5b54-b68d-412281a2a84c ← SUPERIO\GUID_5c917039-d938-5c9a-b22a-9c392b1534f3
│                         d83a4cd0-fce6-58a8-93c7-bcee89b41dcf ← SuperIO-FuSuperioIt89Device
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Update requires a reboot
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Signed Payload
│   
└─TPM:
      Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
      Current version:    7.85.17.51968
      Vendor:             Infineon (TPM:IFX)
      GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
                          5eebb112-75ad-5536-b173-a11eb3399402 ← TPM\VEN_IFX&DEV_0000
                          ddf995da-1b32-5a8a-bc1b-8d5af4b38b51 ← TPM\VEN_IFX&MOD_SLB9670
                          6d81ab63-db2e-50ac-934f-6be9accf5e02 ← TPM\VEN_IFX&DEV_0000&VER_2.0
                          301555de-680d-5ddc-b995-7553fc9138f1 ← TPM\VEN_IFX&MOD_SLB9670&VER_2.0
      Device Flags:       • Internal device
   
```

Sean-StarLabs, 2023-03-12:
It sounds like you've installed flashrom in a different place; I've never tried it on Arch (PKGBUILD has always sufficed) but I think it should be `meson build --prefix=/usr`

sim31, 2023-03-13:
I used [this PKGBUILD from AUR](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=flashrom-git) to install, with a modification to take v1.3.0 tag instead of master (later tried to take master as well, but did not help). This PKGBUILD uses [arch-meson wrapper](https://github.com/archlinux/svntogit-packages/blob/packages/meson/trunk/arch-meson), which adds `--prefix=/usr` by default.

I even installed the flashrom 1.2 from main repos again, just to see if it instals to the same place. From pacman I can tell that both installations own the same filenames in the same locations.

Sean-StarLabs, 2023-03-13:
Not sure - using a Live USB would be the best bet until the Arch packages get sorted out (shouldn't be long given that it's Arch!)

closed 2023-03-23 by Sean-StarLabs