StarLabsLtd/firmware #85: [StarLite Mk IV] [Coreboot] Shutdown hangs the machine

ezaquarii, 2023-03-12:
Trying to shutdown the machine from Open/Free/NetBSD using `halt -p` hangs it.
Shutting it down from coreboot shell (no OS) using `reset` command hangs it.
Shutting down from OpenBSD bootloader using `machine poweroff` hangs it.

Shutting down from Linux works, suggesting that it must be related to some early hardware setup.

Sean-StarLabs, 2023-03-12:
We can't get involved with bsd but `reset -s` from the EFI shell should shut it down. Assuming that's what you meant, what EC version are you using?

ezaquarii, 2023-03-12:
@Sean-StarLabs 1.04 IIRC. How can I double-check the version?

ezaquarii, 2023-03-12:
@Sean-StarLabs 
```
# cat /sys/class/dmi/id/ec_firmware_release
1.4
```

ezaquarii, 2023-03-12:
@Sean-StarLabs I found something interesting. When booting Linux and doing `halt -p`, the system shuts down correctly and the bug goes away. I can properly shut it down:

1. EFI shell `reset -s`
2. OpenBSD bootloader `machine poweroff`
3. OpenBSD `halt -p`

**UNTIL** I power-off the machine during boot. 

The bug is triggered with the following procedure:

1. power cycle system
2. when coreboot prompt is shown (or GRUB), press power off button
3. power on again and enter EFI shell
4. `reset -s` leaves system with blue LEDs illuminated, `machine poweroff` and `halt -p` no longer work
5. boot Linux and `halt -p` to fix the problem

It looks like powering off the system from EFI leaves firmware in some weird state and the state is persisted between power cycles.

Sean-StarLabs, 2023-04-21:
`8.37` in testing __might help - we're quite limited on XHCI control in coreboot

ezaquarii, 2023-04-22:
@Sean-StarLabs 8.37 did not solve the problem. I observed another thing:

1. boot linux and `halt -p`
2. boot efi shell and `reset -s` to shutdown (works)
3. boot openbsd and `halt -p` (works)
4. boot openbsd and leave the system online for some time
5. `halt -p` will hang the machine (as well as `reset -s` in EFI).

Whatever Linux boot process does to the power management, I suspect firmware goes hairy in the background after some time.

Sean-StarLabs, 2023-04-23:
Have you tried using shutdown? I wonder if halt doesn't let coreboot know it's going down, so it doesn't reset XHCI

ezaquarii, 2023-04-24:
I tested both `halt` and `shutdown` on OpenBSD and both result in hang. I'll ask on the mailing list to see if there is a way to disable xhci via software.

I could also test if xhci can be disabled in firmware. Is it possible?

ezaquarii, 2023-04-25:
This problem happens in EFI firmware as well, when issuing `reset -s`. Does it perform shutdown via S5?

Sean-StarLabs, 2023-04-25:
I can't replicate that; can you make it happen without BSD in the equation?

> Does it perform shutdown via S5?

Not sure what you mean?

ezaquarii, 2023-04-25:
Yes, when I boot into EFI shell reset -s hangs it.

But as I mentioned in the original report, if you run Linux first and shut down machine from there, it "fixes" it somehow for a while.

I can take a look later if it's possible to trigger that from EFI. I think i did that by doing some hard reset during boot, but need to sit down at home and fiddle with it to give you a detailed procedure.

In the meantime, I propose that you download OpenBSD install img, write it to USB stick. When it boots, it prompts user to run shell or installer. Just choose shell and do halt -p. It should trigger the failure.

When you power it off, try EFI shell reset -s. It should fail as well.

Booting Linux should fix it.

ezaquarii, 2023-04-25:
1. run linux
2. shutdown as ususal
4. run EFI shell
5. `reset -s` powers off the machine
7. run EFI shell again
6. press power button to force poweroff
7. run EFI shell again
8. `reset -s` hangs the machine from now on
9. boot linux
10. shutdown with `halt -p` to "fix" the problem

ak-coram, 2023-06-22:
I have the same problem on Linux 6.3.8: shutdown doesn't turn the machine off completely and I have to hold the power button to do so. I have been encountering this issue since I've updated to the 8.50 BIOS.

Sean-StarLabs, 2023-06-22:
@ak-coram It's different, but now fixed in `8.51` - that's going up to the LVFS now so should be available in about 24 hours

ak-coram, 2023-06-22:
@Sean-StarLabs: thanks for the letting me know, does this also fix the issue where it sometimes doesn't come back from sleep (screen doesn't turn on)?

Sean-StarLabs, 2023-06-22:
Not aware of such an issue - was that only introduced in `8.50`?

ak-coram, 2023-06-22:
@Sean-StarLabs: I've had it since 8.37, it occurs pretty frequently (about 1 in 3 times) when waking from sleep.

ak-coram, 2023-06-22:
It might be also something from an earlier release, since I've skipped a bunch of updates before 8.37.

ak-coram, 2023-06-22:
@Sean-StarLabs: okay, thanks!

Sean-StarLabs, 2023-09-12:
Can't change this one; coreboot can only enumerate things as PCI devices, which the PMC isn't. Changing that would involving changing how coreboot works.

AMI won't have this issue though

closed 2023-09-12 by Sean-StarLabs

ezaquarii, 2023-09-13:
@Sean-StarLabs can you explain what the issue is? it's marked as completed but explaination suggests wontfix?

Sean-StarLabs, 2023-09-27:
Long story short, the Power Management Controller isn't accessible, which is needed to turn off the USB controller. coreboot can only handle PCI devices but the PMC on GLK isn't one. It was disabled years ago (https://review.coreboot.org/c/coreboot/+/15530) to get the GPIO controllers working.

So yes, it's a "wontfix", as it would involve rewriting coreboot.