StarLabsLtd/firmware #87: EC stills on 1.01 version with coreboot 8.32

cichy1173, 2023-03-14:
Hi!
I updated coreboot to 8.32 and this update ships EC / ITE 1.03. Readme says updating with DC Jack cable will also update EC automatically. I did that, but I have still EC 1.01. 

`fwupdmgr get-devices`

```bash
StarBook (bios):
│     Obecna wersja:      8.32
│     Dostawca:           Star Labs (DMI:coreboot)

SuperIO FuSuperioIt89Device:

      Podsumowanie:       Embedded controller
      Obecna wersja:      01.01
      Dostawca:           DMI:Star Labs
      Czas trwania instalacji:20 sekund

```

Sean-StarLabs, 2023-03-14:
How did you install 8.32?

Sean-StarLabs, 2023-03-14:
That'll be why - there'll be the odd update where updating with flashrom is fine, but generally, it's a bad idea.

It should be possible to trigger it manually, but it's dangerous if not followed to the letter, so EFI Shell would be your best bet now.

cichy1173, 2023-03-14:
> That'll be why - there'll be the odd update where updating with flashrom is fine, but generally, it's a bad idea.
> 
> It should be possible to trigger it manually, but it's dangerous if not followed to the letter, so EFI Shell would be your best bet now.

Can you tell me how to do it safely?

And how can I update bios safely next time? Of course without flashrom

Sean-StarLabs, 2023-03-14:
100% safely, no - only fwupd can do that. I can do 92.5% safely:

* Charger must be connected
* Battery must be charged to 30%
* No external devices
* Compile this with GCC:
```
#include <stdio.h>
#include <stdbool.h>
#include <stdint.h>
#include <unistd.h>
#include <sys/io.h>

/* From coreboot's src/include/pc80/mc146818rtc.h file */
#define RTC_BASE_PORT 0x70

/* 
 * This is the offset of the first of the two checksum bytes
 * we may want to figure out how we can determine this dynamically
 * during execution.
 */
#define CMOS_CHECKSUM_OFFSET 123

void cmos_checksum_write(uint16_t checksum);
void cmos_write_byte(uint8_t addr, uint8_t val);

int main(void)
{
	/* Call ioperm() to grant us access to ports 0x70 and 0x71 */
	ioperm(RTC_BASE_PORT, 2, true);

	/* Write a default value to the CMOS checksum */
	cmos_checksum_write(0xffff);
}

void cmos_checksum_write(uint16_t checksum)
{
	unsigned char lo, hi;

	/* The checksum is stored in a big-endian format. */
	hi = (unsigned char)(checksum >> 8);
	lo = (unsigned char)(checksum & 0x00ff);
	cmos_write_byte(CMOS_CHECKSUM_OFFSET, hi);
	cmos_write_byte(CMOS_CHECKSUM_OFFSET + 1, lo);
}

void cmos_write_byte(uint8_t addr, uint8_t val)
{
	/* Reject addresses in the second bank */
	if (addr >= 128)
		return;

	outb(addr, RTC_BASE_PORT);
	outb(val, RTC_BASE_PORT + 1);
}
```
* Run it with sudo
* Shutdown and then wait 8 minutes
* Turn it on and wait until you see the Star logo

> And how can I update bios safely next time? Of course without flashrom

With fwupd

cichy1173, 2023-03-14:
> 100% safely, no - only fwupd can do that. I can do 92.5% safely:

Ok, so I don't want to do this. 

Downgrading to 8.31 and upgrading to 8.32 is an option? Or waiting for newer version than 8.32?

Sean-StarLabs, 2023-03-14:
With fwupd? You could just re-install 8.32

cichy1173, 2023-03-14:
> With fwupd? You could just re-install 8.32

Is this 100% safe?  Also how can I do that? I updated only with flashrom in the past :/

Thank you for your answers <3

Sean-StarLabs, 2023-03-14:
https://starlabsltd.github.io/firmware/methods
`fwupdmgr reinstall` 

> Is this 100% safe? 

The software is - only thing you could do to "annoy" it would be disconnect the charger

Sean-StarLabs, 2023-03-14:
Cool - didn't know the flatpak worked, it never used to.

Easiest way to answer:
```

const struct fan_curve code quiet_mode = {
	//				1	2	3	4	5	6	7	8	9	10	11
	/* .start_temp	= */	{	0,	55,	60,	65,	70,	75,	80,	85,	90,	95,	100},
	/* .duty	= */	{	0,	0,	0,	20,	25,	30,	35,	40,	50,	60,	70}
};

const struct fan_curve code normal_mode = {
	//				1	2	3	4	5	6	7	8	9	10	11
	/* .start_temp	= */	{	0,	55,	60,	65,	70,	75,	80,	85,	90,	95,	100},
	/* .duty	= */	{	0,	0,	30,	35,	40,	45,	50,	55,	60,	65,	70}
};

const struct fan_curve code aggressive_mode = {
	//				1	2	3	4	5	6	7	8	9	10	11
	/* .start_temp	= */	{	0,	55,	60,	65,	70,	75,	80,	85,	90,	95,	100},
	/* .duty	= */	{	0,	20,	25,	30,	40,	50,	70,	80,	90,	95,	100}
};

/*
 * Hook:	500ms (b)
 * Purpose:	Update the fan duty
 */
void update_fan_rpm(void)
{
	struct fan_curve fan_profile;

	int required_duty		= 0;
	int i;
	int duty_change;
	int max_change;
	int new_duty;
	int fan_multiplier;

	/* Get the fan profile that is selected */
	switch (emem_fan_mode) {
	case FAN_MODE_0:
		fan_profile		= quiet_mode;
		fan_multiplier		= 1;
		break;
	case FAN_MODE_2:
		fan_profile		= normal_mode;
		fan_multiplier		= 8;
		break;
	default:
		fan_profile		= aggressive_mode;
		fan_multiplier		= 4;
                break;
	}

	for (i = 0; i < 11; i++) {
		if (emem_cpu_temperature < fan_profile.start_temp[i])
			break;
		required_duty = fan_profile.duty[i];
	}

	duty_change = required_duty - eram_fan_duty;
	max_change = (MIN_DUTY_CHANGE * fan_multiplier) * duty_change / 100;

	/* If the duty is descreasing, decrease in smaller increments */
	if (required_duty < eram_fan_duty)
		max_change = max_change / fan_multiplier;

	new_duty = floor(eram_fan_duty + max_change);
```

cichy1173, 2023-03-14:
Thank you for that code and help.

closed 2023-03-14 by cichy1173

david-mohr, 2023-03-20:
I have a related question about coreboot. I was also using `flashrom` and will now switch to `fwupdmgr` with `lvfs-testing` (glad I read this issue!). However, I have a LUKS encrypted drive. I remember reading about problems with EC updates and LUKS drives, but looking at the doco, the only mention is under "ITE requirements" and I'm not sure it that applies in this case (it reads like another type of BIOS?).

Do I simply use `fwupdmgr`, or is there something else that I need to be doing instead?

cichy1173, 2023-03-21:
> I have a related question about coreboot. I was also using `flashrom` and will now switch to `fwupdmgr` with `lvfs-testing` (glad I read this issue!). However, I have a LUKS encrypted drive. I [...]

I have encrypted drive too and everything works just fine.