StarLabsLtd/firmware #89: [StarBook Mk VI - Intel - Coreboot] Battery won't charge following ite update to 1.03

divico, 2023-03-18:
Following an update of the ite to 1.03,  the battery won't charge anymore. I tried the 65W GaN charger with usb-c and the jack, as well as two docking station, StarPort 12 and a Dell WD19TB. In all cases the gnome battery indicator is cyclically switching between charging and discharging and the battery is never charging. Also the LED next to the usb-c port remains always red. Also when powered-off the battery won't charge. 

https://user-images.githubusercontent.com/11789268/226138698-ef9ac237-acdc-4a4d-ad87-61acb4cbcffc.mp4

https://user-images.githubusercontent.com/11789268/226138705-43534e3c-b898-4bc9-8d66-1bc3da8a3856.mp4

https://user-images.githubusercontent.com/11789268/226138716-4d73f755-34dd-4c7d-99de-1c3773c4ba06.mp4

I made the update via fwupd and updated Coreboot from 8.31 to 8.32, which includes the update of the ite to 1.03. The update process went well and after the update fwupd reported the correct versions of 8.32 and 1.03, respectively.

Downgrading the ite to 1.02 using fwupd fixes the issue, the battery is charging normally. 

I'm not sure if the following may be related so in doubt I mention it:
After several reboots and a long press (>1min) on the power button with the laptop powered-off, fwupd reported a weird version number of the ite 01.1.07
`
─StarBook (bios):
│     Device ID:          dbee8bd3b1ae0316ad143336155651eedb495a0e
│     Current version:    8.32
│     Vendor:             Star Labs (DMI:coreboot)
│     Release Branch:     coreboot
│     Problems:           [31m[1m• System power is too low to perform the update[0m
│     GUIDs:              230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│                         7227cf32-bb0f-5894-8d66-c91a43c10eae ← FLASHROM\VENDOR_Star Labs&PRODUCT_StarBook&REGION_BIOS
│                         57184d37-4abe-517a-9039-5f6043f7c5a2 ← FLASHROM\GUID_5c917039-d938-5c9a-b22a-9c392b1534f3
│                         8d2aeb7b-6a2f-5f11-b149-0af5c2cbf6a0 ← Star Labs&B6-I&StarBook&B6-I&Star Labs&StarBook
│                         4b23d672-a538-5aad-af77-dfea6e4530c5 ← Star Labs&B6-I&StarBook&B6-I
│                         afe749a1-665c-5ed7-90d6-d89bcd0560cd ← Star Labs&B6-I&StarBook
│                         64640287-8a2e-5ad9-b7b2-ec7378f4297e ← Star Labs&B6-I&Star Labs&StarBook
│                         8c994a92-7ef8-5d68-80b5-99ead7cf4686 ← Star Labs&B6-I&StarBook&B6-I&coreboot
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs shutdown after installation
│                         • Cryptographic hash verification is available
│                         • Updatable
│                         • Device supports switching to a different branch of firmware
│   
├─SuperIO FuSuperioIt89Device:
│     Device ID:          3ecf3fbe07ed8933226b294c8c19f4f9c8ebd71a
│     Summary:            Embedded controller
│     Current version:    01.107
│     Vendor:             DMI:Star Labs
│     Install Duration:   20 seconds
│     Problems:           [31m[1m• System power is too low to perform the update[0m
│     GUIDs:              55a8a029-ca9f-5b54-b68d-412281a2a84c ← SUPERIO\GUID_5c917039-d938-5c9a-b22a-9c392b1534f3
│                         d83a4cd0-fce6-58a8-93c7-bcee89b41dcf ← SuperIO-FuSuperioIt89Device
│     Device Flags:       • Internal device
│                         • Update requires a reboot
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Updatable
│                         • Signed Payload

`
I'm available for debugging, however I'm not sure were to start.

Sean-StarLabs, 2023-03-18:
How did you install `8.32`?

divico, 2023-03-19:
Using fwupd. I enabled lvfs-testing and installed it using Gnome firmware. After the install, I rebooted and checked the coreboot and ite version reported by Gnome firmware, both were correct (i.e. 8.32, 1.03). I'am running Arch linux

autry7, 2023-03-19:
I also had this issue. I downgraded to 8.31 because of the SSD issue, but the ite stayed on 1.03. I didn't notice until the laptop died that it wouldn't charge. Downgraded ite through efi and battery charges as it should now.

autry7, 2023-03-19:
1.03 works with AMI bios as intended. Hopefully that helps narrow it down.

Sean-StarLabs, 2023-03-19:
It's not a problem with `1.03` itself; it's memory overflow. Usually, it's caused by EC writes when they're in programming mode - i.e. after `fwupd` tells you to shut down, you do something that'll save something to the EC memory. That can be anything from not entering S0, entering S3/S4, changing the keyboard backlight brightness, disabling the trackpad etc.

You said you rebooted - if that was a indeed a reboot rather than a shutdown - mystery solved. 

If not, I'd take a stab in the dark and say Arch have added a feature and not to told anyone (just as they've done it before with mtd).

berger321, 2023-03-20:
I have a similar issue. I have downgraded the bios version but how did you downgrade the EC with fwupd?

Sean-StarLabs, 2023-03-20:
You can't - it can only be done in the EFI Shell

divico, 2023-03-20:
> It's not a problem with `1.03` itself; it's memory overflow. Usually, it's caused by EC writes when they're in programming mode - i.e. after `fwupd` tells you to shut down, you do something that'll [...]

Thanks for the input Sean. I reinstalled this morning both 8.32 and 1.03 and I took care to not do anything that would involve the EC memory. I haven't seen the issue I reported earlier for the whole day, so from my side we can close this issue. I apologize for the noise. I can't  tell what I did after the first upgrade that would involve the EC memory and lead to the issue though.

commit 7d9f3ff0 (2023-03-20): Added StarBook MkVI-Intel ite 1.04

Fixes #89
closed 2023-03-20 by Sean-StarLabs