StarLabsLtd/firmware #91: Suspend doesn't work, coreboot firmware Starbook 2022

aaron-gh, 2023-03-22:
Tried on both stable and testing. No suspend states s2idle or deep work. Work fine under AMI
Tried with latest linux mainline kernel as well as lts

Sean-StarLabs, 2023-03-22:
Going to need some more info as it categorically works in our tests. 

How are you testing it?
What do you see (i.e. what happens when you open the lid)?
What distro are you using?
Do you have any external devices connected?
Does the charger being connected make any difference?
What SSD/memory do you have?

aaron-gh, 2023-03-22:
Here's some more info:

When I open the lid, the system is on the lock screen, however it reaches that state before I open the lid (it goes to lock screen as soon as I close it or issue any other call to suspend) I know this because I use a screen reader.

Currently running Ubuntu mate 22.04

It does not make a difference if the charger is connected

eOutput of some commands that might be useful:

uname -r:
6.1.13-060113-generic

lspci -k
00:00.0 Host bridge: Intel Corporation Device 4621 (rev 02)
	Subsystem: Intel Corporation Device 7270
	Kernel driver in use: igen6_edac
	Kernel modules: igen6_edac
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
	DeviceName: VGA compatible controller
	Subsystem: Intel Corporation Alder Lake-P Integrated Graphics Controller
	Kernel driver in use: i915
	Kernel modules: i915
00:08.0 System peripheral: Intel Corporation 12th Gen Core Processor Gaussian & Neural Accelerator (rev 02)
	Subsystem: Intel Corporation Device 7270
00:14.0 USB controller: Intel Corporation Alder Lake PCH USB 3.2 xHCI Host Controller (rev 01)
	Kernel driver in use: xhci_hcd
	Kernel modules: xhci_pci
00:14.2 RAM memory: Intel Corporation Alder Lake PCH Shared SRAM (rev 01)
	Subsystem: Intel Corporation Device 7270
00:15.0 Serial bus controller: Intel Corporation Alder Lake PCH Serial IO I2C Controller #0 (rev 01)
	Subsystem: Intel Corporation Device 7270
	Kernel driver in use: intel-lpss
	Kernel modules: intel_lpss_pci
00:1c.0 PCI bridge: Intel Corporation Device 51bc (rev 01)
	Kernel driver in use: pcieport
00:1d.0 PCI bridge: Intel Corporation Device 51b0 (rev 01)
	Kernel driver in use: pcieport
00:1e.0 Communication controller: Intel Corporation Alder Lake PCH UART #0 (rev 01)
	Subsystem: Intel Corporation Device 7270
	Kernel driver in use: intel-lpss
	Kernel modules: intel_lpss_pci
00:1f.0 ISA bridge: Intel Corporation Alder Lake PCH eSPI Controller (rev 01)
	Subsystem: Intel Corporation Device 7270
00:1f.3 Audio device: Intel Corporation Alder Lake PCH-P High Definition Audio Controller (rev 01)
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel, snd_sof_pci_intel_tgl
00:1f.4 SMBus: Intel Corporation Alder Lake PCH-P SMBus Host Controller (rev 01)
	Subsystem: Intel Corporation Device 7270
	Kernel driver in use: i801_smbus
	Kernel modules: i2c_i801
00:1f.5 Serial bus controller: Intel Corporation Alder Lake-P PCH SPI Controller (rev 01)
	Subsystem: Intel Corporation Device 7270
	Kernel driver in use: intel-spi
	Kernel modules: spi_intel_pci
01:00.0 Network controller: Intel Corporation Wi-Fi 6 AX210/AX211/AX411 160MHz (rev 1a)
	Subsystem: Intel Corporation Wi-Fi 6 AX210 160MHz
	Kernel driver in use: iwlwifi
	Kernel modules: iwlwifi
02:00.0 Non-Volatile memory controller: Phison Electronics Corporation E18 PCIe4 NVMe Controller (rev 01)
	Subsystem: Phison Electronics Corporation E18 PCIe4 NVMe Controller
	Kernel driver in use: nvme
	Kernel modules: nvme

I have 32gb ram and the PCIE gen 4 1TB stardrive.

It does not matter what Distro/Kernel I use, tried with debian stable/testing/unstable, ubuntu 22.04/22.10, Arch linux, Fedora 37.

aaron-gh, 2023-03-22:
Couple more pieces of info that might help:
Neither mem sleep or s2idle work as sleep states, tested by doing the following:
Tried lid close/systemctl suspend first, as that is what the distro uses.
Then tried writing to the kernel directly with:
`# echo "mem" > /sys/power/state`
`# echo "s2idle" > /sys/power/mem_sleep`
`# echo "mem" > /sys/power/state`
`# echo "freeze" > /sys/power/state`

Sean-StarLabs, 2023-03-23:
> When I open the lid, the system is on the lock screen, however it reaches that state before I open the lid (it goes to lock screen as soon as I close it or issue any other call to suspend) I know this because I use a screen reader.

Sounds like software in play - are all of these tests with the same screen reader installed?

aaron-gh, 2023-03-23:
No, tests were also conducted with screen readers and all other assistive technology tools disabled.

It is demonstratible by playing a sound file, and then requesting the computer enter a sleep state.
You can hear a speaker click, and the audio will stop for a moment. The speakers will click again and the audio resumes. This is on a fresh install, all updates completed, playing a media file with the built-in media player.
It can also be seen by connecting an external device. The device will power down for a moment, and then power back up, but again, even without external devices connected, this happens
Using the AMI bios makes the issue go away entirely.

Sean-StarLabs, 2023-03-23:
Okay - in the case, can you capture a `dmesg` whilst try to suspend? i.e. `sudo dmesg -wH > dmesg.txt`, suspend, wait 60 seconds, resume, stop.

aaron-gh, 2023-03-23:
Hi,
Here is the requested log.
[dmesg.txt](https://github.com/StarLabsLtd/firmware/files/11049243/dmesg.txt)

Sean-StarLabs, 2023-03-23:
Thanks, any USB devices connected?

aaron-gh, 2023-03-23:
No.

closed 2023-03-28 by Sean-StarLabs

aaron-gh, 2023-04-05:
This is still an issue with latest coreboot from testing and ite 1.0.7

Sean-StarLabs, 2023-04-05:
What version is latest coreboot?

aaron-gh, 2023-04-05:
8.36

aaron-gh, 2023-04-13:
Still present on 8.37.

Sometimes, instead of exiting s3 as soon as it is entered, the system will reboot instead. There is no pattern to either failure, 1 or the other happens every time.

Sean-StarLabs, 2023-04-13:
Can you post a new dmesg please: