StarLabsLtd/firmware #93: Starbook MK V: Excessive Power Draw from USB-C Display

stan-janssen, 2023-03-27:
When I connect my StarBook MK V to a Dell S2719DC over USB-C, the monitor shuts down the connection with an "Excessive Power Draw from connected device" warning. The monitor is capable of delivering 45W PD, and I think the laptop want to draw 65W. 

My other laptop, a Thinkpad T15, works fine with this display.

Could it be that the Starbook does not take into account the maximum power delivery from the USB-C power source? Would it be possible to include this functionality? 

At the moment, I cannot use this display with this laptop when the laptop is not connected to its own power brick over the barrel jack.

Sean-StarLabs, 2023-03-27:
Which variant and firmware versions?

stan-janssen, 2023-03-27:
This is the Intel i7 variant with firmware version 8.15.

stan-janssen, 2023-03-27:
```
$ cat /sys/class/dmi/id/ec_firmware_release
1.3
```

stan-janssen, 2023-03-27:
Yes, that seems to work! Thanks.

It does require rebooting though (at least, that is what the Coreboot application tells me); is there something that can be done to auto-detect the max power from the power source and adjust the laptop's power draw accordingly?

Sean-StarLabs, 2023-03-27:
Not in coreboot; AMI can

stan-janssen, 2023-03-27:
Is this something that I should report to the Coreboot developers or is that not helpful?

https://ticket.coreboot.org/projects/coreboot

closed 2023-03-28 by Sean-StarLabs