StarLabsLtd/firmware #96: Starbook MkVI - Intel - coreboot: reboot during sleep

donnekgit, 2023-03-30:
I've been having a recurrent issue where when the laptop is asleep it will (sometimes) reboot.  When you awaken it, you get the following message about uefi.shell:
[image]

I installed fwupdmgr and updated the firmware, and although this didn't stop the rebooting, it booted back to the login screen instead of that uefi message (though a few of these did appear too, ie it doesn't go to login screen every time).

It's not a major issue, because the laptop boots so fast, but it can be a bit of a pain if you have been doing some writing or coding, and forget to save - that work is then lost.

Any idea what might be causing this?

Sean-StarLabs, 2023-03-30:
What version?

donnekgit, 2023-03-30:
Sorry:
libflashrom1:  1.2-10+j
fwupd 1.8.9+j
fwupd-signed: 1.51~22.04.1+1.2-3ubuntu0.2
Star Labs StarBook (bios): 8.25

I've just seen there's an 8.31 for the bios, so should I try updating that?

Sean-StarLabs, 2023-03-30:
Yes please

donnekgit, 2023-03-30:
OK, done that.  I'll leave it for a few days and see if the issue recurs. :-)

donnekgit, 2023-04-01:
OK.  Since then, I've had two reboots. This is with an external HDMI screen attached, and using KDE → Power/Session → Sleep.  There seems to be no consist reason for why it sometimes wakes up from sleep OK, and other times has rebooted.

Sean-StarLabs, 2023-04-01:
So the original issue no longer occurs?

donnekgit, 2023-04-01:
It seems not to do that (much) any more - it goes from the StarLabs boot logo straight into the Kubuntu login screen.

Sean-StarLabs, 2023-04-01:
Hmm, when it goes to EFI shell now, do you see an FS0 listed?

The rebooting behaviour will be #91 - planning to do a stable release for that on Monday but thought the SSD initialisation one was done

donnekgit, 2023-04-02:
It's only done that once recently, and that was the pic I sent.

What I'm trying now is unplugging the HDMI cable to the other screen, and then just closing the lid to put it to sleep, rather than select Sleep.

Sean-StarLabs, 2023-04-02:
I got from your first message that was what was happening every time - so just to confirm, since 8.31, that no longer happens, right?

donnekgit, 2023-04-02:
Sorry.  That WAS happening, then I updated the firmware to 8.25 about 3 weeks, and it seemed to use the login screen after that.  But on Thursday it did use the EFI screen.  Since updating to 8.31 a couple of days ago, it has only used the login screen on the two reboots it has done since then.

Sean-StarLabs, 2023-04-04:
Can you try `8.36`? That's got the fix for #91 so should work, assuming you're using S3 (the default suspend to ram) rather than S4 (hibernate).

donnekgit, 2023-04-05:
The current firmware is 8.31, but 8.36 doesn't seem to be available:
```
$ fwupdmgr update
Devices with no available firmware updates: 
 • Star Drive PCIe SSD
 • SuperIO FuSuperioIt89Device
Devices with the latest available firmware version:
 • StarBook (bios)
 • Unifying Receiver
```

cichy1173, 2023-04-05:
> The current firmware is 8.31, but 8.36 doesn't seem to be available:
> 
> ```
> $ fwupdmgr update
> Devices with no available firmware updates: 
>  • Star Drive PCIe SSD
>  • SuperIO FuSuperioIt89Device
> Devices with the latest available firmware version:
>  • StarBook (bios)
>  • Unifying Receiver
> ```

8.36 is available in testing branch. You need to switch to this branch first.

donnekgit, 2023-04-06:
OK.  Installed that, and it's a backward step.  Now, when it awakes from sleep, there's nothing on the screen (just black), no way of logging in, etc.  You have to manually switch the machine off and then reboot.  I've reverted to 8.31.

Sean-StarLabs, 2023-04-11:
Ah, I'd fixed this before but the patch got messed up when it was upstreamed. Should be fixed now in `8.37`

donnekgit, 2023-04-12:
Tried 8.37, but although it came up fine the first time I used sleep, the second time it just produced the blank screen, requiring a reboot.  So I've downgraded to 8.31 again.   I think it's something to do with a HDMI screen being attached, so over the next week I'm going to pull out the HDMI cable before leaving the laptop, and just close the lid rather than select the sleep option, and then I can see if there are any reboots.  If there aren't, that would confirm that it's a HDMI issue.

Sean-StarLabs, 2023-04-14:
Interesting - Can you try with HDMI connected and use the terminal command `sudo systemctl suspend`?

donnekgit, 2023-04-15:
I've done that over two days, and it seems to work, but I need another few days to test fully.

Sean-StarLabs, 2023-04-23:
Adjustments in `8.40` should handle that, which is now stable, so I'll close this for now. Please re-open if it occurs on that release.

donnekgit, 2023-04-26:
URGENT: Laptop is now unbootable.

I went to update to 8.40, couldn't see it in stable, and then went to testing and installed 8.41.  The machine is now unbootable, and showing the following:

[image]

Can you let me know asap how to revert from this, since I need to work on the laptop.

In the longer term, would it be a good idea to switch from coreboot to AMI?

cichy1173, 2023-04-26:
> URGENT: Laptop is now unbootable.
>
> I went to update to 8.40, couldn't see it in stable, and then went to testing and installed 8.41. The machine is now unbootable, and showing the following: [...]

Coreboot 8.40 is stable, 8.41 is `testing`. Check bootable devices in bios, maybe OS does not recognize your drive (looks like that)

donnekgit, 2023-04-26:
I couldn't see 8.40 in stable.  Can you give me a bit more info?  Hold down DEL while booting?  And if BIOS doesn't recognise the drive, what then?

cichy1173, 2023-04-26:
> I couldn't see 8.40 in stable. Can you give me a bit more info? Hold down DEL while booting? And if BIOS doesn't recognise the drive, what then?

If I remember right, you should click F2 when Star Labs logo appears. Then go to Bootable devices or something like that and look for drive.

about updates you can read here: https://fwupd.org/lvfs/devices/com.starlabs.B6-I.coreboot and here: https://cichy1173.github.io/starbook_blog/

donnekgit, 2023-04-26:
Phew!  Thanks for coming back so quickly.  The UEFI entry was first on the list, so once I chose the disk entry it booted, and I was then able to downgrade to 8.40.  I think I'd forgotten to refresh first, and that was why 8.40 didn't show on the list.  Maybe there should be a note added to https://starlabsltd.github.io/firmware/methods/lvfs_terminal to say that a refresh is essential before update, just to remind people?

The only USB item attached at the time of the photo was a USB mouse.

Anyway - thanks again.

Sean-StarLabs, 2023-04-26:
Cool

> I think I'd forgotten to refresh first, and that was why 8.40 didn't show on the list.

It can't be that if you could see `8.41`, as that was added after `8.40`.

> The only USB item attached at the time of the photo was a USB mouse.

In that case, In that photo, `fs0:` would have been your SSD.

donnekgit, 2023-04-26:
> It can't be that if you could see 8.41, as that was added after 8.40.

Hmm. That's what I thought, but it definitely wasn't visible on either stable or testing (I checked twice in each place).  It only became available after downgrading to 8.31 (of fond memory) and running refresh.

> In that case, In that photo, fs0: would have been your SSD.

Oh, OK.  I'll remember that.

fnetX, 2023-04-29:
I am running 8.40 but a suspend via my desktop environment caused a reboot upon waking up again. I don't quite know what is the recommended solution. Should I use `systemctl suspend` instead?

Sean-StarLabs, 2023-04-29:
Can't say without knowing more details; using the terminal will just prove it's a software bug

fnetX, 2023-05-04:
I used `systemctl suspend` recently, and just now hit this problem again. After waking up the laptop, the screen was blank (backlight on) for a moment, then the boot procedure began.

<details><summary>

`fnetx@starbook:~$ sudo journalctl -b-1 --no-pager --lines 70`

</summary>

~~~
May 04 22:30:01 starbook.m.codeberg.org CRON[179823]: pam_unix(cron:session): session closed for user root
May 04 22:33:38 starbook.m.codeberg.org systemd[1]: Started anacron.service - Run anacron jobs.
May 04 22:33:38 starbook.m.codeberg.org anacron[179869]: Anacron 2.3 started on 2023-05-04
May 04 22:33:38 starbook.m.codeberg.org anacron[179869]: Normal exit (0 jobs run)
May 04 22:33:38 starbook.m.codeberg.org systemd[1]: anacron.service: Deactivated successfully.
May 04 22:35:01 starbook.m.codeberg.org CRON[179912]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
May 04 22:35:01 starbook.m.codeberg.org CRON[179914]: (root) CMD (command -v debian-sa1 > /dev/null && debian-sa1 1 1)
May 04 22:35:01 starbook.m.codeberg.org CRON[179912]: pam_unix(cron:session): session closed for user root
May 04 22:38:51 starbook.m.codeberg.org wpa_supplicant[1104]: wlp1s0: WPA: Group rekeying completed with 9c:c7:a6:d8:ae:03 [GTK=CCMP]
May 04 22:45:01 starbook.m.codeberg.org CRON[179982]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
May 04 22:45:01 starbook.m.codeberg.org CRON[179983]: (root) CMD (command -v debian-sa1 > /dev/null && debian-sa1 1 1)
May 04 22:45:01 starbook.m.codeberg.org CRON[179982]: pam_unix(cron:session): session closed for user root
May 04 22:48:51 starbook.m.codeberg.org wpa_supplicant[1104]: wlp1s0: WPA: Group rekeying completed with 9c:c7:a6:d8:ae:03 [GTK=CCMP]
May 04 22:52:13 starbook.m.codeberg.org rtkit-daemon[1333]: Supervising 9 threads of 6 processes of 1 users.
May 04 22:52:13 starbook.m.codeberg.org rtkit-daemon[1333]: Supervising 9 threads of 6 processes of 1 users.
May 04 22:52:23 starbook.m.codeberg.org systemd[1505]: Started vte-spawn-0f289fb5-37ff-4f32-b7bb-6c8aeecf778b.scope - VTE child process 180127 launched by gnome-terminal-server process 2358.
May 04 22:52:25 starbook.m.codeberg.org polkitd[1032]: Registered Authentication Agent for unix-process:180134:19967732 (system bus name :1.851 [/usr/bin/pkttyagent --notify-fd 5 --fallback], object path /org/freedesktop/PolicyKit1/AuthenticationAgent, locale en_GB.UTF-8)
May 04 22:52:25 starbook.m.codeberg.org systemd-logind[1044]: The system will suspend now!
May 04 22:52:25 starbook.m.codeberg.org NetworkManager[1094]: <info>  [1683233545.5534] manager: sleep: sleep requested (sleeping: no  enabled: yes)
May 04 22:52:25 starbook.m.codeberg.org NetworkManager[1094]: <info>  [1683233545.5535] device (p2p-dev-wlp1s0): state change: disconnected -> unmanaged (reason 'sleeping', sys-iface-state: 'managed')
May 04 22:52:25 starbook.m.codeberg.org ModemManager[1127]: <info>  [sleep-monitor-systemd] system is about to suspend
May 04 22:52:25 starbook.m.codeberg.org NetworkManager[1094]: <info>  [1683233545.5540] manager: NetworkManager state is now ASLEEP
May 04 22:52:25 starbook.m.codeberg.org NetworkManager[1094]: <info>  [1683233545.5541] device (wlp1s0): state change: activated -> deactivating (reason 'sleeping', sys-iface-state: 'managed')
May 04 22:52:25 starbook.m.codeberg.org polkitd[1032]: Unregistered Authentication Agent for unix-process:180134:19967732 (system bus name :1.851, object path /org/freedesktop/PolicyKit1/AuthenticationAgent, locale en_GB.UTF-8) (disconnected from bus)
May 04 22:52:25 starbook.m.codeberg.org dbus-daemon[1014]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.10' (uid=0 pid=1094 comm="/usr/sbin/NetworkManager --no-daemon")
May 04 22:52:25 starbook.m.codeberg.org systemd[1]: Starting NetworkManager-dispatcher.service - Network Manager Script Dispatcher Service...
May 04 22:52:25 starbook.m.codeberg.org dbus-daemon[1014]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
May 04 22:52:25 starbook.m.codeberg.org systemd[1]: Started NetworkManager-dispatcher.service - Network Manager Script Dispatcher Service.
May 04 22:52:25 starbook.m.codeberg.org kernel: wlp1s0: deauthenticating from 9c:c7:a6:d8:ae:03 by local choice (Reason: 3=DEAUTH_LEAVING)
May 04 22:52:25 starbook.m.codeberg.org dbus-daemon[1533]: [session uid=1000 pid=1533] Activating service name='org.cinnamon.ScreenSaver' requested by ':1.494' (uid=1000 pid=180142 comm="/usr/bin/python3 /usr/bin/cinnamon-screensaver-com")
May 04 22:52:25 starbook.m.codeberg.org org.cinnamon.ScreenSaver[180169]: Fractional scaling active: False
May 04 22:52:25 starbook.m.codeberg.org org.cinnamon.ScreenSaver[180169]: Trying to connect to logind...
May 04 22:52:25 starbook.m.codeberg.org dbus-daemon[1533]: [session uid=1000 pid=1533] Successfully activated service 'org.cinnamon.ScreenSaver'
May 04 22:52:25 starbook.m.codeberg.org org.cinnamon.ScreenSaver[180169]: Starting screensaver...
May 04 22:52:25 starbook.m.codeberg.org wpa_supplicant[1104]: wlp1s0: CTRL-EVENT-DISCONNECTED bssid=9c:c7:a6:d8:ae:03 reason=3 locally_generated=1
May 04 22:52:25 starbook.m.codeberg.org wpa_supplicant[1104]: wlp1s0: CTRL-EVENT-DSCP-POLICY clear_all
May 04 22:52:25 starbook.m.codeberg.org NetworkManager[1094]: <info>  [1683233545.7954] device (wlp1s0): supplicant interface state: completed -> disconnected
[... cut ...]
May 04 22:52:26 starbook.m.codeberg.org systemd[1]: syncthing-resume.service: Deactivated successfully.
May 04 22:52:26 starbook.m.codeberg.org systemd[1]: Finished syncthing-resume.service - Restart Syncthing after resume.
May 04 22:52:26 starbook.m.codeberg.org systemd-sleep[180234]: Entering sleep state 'suspend'...
May 04 22:52:26 starbook.m.codeberg.org kernel: PM: suspend entry (deep)
May 04 22:52:26 starbook.m.codeberg.org wpa_supplicant[1104]: wlp1s0: CTRL-EVENT-DSCP-POLICY clear_all
~~~

I hope this helps ...
</details>

fnetX, 2023-05-04:
I noticed that:

~~~
cat /sys/class/dmi/id/ec_firmware_release
1.1
~~~

Is this a problem? Reading here https://starlabsltd.github.io/firmware/started I still miss the context of what all these different things (EC, ITE etc) are about and what I need. I follow the upgrade procedure for coreboot via lvfs / gnome-firmware or fwupdmgr.

Sean-StarLabs, 2023-05-04:
> I hope this helps ...

What distro are you using?

> Is this a problem?

No, EC doesn't influence S3

fnetX, 2023-05-04:
Debian GNU/Linux 12 (Bookworm).

fnetX, 2023-05-17:
I'm still greeted with a reboot from time to time. Is there anything I should try or further logs / details I could provide?

donnekgit, 2023-05-17:
Confirmed.  The number of reboots is way down (it's maybe only 20% of what it was), but still happens every so often.

donnekgit, 2023-05-22:
Not sure if this is still being looked at (unclear how to reopen a topic once closed), but I just wanted to log what seems to be a pattern.  If I am attached to a HDMI screen (Acer K242HL), and then put the laptop to sleep using K → Sleep, and then decide to move to another room and unplug the HDMI cable, when I then open the laptop to resume from sleep it reboots fairly consistently.  So I think there may be some sort of HDMI issue.  Can any of you other guys test this?

Sean-StarLabs, 2023-05-31:
I always thought anyone could re-open an issue, seems like it's not possible to enable.

Anyway, could you please try `8.43` with any EC higher than `1.11`?

fnetX, 2023-05-31:
Will do. Can you point me on how to update the EC? I didn't yet come across note about this, I fear.

Sean-StarLabs, 2023-06-01:
> Will do. Can you point me on how to update the EC? I didn't yet come across note about this, I fear.

https://support.starlabs.systems/kb/firmware/efi-shell

Thanks for the info

donnekgit, 2023-06-01:
> Anyway, could you please try `8.43` with any EC higher than `1.11`?

Sure.  Away from the machine atm, but will do asap.  Thanks.

fnetX, 2023-06-01:
~~~
$ cat /sys/class/dmi/id/ec_firmware_release
1.13
~~~

But how do I upgrade to 8.43 safely? I only see LVFS / fwupd as supported method, and the newest release offered is 8.41.

Sean-StarLabs, 2023-06-22:
@fnetX Can you try `8.51`? On it's way to the LVFS now so should be available in about 24 hours

Sean-StarLabs, 2023-07-12:
Can you try now?

fnetX, 2023-07-16:
Installed it. The "coreboot-configurator" broke with

~~~
nvramtool: CMOS option table not found in coreboot table.  Apparently, the coreboot installed on this system was built without selecting CONFIG_USE_OPTION_TABLE.
~~~

and I'll test now.

What really tackles me is the fact that often when this random reboot happens, external USB is broken. Everything slows to a halt, the countdown in EFI shell alike "Will continue in 5 seconds" takes literally forever to count down one second, same with GRUB, and input via external keyboard does not work until I plug it off and on again. When plugging it off during countdown in GRUB / EFI Shell, it continues normally.

Also, I recently had a problem when awaking from standby failed with disk errors. First, the screensaver failed to continue after resume and crashed. Switching to tty0 gave me some disk i/o errors:
[image]

After reboot via sysrq, the EFI shell appeared:

[image]

And the boot from file dialog was also empty:

[image]

This happens often during the standby issue: Selecting "Boot from file" directly does not work, it does not show the drive.

I have to properly power off the laptop, and power it on again. Then everything works as expected. I somehow link these issues together, and suspect it might either be related to the drive, the USB network, or both.

Sean-StarLabs, 2023-07-16:
Interesting, two questions:
What SSD are you using?
What external devices are connected and to where?

fnetX, 2023-07-17:
I am using the 500 GB Star Drive PCIe SSD.

I am switching between two setups, both with a mouse and keyboard connected to a USB hub, which is connected to the USB2.0 port at the right.

I sometimes saw the rebooting without this setup, but I do not recall if I just unplugged it from the setup when in standby. I'll try to gather precise environments each time the bug happens, but since it doesn't happen all too often and in unexpected situations ... It's like taking the laptop from the bag, ah it reboots instead from returning from standby - I don't always recall the setup when I put it into standby.

Sean-StarLabs, 2023-07-17:
So just USB 2.0 devices in the equation? No displays, charger etc.

Sean-StarLabs, 2023-07-21:
Found something peculiar which is fixed in `8.60`, not sure if it was actually causing an issue but it would explain it if it was

Sean-StarLabs, 2023-09-12:
Assuming solved, re-tested with all distros on `8.94`. If it's present with that, please open a new issue

closed 2023-09-12 by Sean-StarLabs

cichy1173, 2023-09-19:
> Assuming solved, re-tested with all distros on `8.94`. If it's present with that, please open a new issue

Will 8.94 appear in fwupd as a stable release? Last version of coreboot available in fwupd (lvfs) is 8.80 (as a testing)

Sean-StarLabs, 2023-09-20:
Yes, should be there this afternoon

cichy1173, 2023-10-01:
> Yes, should be there this afternoon

I see only 8.95 marked as lvfs-testing, not stable.

[image]

Sean-StarLabs, 2023-10-02:
Sorry - I'll push 8.97 stable tomorrow morning (left it doing S3 cycles) - few other very small fixes in there.

fnetX, 2023-11-17:
Hi, my starbook was in service a while back, and afterwards the issue was worse than ever. More than half of the times putting my laptop to standby would break it. Interestingly, sometimes it did not immediately cause the efi shell screen as sometimes pictured above, but allow me to "use" the laptop, but the filesystem was throwing I/O errors all the time (no reading or writing possible, thus no access to logfiles and similar). Rebooting the machine then also lead to the EFI error screen.

What I finally did: I toggled the "Third-party SSD support" option in coreboot. I had no trouble afterwards (for over a month now). I use the official StarDrive SSD shipped with my laptop, though. Just giving this as a pointer if this issue still exists.

I am on 8.95 after the service.