StarLabsLtd/firmware #98: problems after upgrading to 8.37 on lite mkIII and labtop mk3

barbastrejo12, 2023-04-12:
I have been running debian stable on both machines. The lite mkIII after the upgrading only starts with the power cord connected. The Labtop mkIII does not boot the OS it goes directly to the uefi shell, in the coreboot bios interface it does not see debian as a boot option anymore

Sean-StarLabs, 2023-04-12:
You'll need to re-add the Debian boot entry:
```
sudo efibootmgr -c -w -d /dev/sda1 -p 1 -L debian -l \\EFI\\debian\\grubx64.efi
```
I'll make a new build for the Lite

barbastrejo12, 2023-04-12:
`sudo efibootmgr -c -w -d /dev/sda1 -p 1 -L debian -l \\EFI\\debian\\grubx64.efi`
that command outputs this:
`Could not prepare Boot variable: No such file or directory`

barbastrejo12, 2023-04-12:
BootCurrent: 0003
Timeout: 2 seconds
BootOrder: 0004,0000,0001,0002
Boot0000* Enter Setup
Boot0001* UEFI PCIe SSD DDC30789174C00029358 1
Boot0002* UEFI Shell
Boot0003* grubx64.efi
Boot0004* debian

a file was created named debian but it seems empty. I rebooted and still it went directly to uefi shell. I tried adding the option manually in the coreboot menu as I did before in the past and it worked, but this time it seems to not hold the changes. I could roll back to the previous version in any case(if it`s safe to do so).

barbastrejo12, 2023-04-13:
Problem solved, I saw another person's related problem on a lite mkIII on Debian forum and it fixed my problem on the Labtop mkIII as well.
https://forums.debian.net/viewtopic.php?t=152138&hilit=uefi
I will be waiting for the new version of the lite mkIII, thanks.

closed 2023-04-13 by barbastrejo12