StarLabsLtd/firmware #99: After upgrade to 8.38 on lite mkIII it does not boot anymore, with or without the powercord

barbastrejo12, 2023-04-15:
After upgrade to 8.38 on lite mkIII it does not boot anymore, with or without the powercord. I remember similar issues in the past, should I unplug the battery and wait 60 seconds?

barbastrejo12, 2023-04-17:
now, maybe because battey went to zero, after plugging the cable and pressing power button, the power led holds but the screen remains black.

Sean-StarLabs, 2023-04-17:
Argh, I know what that it - short version, it'll need a re-flash. Did you have a programmer?

barbastrejo12, 2023-04-17:
I have the ch341a

Sean-StarLabs, 2023-04-17:
The tom caching interferes with the IFWI version, triggering the anti-rollback. 

If you flash it with the ch341a_spi back to `8.37`, or any other version, that'll bring it back

barbastrejo12, 2023-04-17:
8.37 only starts with power cord, I will flash the 8.20, it has fewer bugs. Is there some documentation of the lite mkIII? I am not sure which one is the chip to clip with the programmer

closed 2023-04-21 by Sean-StarLabs