StarLabsLtd/firmware #101: [Starbook MkVI - Intel][coreboot] Third-party SSDs

bryanvobo, 2023-04-21:
I've been using firmware version 8.31 (stable) for a couple of months with a Samsung-brand NVMe SSD. I've tried each of the "testing" releases since that one but 8.31 is the only version that recognizes and boots from my SSD. I was hopeful that the most recent 8.40 (stable) version would also work but it does not.

The changelog entries for version 8.32 include:
> Remove the hotplug workaround for SSDs and set coreboot to not disable PCI devices

Can that be re-added into the next release, or would that have adverse effects on other PCI devices? Is there another artificial slowdown (`sleep`) that could be introduced to allow enough time for third-party drives to be detected?

Sean-StarLabs, 2023-04-21:
The workaround that did it for the WD drives is still there, bound to a CMOS option. Try `sudo nvramtool -w pcie_hot_plug=Enable`

bryanvobo, 2023-04-21:
I flashed 8.40, booted from a USB stick, ran `sudo nvramtool -w pci_hot_plug=Enable` (`pci` instead of `pcie`), verified it with `nvramtool -a`, rebooted, and it didn't recognize my SSD.

Sean-StarLabs, 2023-04-21:
Hmmm, it's likely the other boot optimisations mean it's still too quick with that workaround (it was 7.2 seconds with that workaround, now it's 1.1)., I'll try to add an option to specify the amount of delay

Sean-StarLabs, 2023-04-23:
Can you try `8.41`? That, with `pci_hot_plug=Enable` might be enough

bryanvobo, 2023-04-23:
That worked! `8.41` alone didn't see the SSD but adding `pci_hot_plug=Enable` got it detected.

Note: "UEFI Shell" had top priority in the boot order so I needed to rearrange it so that the SSD was booted from first.

Thanks Sean!