StarLabsLtd/firmware #102: [Starbook MkVI - Intel][coreboot] Boot entries lost after firmware upgrade

fnetX, 2023-04-25:
Hey,

I'm having a StarBook VI Intel with coreboot. Recently, I was frequently updating coreboot, because I am also affected by #96 and I hoped for a fix (running 8.40 since today).

However, there is a big inconvenience: After each upgrade, the boot entry from Grub (Debian 12) is lost. After the firmware upgrade, I am greeted with the EFI shell, and `exit`ing brings me to a boot manager with only the uefi entry. I need to go through "Boot from file" and reinstall grub.

During the firmware upgrade to 8.40 something changed: At first, the "Boot from file" dialog was empty (not showing my SSD). After a little bit of panic fiddling, I found it when rebooting, then pressing enter for "boot immediately" which opened the UEFI shell (with different messages than opening it normally). Typing `exit` there got me to the ui where "Boot from file" showed my SSD. Opening the boot manager without the "boot immediately" workaround didn't allow to select the SSD.

It is not a pressing matter, but I wonder if there is a way to avoid these manual steps and have the firmware upgrade preserve the boot entries.

Sean-StarLabs, 2023-04-25:
Howdy,

The short answer is no; everything is "new" after an update on coreboot, so the boot entries will always go. edk2 will automatically scan and try to add entries.

It doesn't work for Debian, see https://forums.debian.net/viewtopic.php?t=152138&hilit=uefi. One users said the solution in that did work:
```
grub-install --force-extra-removable
dpkg-reconfigure grub-efi-amd64
```

fnetX, 2023-04-29:
Ah I see. I don't think that the installation on removable media (there aren't any for me) would help, but I'll check that out. Thank you.

closed 2023-04-29 by fnetX