StarLabsLtd/firmware #104: [Starbook MkVI - Intel][coreboot] OSResearch Heads support?

Hacksawfred3232, 2023-04-26:
I just wanted to know if Heads could be supported as a alternative payload for coreboot on this device?
https://github.com/osresearch/heads

This would pair well with Qubes OS, which is now stable - somewhat. https://www.qubes-os.org/hcl/#star-labs_starbook-mk-vi_i7-1260p_integrated-graphics-iris-xe_hsf3232_r4-1

Sean-StarLabs, 2023-04-26:
There's nothing stopping it, would just need to be ported

Hacksawfred3232, 2023-04-26:
> There's nothing stopping it, would just need to be ported

- I've noted that the latest available Coreboot version for Heads is 4.19, while we're on 8.40+.
- There is also mentions of a i386 - 32bit architecture for compiling the firmware itself. Unless that's par for the course.
- Lastly, there is also the question of needing all regions unlocked for some of their functions. Unless that's just a Lenovo thing. 

This is why I held off on trying to see if I can do the port job myself, since it looked like it could involve a lot more effort than tweaking some options and quick code patches. Anything more, and I would have to familiarize myself with both the coreboot firmware and Heads source code.

I'll post an issue on their git repo, see what would be required to port things over.

Sean-StarLabs, 2023-04-26:
4.19 is the latest version of coreboot. 8.40 is the local version number.

I'm not familiar with heads, but unlocking all regions sounds like a terrible idea.

Mailing lists are a much better place for questions :)

Hacksawfred3232, 2023-04-26:
> 4.19 is the latest version of coreboot. 8.40 is the local version number.

Just noticed that now! :facepalm: Thanks for the heads up.

tlaurion, 2023-05-27:
> There's nothing stopping it, would just need to be ported
@Sean-StarLabs @Hacksawfred3232 

I updated heads ticket. Might want to contribute there for it to happen!

tlaurion, 2023-05-27:
> unlocking all regions sounds like a terrible idea.

This is a blob redistribution issue mitigation, mainly. Lenovo ships with a neuteurable ME, which blobs cannot be redistributed legally. So those blobs are downloaded from Lenovo, extracted and out in local build system, GBE is generated, and ifd is provided.

There is nothing preventing us to relock ME region prior of flashing, but as today, internal upgrades simply flash the whole SPI, including ifd, ME, GBE and the BIOS region.

Otherwise we could not take advantage of ME freed space (~5mb) that needs to be relegated to BIOS region.

At the end of that process on sandy/ivy bridge, ME region is <=98kb and BIOS region for xx30 is >11.5mb, where xx20 is ~7.8.

Agreed that ME and IFD could relock those regions. But it is a nice auditability feature to be able to use flashrom directly whole SPI chip(s) where ME doesn't exist anymore (BUP only on xx20, BUP+ROMP on xx30).

For those Lenovo, ME is neutered not just deactivated per HAP. Older models could have ME removed. Nowadays ME is still in the ROM, just disagled/deactivated, where possible.