StarLabsLtd/firmware #105: [Starbook MkVI -Intel][Coreboot] Suspend still not working

aaron-gh, 2023-05-05:
No mode of suspend works at all, the machine just wakes up as soon as suspend is attempted.

Does not matter if attempting s2idle or s3.

All coreboot settings are default, other than max charge, which is 80%.

Not sure what logs would be most helpful to look into this issue.

divico, 2023-05-06:
I use suspend heavily and never had any issue with suspend to ram `S3` or suspend to disk `S4` running coreboot `8.40` and EC `1.12`. If I'm not mistaken, `s2idle` is pure software sleep state which is always available, regardless of hardware. Are  you sure it is not pure software configuration issue? You could try booting into a fresh Ubuntu LTS or another distrib known to work well.

aaron-gh, 2023-05-06:
Using the same combination. s3 and s4 both fail, so does s2idle.

On Ubuntu 22.04, but have also tried with both Debian stable and Debian testing.

Sean-StarLabs, 2023-05-06:
It's certainly sounds like software - testing with `sudo systemctl suspend` from a live USB with zero modifications would rule that out though.

Or we could look at the coreboot debug log - you'd need to get `cbmem` from `util` in the coreboot repository.

If not, logs - probably all of them! If you can ping an email to support, we can get a full set and see if we can see anything.

S2idle isn't supported so that won't work.

aaron-gh, 2023-05-06:
Hi,

sudo systemctl suspend on a live USB also fails.

I will work on getting the logs gathered, including cbmem.

For clarification, all the things I have tried were with both charger connected and disconnected.

mysfitt, 2023-05-18:
I just wanted to add that I had the same issue when I updated to ITE 1.13 and Coreboot 8.41 recently. Dmesg output indicated that it was a spurious USB wakeup that was happening pretty much constantly on the USB3 bus. So essentially, the machine would go to S3 and the USB bus would wake it up immediately. And this would loop over and over while the machine was supposed to be sleeping (lid switch event). I downgraded coreboot to 8.40 and everything is working again.

aaron-gh, 2023-05-24:
This is the same as what I am experiencing. However in my case downgradng did not fix the issue.

Sean-StarLabs, 2023-07-21:
Going to merge this into #96 as they're very similar

closed 2023-07-21 by Sean-StarLabs